LORD-MicroStrain/MSCL #1: Getting error trying to build MSCL for python

crosvera, 2015-09-30:
Hello,

I'm trying to build the package under Ubuntu 14.04, I ran the "buildAll.sh" script but I get some errors related to the "target"

========================================================
(buildAll.sh) STARTING MSCL BUILD
(buildAll.sh) Creating the build\swig-python directory
(buildAll.sh) Running BJAM (C++)
warning: No toolsets are configured.
warning: Configuring default toolset "gcc".
warning: If the default is wrong, your build may not work correctly.
warning: Use the "toolset=xxxxx" option to override our guess.
warning: For more configuration options, please consult
warning: http://boost.org/boost-build2/doc/html/bbv2/advanced/configuration.html
/usr/share/boost-build/build-system.jam:197: in find-target
warning: object is empty
/usr/share/boost-build/build-system.jam:661: in load
/usr/share/boost-build/kernel/modules.jam:289: in import
/usr/share/boost-build/kernel/bootstrap.jam:139: in boost-build
/usr/share/boost-build/boost-build.jam:8: in module scope
notice: could not find main target MSCL//stage_c++
notice: assuming it is a name of file to create.
error: Project target requested but not yet assigned for module 'Jamfile</home/crosvera/MSCL/BuildScripts>'.

(buildAll.sh) Running BJAM (Python)
warning: No toolsets are configured.
warning: Configuring default toolset "gcc".
warning: If the default is wrong, your build may not work correctly.
warning: Use the "toolset=xxxxx" option to override our guess.
warning: For more configuration options, please consult
warning: http://boost.org/boost-build2/doc/html/bbv2/advanced/configuration.html
/usr/share/boost-build/build-system.jam:197: in find-target
warning: object is empty
/usr/share/boost-build/build-system.jam:661: in load
/usr/share/boost-build/kernel/modules.jam:289: in import
/usr/share/boost-build/kernel/bootstrap.jam:139: in boost-build
/usr/share/boost-build/boost-build.jam:8: in module scope
notice: could not find main target MSCL//stage_python
notice: assuming it is a name of file to create.
error: Project target requested but not yet assigned for module 'Jamfile</home/crosvera/MSCL/BuildScripts>'.

(buildAll.sh) Running BJAM (Python) (2)
warning: No toolsets are configured.
warning: Configuring default toolset "gcc".
warning: If the default is wrong, your build may not work correctly.
warning: Use the "toolset=xxxxx" option to override our guess.
warning: For more configuration options, please consult
warning: http://boost.org/boost-build2/doc/html/bbv2/advanced/configuration.html
/usr/share/boost-build/build-system.jam:197: in find-target
warning: object is empty
/usr/share/boost-build/build-system.jam:661: in load
/usr/share/boost-build/kernel/modules.jam:289: in import
/usr/share/boost-build/kernel/bootstrap.jam:139: in boost-build
/usr/share/boost-build/boost-build.jam:8: in module scope
notice: could not find main target MSCL//stage_python
notice: assuming it is a name of file to create.
error: Project target requested but not yet assigned for module 'Jamfile</home/crosvera/MSCL/BuildScripts>'.

Do you have any idea?

Thanks in advance,
Regards

rwslord, 2015-09-30:
hmm...weird. Running "buildAll.sh" is the right thing to do.
Could you tell me what version of bjam you are running?

crosvera, 2015-09-30:
Hello Richard,

Thanks for the reply, actually the bjam version is:
% bjam -v
Boost.Jam  Version 2011.12.1. OS=LINUX.
   Copyright 1993-2002 Christopher Seiwald and Perforce Software, Inc.
   Copyright 2001 David Turner.
   Copyright 2001-2004 David Abrahams.
   Copyright 2002-2008 Rene Rivera.
   Copyright 2003-2008 Vladimir Prus.

Best regards.

rwslord, 2015-09-30:
No problem! That's the same version of bjam I am using on our linux build machine as well (although it is Debian, not Ubuntu, so there's a difference there).

Could you try this:
-create a file in your home directory called "user-config.jam"
-within it put the following:
```
using gcc
:	host
:	g++
:	<cxxflags>-I/usr/include/python2.7
;
```
-save and try building again

crosvera, 2015-09-30:
Ok, I did it. but I still getting the error:

% cat ~/user-config.jam
using gcc
:   host
:   g++
:   <cxxflags>-I/usr/include/python2.7
;

% cd ~/MSCL/BuildScripts

% sh buildAll.sh
========================================================
(buildAll.sh) STARTING MSCL BUILD
(buildAll.sh) Creating the build\swig-python directory
(buildAll.sh) Running BJAM (C++)
/usr/share/boost-build/build-system.jam:197: in find-target
warning: object is empty
/usr/share/boost-build/build-system.jam:661: in load
/usr/share/boost-build/kernel/modules.jam:289: in import
/usr/share/boost-build/kernel/bootstrap.jam:139: in boost-build
/usr/share/boost-build/boost-build.jam:8: in module scope
notice: could not find main target MSCL//stage_c++
notice: assuming it is a name of file to create.
error: Project target requested but not yet assigned for module
'Jamfile</home/crosvera/MSCL/BuildScripts>'.

(buildAll.sh) Running BJAM (Python)
/usr/share/boost-build/build-system.jam:197: in find-target
warning: object is empty
/usr/share/boost-build/build-system.jam:661: in load
/usr/share/boost-build/kernel/modules.jam:289: in import
/usr/share/boost-build/kernel/bootstrap.jam:139: in boost-build
/usr/share/boost-build/boost-build.jam:8: in module scope
notice: could not find main target MSCL//stage_python
notice: assuming it is a name of file to create.
error: Project target requested but not yet assigned for module
'Jamfile</home/crosvera/MSCL/BuildScripts>'.

(buildAll.sh) Running BJAM (Python) (2)
/usr/share/boost-build/build-system.jam:197: in find-target
warning: object is empty
/usr/share/boost-build/build-system.jam:661: in load
/usr/share/boost-build/kernel/modules.jam:289: in import
/usr/share/boost-build/kernel/bootstrap.jam:139: in boost-build
/usr/share/boost-build/boost-build.jam:8: in module scope
notice: could not find main target MSCL//stage_python
notice: assuming it is a name of file to create.
error: Project target requested but not yet assigned for module
'Jamfile</home/crosvera/MSCL/BuildScripts>'.

(buildAll.sh) Building Unit Tests
notice: could not find main target MSCL_Unit_Tests
notice: assuming it is a name of file to create.
error: Project target requested but not yet assigned for module
'Jamfile</home/crosvera/MSCL/BuildScripts>'.

(buildAll.sh) Running Unit Tests
buildAll.sh: 26: buildAll.sh: cannot create
MSCL_Unit_Tests/unitTestResults.xml: Directory nonexistent
(buildAll.sh) Zipping the MSCL Output
bash: BuildScripts/zipOutput.sh: No such file or directory
(buildAll.sh) DONE

This is getting weird...

crosvera, 2015-09-30:
I think now I get it.

My problem  was trying to run buildAll.sh from BuildScripts subdirectory,
must be run in from the root-project directory. Now it's compiling...

Thank you Richards

rwslord, 2015-09-30:
ok, I think the problem is that you are running buildAll.sh from the wrong location...I need to add documentation on this :) 

bjam (and therefore the buildAll.sh script) needs to be run from the same directory as the Jamroot.jam file. So go up a folder and run the script from there (```bash BuildScripts/buildAll.sh```)

rwslord, 2015-09-30:
ha, beat me to it!

closed 2015-09-30 by rwslord