LORD-MicroStrain/MSCL #2: Compiling problems with 3 targets

crosvera, 2015-10-01:
Hello @rwslord ,

I cloned the project and tried to compile it under Ubuntu 14.04 but bjam gave me the following problem:

    % bjam MSCL//stage_c++ release
    ...patience...
    ...found 835 targets...
    ...updating 3 targets...
    gcc.compile.c++ /home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release/source/mscl/Utils.o
    MSCL/source/mscl/Utils.cpp: In function ‘void mscl::Utils::filterSensorcloudName(std::string&)’:
    MSCL/source/mscl/Utils.cpp:406:68: error: no matching function for call to ‘regex_replace(std::string&, std::regex, const char [2])’
        str = std::regex_replace(str, std::regex("[^a-zA-Z0-9-_.]"), "-");
                                                                        ^   
    MSCL/source/mscl/Utils.cpp:406:68: note: candidates are:
    In file included from /usr/include/c++/4.8/regex:62:0,
                     from MSCL/source/stdafx.h:42,
                     from MSCL/source/mscl/Utils.cpp:14:
    /usr/include/c++/4.8/bits/regex.h:2162:5: note: template<class _Out_iter, class _Bi_iter, class _Rx_traits, class _Ch_type> _Out_iter std::regex_replace(_Out_iter, _Bi_iter, _Bi_iter, const std::basic_regex<_Ch_type, _Rx_traits>&, const std::basic_string<_Ch_type>&, std::regex_constants::match_flag_type)
         regex_replace(_Out_iter __out, _Bi_iter __first, _Bi_iter __last,
         ^                                                                                                                                                                                       
    /usr/include/c++/4.8/bits/regex.h:2162:5: note:   template argument deduction/substitution failed:
    MSCL/source/mscl/Utils.cpp:406:68: note:   deduced conflicting types for parameter ‘_Bi_iter’ (‘std::basic_regex<char>’ and ‘const char*’)
        str = std::regex_replace(str, std::regex("[^a-zA-Z0-9-_.]"), "-");
                                                                        ^   
    In file included from /usr/include/c++/4.8/regex:62:0,
                     from MSCL/source/stdafx.h:42,
                     from MSCL/source/mscl/Utils.cpp:14:
    /usr/include/c++/4.8/bits/regex.h:2182:5: note: template<class _Rx_traits, class _Ch_type> std::basic_string<OutputT> std::regex_replace(const std::basic_string<OutputT>&, const std::basic_regex<_Ch_type, _Rx_traits>&, const std::basic_string<OutputT>&, std::regex_constants::match_flag_type)
         regex_replace(const basic_string<_Ch_type>& __s,
         ^   
    /usr/include/c++/4.8/bits/regex.h:2182:5: note:   template argument deduction/substitution failed:
    MSCL/source/mscl/Utils.cpp:406:68: note:   mismatched types ‘const std::basic_string<OutputT>’ and ‘const char [2]’
        str = std::regex_replace(str, std::regex("[^a-zA-Z0-9-_.]"), "-");
                                                                        ^   

        "g++"  -ftemplate-depth-128 -I/usr/include/python2.7 -O3 -finline-functions -Wno-inline -Wall -fPIC -std=c++11 -DNDEBUG -DUNIX_SOCKETS  -I"MSCL/source" -c -o "/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release/source/mscl/Utils.o" "MSCL/source/mscl/Utils.cpp"

    ...failed gcc.compile.c++ /home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release/source/mscl/Utils.o...
    ...skipped <p/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release>libmscl.so for lack of <p/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release>source/mscl/Utils.o...
    ...skipped <p/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/Output/C++/Release>libmscl.so for lack of <p/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release>libmscl.so...
    ...failed updating 1 target...
    ...skipped 2 targets...

Any idea? 

Best regards

rwslord, 2015-10-01:
Can you check your version of gcc?

rwslord, 2015-10-01:
It looks like std::regex (part of C++11) is only available in gcc 4.9+ 
From your output, it looks like you might have 4.8. Are you able to upgrade?

crosvera, 2015-10-01:
% g++ --version
    g++ (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4
    Copyright (C) 2013 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Yeah I'm compiling the source within a VM with debian (gcc 4.9) and seems it's working. I will update my gcc version in my Ubuntu machine and I will tell you if it worked.

Regards and thanks for your reply again.

rwslord, 2015-10-01:
Ok, yea let me know how it goes.

Also, if you'd like to discuss details on the project you are working on, feel free to contact me at richard.stoneback@lord.com. The priorities of adding features is mainly driven by customers right now, so if there are any functions missing that you need, I might be able to push them up the list.

crosvera, 2015-10-01:
Ok, now with `% bjam MSCL//stage_c++ release` works and copile the c++ code.

Now I have a new error in the next command, but I will open a new issue for it.

Regards

closed 2015-10-01 by crosvera