LORD-MicroStrain/MSCL #3: Problems with building the python stage

crosvera, 2015-10-01:
Hello @rwslord,

I've successful compiled the source code with `bjam MSCL//stage_c++ release`, but when I want to compile the Python module I got this error:

    % bjam MSCL//stage_python release                                                                                                                                                            
    ...patience...
    ...found 839 targets...
    ...updating 4 targets...
    swig.swig /home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp
    /bin/sh: 2: Syntax error: "(" unexpected

        swig -c++ -DUNIX_SOCKETS -outdir /home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/swig-python -python -o /home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp MSCL/source/mscl/Wrapper/MSCL_Main_Interface.i

    ...failed swig.swig /home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp...
    ...skipped <p/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release>source/mscl/Wrapper/MSCL_Main_Interface.o for lack of <p/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release-object(c-scanner)@993>source/mscl/Wrapper/MSCL_Main_Interface.cpp...
    ...skipped <p/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release>_mscl.so for lack of <p/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release>source/mscl/Wrapper/MSCL_Main_Interface.o...
    ...skipped <p/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/Output/Python>_mscl.so for lack of <p/home/crosvera/Dropbox (Damage Monitor)/trabajo/MSCL/build/MSCL/gcc-host/release>_mscl.so...
    ...failed updating 1 target...
    ...skipped 3 targets...

Any idea why?

Regards

rwslord, 2015-10-01:
It looks like there might be a bug in the swig.jam file (in the MSCL folder) that isn't putting quotes around the path. So the space in "Dropbox (Damage Monitor)" is likely causing the problem.

I will hopefully fix this soon. In the meantime, you can try either getting rid of the space in your path, or changing the line in swig.jam from:

```swig -c++ $(OPTIONS) -o $(<) $(>)```
to
```swig -c++ $(OPTIONS) -o "$(<)" "$(>)"```

crosvera, 2015-10-01:
Well editing swig.jam at those line you said and the Jamfile at lines 74 and 86:

	<swig-options>"-outdir $(SWIG_PYTHON)"
to:
	<swig-options>"-outdir \"$(SWIG_PYTHON)\""

Solves the problem. Thanks

I made a pull request to fix this.
Regards

closed 2015-10-01 by crosvera