LORD-MicroStrain/MSCL #5: Invalid Configuration

crosvera, 2015-10-09:
Hello there,

I have a G-Link2 mode and I want to start sampling in LowDutyCycle, but setting the configuration I got a "Invalid Configuration" Error, my code is the following:

    import sys
    import time

    import mscl

    def node_set_idle(node, tries=20):
        print "Attemting to set in Idle mode node"
        idleStatus = node.setToIdle()
        n = 0
        while not idleStatus.complete():
            sys.stdout.write(".")
            if n == 10:
                idleStatus.cancel()
            n += 1
            sys.stdout.flush()
            time.sleep(0.7)
        print ""

        result = idleStatus.result()
        if result == mscl.SetToIdleStatus.setToIdleResult_success:
            print "\nNode now stoped and in Idle mode"
        elif result == mscl.SetToIdleStatus.setToIdleResult_canceled:
            print "\nSet to Idle was canceled"
        else:
            print "\nSet to Idle failed"

    connection = mscl.Connection.Serial(sys.argv[1], 921600)

    basestation = mscl.BaseStation(connection)

    node = mscl.WirelessNode(int(sys.argv[2]), basestation)

    node_set_idle(node)

    response = node.ping()

    if response.success():
        print response.baseRssi()
        print response.nodeRssi()

    time.sleep(0.5)
    print node.getNumDatalogSessions()
    time.sleep(0.5)

    print node.getNumSweeps()

    time.sleep(1)
    time.sleep(1)

    config = mscl.WirelessNodeConfig()
    #config.bootMode(mscl.WirelessTypes.bootMode_normal)
    #config.inactivityTimeout(7200)
    config.samplingMode(mscl.WirelessTypes.samplingMode_nonSync)
    config.sampleRate(mscl.WirelessTypes.sampleRate_16Hz) #sampleRate_16Hz)
    #config.unlimitedDuration(True)

    while True:
        try:
            #node_set_idle(node)
            #time.sleep(0.8)
            node.applyConfig(config)
            time.sleep(1)
        except (mscl.Error_InvalidNodeConfig, mscl.Error_Communication), e:
            print e
            continue
        break

    print node.startNonSyncSampling()

When I run the script I got:

    Attemting to set in Idle mode node
    ..

    Node now stoped and in Idle mode
    -36
    -32
    0
    100
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.
    Invalid Configuration.

Am I doing right the configuration?
Best Regards.

rwslord, 2015-10-09:
So the G-Link2 node, for Low Duty Cycle (non-sync) sampling, only supports rates 32hz - 512 hz. You are attempting to sample at 16Hz, which I believe is your problem.

rwslord, 2015-10-09:
Also, to determine the point of failure causing the Invalid Configuration, you can use the `node.verifyConfig(WirelessNodeConfig, ConfigIssues) function. 

This takes in your config object and an empty `ConfigIssues` container. When it is returned back to you, if it returns false (failed to verify), you can iterate over the `ConfigIssues` container which will have been populated with the issue ids and strings letting you know what errors were found.

rwslord, 2015-10-12:
Did changing the sample rate seem to solve the problem?

crosvera, 2015-10-12:
You was right, I had to change the rate value. Thanks again!

closed 2015-10-12 by crosvera