LORD-MicroStrain/MSCL #6: WirelessNode.channels() method doesn't exists

crosvera, 2015-10-30:
Hello @rwslord 

In the Example file at MSCL_Examples/Wireless/Python/ConfigureNode.py, it does the following:

    # get a list of the supported channels
    supportedChannels = node.channels()

But when I tried it I got:

	---------------------------------------------------------------------------
	AttributeError                            Traceback (most recent call last)
	<ipython-input-30-1dd55c451f70> in <module>()
	----> 1 node.channels()

	/home/crios/venvs/us2dm/lib/python2.7/site-packages/mscl/mscl.pyc in <lambda>(self, name)
	   1504     __setattr__ = lambda self, name, value: _swig_setattr(self, WirelessNode, name, value)
	   1505     __swig_getmethods__ = {}
	-> 1506     __getattr__ = lambda self, name: _swig_getattr(self, WirelessNode, name)
	   1507     __repr__ = _swig_repr
	   1508     def __init__(self, *args):

	/home/crios/venvs/us2dm/lib/python2.7/site-packages/mscl/mscl.pyc in _swig_getattr(self, class_type, name)
	     55     method = class_type.__swig_getmethods__.get(name,None)
	     56     if method: return method(self)
	---> 57     raise AttributeError(name)
	     58 
	     59 def _swig_repr(self):

	AttributeError: channels

What I think is working is the following:
	ac = node.getActiveChannels()
	active_channels = []
	for i in range(1, ac.MAX_CHANNELS+1):
	    if ac.enabled(i): active_channels.append(i)

Best Regards

rwslord, 2015-10-30:
good find. The example code needs to be updated to the new method for channel specific configuration options. I will push this up ASAP.

crosvera, 2015-10-30:
Can you point me the new method name?

Thanks

rwslord, 2015-10-30:
If you just want to get a list of the channels that are enabled, then what you have with the `node.getActiveChannels()` is what you want.

If you want to get or set options on the Node that are specific to channels, this is similar to the way to get/set any other option on the Node, but the functions will have `ChannelMask` parameters.

First off, you can get the list of `ChannelGroups` that are supported by the Node by doing:

```python
features = node.features()
chGroups = features.channelGroups()
```

Once you have the `ChannelGroups`, you can iterate over them to get the info each contains:
```python
for group in chGroups:
  mask = group.channels()  # the mask that is associated with this group
  name = group.name() # the name of the group
  settings = group.settings() # the WirelessTypes.ChannelGroupSettings that apply to this group
```

The `WirelessTypes.ChannelGroupSettings` contain a list of enums that represent different configuration settings that are per-ChannelGroup settings (for instance, if the group's settings contains `WirelessTypes.chSetting_linearEquation`, you know that it supports setting the linear equation for that group).

Once you know each group's `ChannelMask` and that settings that can be applied for it, you can then go back to the `WirelessNode` and use the appropriate function:
```python
node.getLinearEquation(groupMask) # pass the mask for a group that supports linear equation
```

Some `ChannelGroup`s might just contain a single channel (identified via the group's channel mask). However, others may contain multiple channels, indicating that the setting on the Node is applied to all of the channels in the mask.

crosvera, 2015-10-30:
Thanks for the info @rwslord,

Related with this, ¿How can I get the Gain value?, I tried to do the following:

    chGroups = features.channelGroups()
    masks = {g.name():g.channels() for g in chGroups 
             if int(g.name().split(" ")[-1]) in active_channels}

    for m in masks:
        print node.getHardwareGain(masks[m])

But I get this error:

    ---------------------------------------------------------------------------
    Error_NotSupported                        Traceback (most recent call last)
    <ipython-input-158-b6dda7ceb73d> in <module>()
    ----> 1 for m in masks: print node.getHardwareGain(masks[m])

    /home/crios/venvs/us2dm/lib/python2.7/site-packages/mscl/mscl.pyc in getHardwareGain(self, *args)
       1561     def getTimeBetweenBursts(self): return _mscl.WirelessNode_getTimeBetweenBursts(self)
       1562     def getLostBeaconTimeout(self): return _mscl.WirelessNode_getLostBeaconTimeout(self)
    -> 1563     def getHardwareGain(self, *args): return _mscl.WirelessNode_getHardwareGain(self, *args)
       1564     def getHardwareOffset(self, *args): return _mscl.WirelessNode_getHardwareOffset(self, *args)
       1565     def getLinearEquation(self, *args): return _mscl.WirelessNode_getLinearEquation(self, *args)

    Error_NotSupported: The requested ChannelMask is not supported for this Node.

Any idea? Thanks.

rwslord, 2015-10-30:
First off, what type of Node do you have? Just want to make sure it actually supports hardware gain before you invest too much time in it. 

The group name that you are using (`g.name()`) is mainly for readability and doesn't map to anything. It looks like you are comparing it to everything in `active_channels`. Instead, loop over all of the groups, and for each group, loop over all of the settings (`g.settings()`). If a setting is equal to `mscl.WirelessTypes.chSetting_hardwareGain` then you know you can perform `node.getHardwareGain(mask)` where the `mask` parameter is the channel mask for that group (`g.channels()`).

crosvera, 2015-11-02:
Thanks for pointing that @rwslord, it seems that G-Link2 nodes does not supports the gain parameter. I will check with other kind of node, maybe sg-link.

Best Regards

rwslord, 2015-11-02:
No problem. SG-Link should support hardware gain. I will work on updating the example code to reflect these changes.

rwslord, 2015-11-02:
updated example code with release of [v1.20.1](https://github.com/LORD-MicroStrain/MSCL/commit/34aa661e705fba285aaa80516b2444b751fbe972)

closed 2015-11-02 by rwslord