LORD-MicroStrain/MSCL #7: Get DataLogging data

crosvera, 2015-11-04:
Hello @rwslord,

I'm trying to test the DataLogging mode in some G-Link2 nodes using the MSCL framework, I can successfully create an ArmedDataloggingNetwork, add it some nodes and start sampling, but what I really didn't find is an ArmedDataloggingNetwork or WirelessNode object's method to collect the data stored. Maybe I missed something to read in the docs? 

Best regards.

rwslord, 2015-11-04:
You are looking for is the `DatalogDownloader` class. [API reference](http://lord-microstrain.github.io/MSCL/Documentation/MSCL_Public_Documentation/index.html#File:MicroStrain/Wireless/DatalogDownloader.h:DatalogDownloader.DatalogDownloader)

You will want to keep calling `getNextData()` on it until `complete()` returns true.

rwslord, 2015-11-05:
@crosvera - did this work for you?

crosvera, 2015-11-05:
Well @rwslord  I haven't been successful, I have the following code:

    from mscl import mscl
    import sys
    import time

    def node_set_idle(node, tries=20):
        print "Attempting to set in Idle mode node"
        idleStatus = node.setToIdle()
        n = 0
        while not idleStatus.complete():
            sys.stdout.write(".")
            if n == 10:
                idleStatus.cancel()
            n += 1
            sys.stdout.flush()
            time.sleep(0.7)
        print ""

        result = idleStatus.result()
        if result == mscl.SetToIdleStatus.setToIdleResult_success:
            print "\nNode now stoped and in Idle mode"
        elif result == mscl.SetToIdleStatus.setToIdleResult_canceled:
            print "\nSet to Idle was canceled"
        else:
            print "\nSet to Idle failed"

    port = "/dev/ttyUSB0"
    baud_rate = 921600

    node_address = [45635, 45632]

    connection = mscl.Connection.Serial(port, baud_rate)
    basestation = mscl.BaseStation(connection)

    nodes = []
    for na in node_address:
        node = mscl.WirelessNode(na, basestation)
        nodes.append(node)

    config = mscl.WirelessNodeConfig()
    config.samplingMode(mscl.WirelessTypes.samplingMode_armedDatalog)
    config.sampleRate(mscl.WirelessTypes.sampleRate_32Hz)

    for node in nodes:
        node.applyConfig(config)
        time.sleep(1)
        node_set_idle(node)

    network = mscl.ArmedDataloggingNetwork(basestation)

    for node in nodes:
        network.addNode(node)

    #Sampling for 10 secs
    network.startSampling()
    time.sleep(10)

    for node in nodes:
        try:
            node_set_idle(node)
        except mscl.Error_Communication, e:
            print e

    for node in nodes:
        dwn = mscl.DatalogDownloader(node)
        na = node.nodeAddress()
        while not dwn.complete():
            sweep = dwn.getNextData()
            ts = sweep.timestamp()
            pnts = [p.as_float() for p in sweep.data()]
            print ns, ts, pnts

And when I run it I got:

    Attempting to set in Idle mode node
    ..

    Node now stoped and in Idle mode
    Attempting to set in Idle mode node
    ..

    Node now stoped and in Idle mode
    Attempting to set in Idle mode node
    ...........Failed to read EEPROM 108 from the BaseStation.
    Attempting to set in Idle mode node
    ........

    Node now stoped and in Idle mode
    Traceback (most recent call last):
      File "testdl.py", line 73, in <module>
        dwn = mscl.DatalogDownloader(node)
      File "/home/crios/Dropbox (Damage Monitor)/trabajo/mscl/mscl.py", line 1586, in __init__
        this = _mscl.new_DatalogDownloader(*args)
    mscl.mscl.Error_NodeCommunication: Failed to read EEPROM 0 from Node 45635
 

What I'm doing wrong?

Best regards.

rwslord, 2015-11-05:
There are a couple issues. 

**First**, the Set to Idle operation can take up to 2 minutes (usually takes less than 20 seconds). It looks like you are canceling very quickly.

**Second**, I think you may have found a bug with set to idle. I've added this as an issue ( #8 ) and will work on fixing it soon. In the meantime, you can ping the Base Station before doing anything else in your program, which will immediately go and cache the firmware version so that canceling set to idle doesn't have to perform a read eeprom command.

crosvera, 2015-11-06:
It's seems that giving more time to setting the node into idle node is working, so now I can do the DataLogging.

Thanks again @rwslord

rwslord, 2015-11-09:
closing this issue. The other set to idle issue has been created in #8

closed 2015-11-09 by rwslord