LORD-MicroStrain/MSCL #10: MSCL on ARM platforms

crosvera, 2015-12-28:
Has anyone tried it over an ARM platform?

How do you compile it for ARM?

Best regards.

rwslord, 2016-01-04:
Yes, our [WSDA-1500](http://www.microstrain.com/wireless/wsda-1500) uses MSCL, and is an ARM platform.

There shouldn't be much different that you have to do. Just tell bjam/boostbuild to point to the compiler that you want to use: http://www.boost.org/build/doc/html/bbv2/overview/configuration.html

rwslord, 2016-01-07:
Did you get it compiling?

crosvera, 2016-01-07:
I'm on it Richard.

Best Regards

crosvera, 2016-01-14:
Finally we compiled it inside an ARM machine due to the problems found trying to compiling, especially when the python module was generating.

rwslord, 2016-01-14:
Sorry that you had issues compiling. Were there any specific that should be addressed here?

crosvera, 2016-01-14:
Uhm, the main thing is how to build the tool-chain used for cross-compiling. I had installed the arm-linux-gnueabihf-gcc/g++ but at build time I got errors when the build process tried to get some libboost libs and python headers.

Did you compile libboost and python fom ARM target?

best regards

rwslord, 2016-01-14:
For our WSDA-1500, we use [Yocto](https://www.yoctoproject.org/) for building all dependencies for ARM. It will, however, require varying amounts of work to get going depending on your platform. If you happen to be working with a beaglebone black, we may be able to share some resources.

crosvera, 2016-01-18:
Well @rwslord, I will be very thankful if you could share those resources you talked about.

Best regards.

rwslord, 2016-01-18:
Send me an email at richard_stoneback@lord.com

jpherbst, 2016-01-19:
Hey Carlos,

I've added a Yocto layer to our github that contains a recipe for mscl and recipes for all of the dependencies that are not included in standard Yocto.  I tested it with poky-fido-13.0.1 so I would suggest building with that version.

https://github.com/LORD-MicroStrain/meta-lord

You should be able to clone that into your yocto directory, add it to your bblayers.conf, and build.

crosvera, 2016-01-19:
Great! Thank you @rwslord, we are going to try it.
Best regards

closed 2016-01-19 by crosvera

b-harned, 2018-11-06:
Hello @jpherbst @rwslord, I am just starting on getting a Lord IMU working on a BeagleBone Black running Debian 8.7. Do you have any advice before I dive in? Just want to check with you guys in case there has been more testing/development on this since early 2016.

jpherbst, 2018-11-06:
I would recommend cross compiling on a more powerful Debian computer.  If you are able to setup a cross compiling environment with the necessary dependencies you should be able to follow the instructions for Linux and copy the compiled binaries over to the BeagleBone.  Trying to compile on the BeagleBone will take a very long time and you may run our of RAM, as some Raspberry Pi users have found.

We actually use MSCL on a BeagleBone Black in our WSDA-2000 product, but we use Yocto instead of Debian.  Unfortunately binaries generated using Yocto likely will not work in Debian.