LORD-MicroStrain/MSCL #11: Linux Operating System

kerolos, 2016-04-16:
is it possible to run this program in Linux OS ? by using "make file", in the case it can work. Is the code  available ? 
can I read the data from Multi-Inertia sensors simultaneously ?
thanks in advance.

rwslord, 2016-04-18:
Yes, MSCL can be used on Linux. There are currently no pre-built binaries available at this time (something we want to make available in the future). 

However, you can build it yourself by running the [buildAll.sh](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildAll.sh) file. Make sure you run it from the MSCL's root directory (not the build scripts directory). 

Some of the closed issues might help you as well (like #1 and #2).

rwslord, 2016-04-25:
Just saw your other question...you can definitely read data from multiple Inertial sensors plugged in at the same time. Each `InertialNode` would just get its own loop parsing through the data. MSCL spins up a separate thread for each device that you create, so the data is parsed independently for each connection.

rwslord, 2016-06-08:
@kerolos - did you figure out what you needed with MSCL on Linux and with your Inertial sensors?

closed 2018-03-08 by rwslord