LORD-MicroStrain/MSCL #12: Online data reading from wireless sensors

mahatos, 2016-06-08:
Currently we are using Lord wireless sensors for acceleration recording. In my study, I need to access the acceleration data in real time. I contacted the Indian customer support and they advise to go through MSCL package.

So, I am using python program to access the data. My have trouble to retrieve the data. The sample code given in 'ParseData.py' is running an infinite loop and inside that loop I am unable to access 'sweep.data()'.

rwslord, 2016-06-08:
Can you explain more of what you mean by "unable to access sweep.data()"? 
Does the python program throw an error on that line?

mahatos, 2016-06-09:
I have been trying following the example codes and I made like : 

`import sys
sys.path.append("../../dependencies/Python")
import mscl

COM_PORT = "COM12"
NODE_ADDRESS = 31905

try:
    # create a SerialConnection with the specified COM Port, default baud rate of 921600
    connection = mscl.Connection.Serial(COM_PORT)

    # create a BaseStation with the connection
    baseStation = mscl.BaseStation(connection)
    
    # create a WirelessNode with the BaseStation we created
    node = mscl.WirelessNode(NODE_ADDRESS, baseStation)

    # create a SyncSamplingNetwork object, giving it the BaseStation that will be the master BaseStation for the network
    network = mscl.SyncSamplingNetwork(baseStation)

    # add a WirelessNode to the network
    print "Adding node to the network...",
    network.addNode(node)
    print "Done."

    # can get information about the network
    print "Network info: "
    print "Network OK: ", network.ok()
    print "Percent of Bandwidth: ", network.percentBandwidth()
    print "Lossless Enabled: ", network.lossless()
    print "High Capacity Mode: ", network.highCapacity()

    # apply the network configuration to every node in the network
    print "Applying network configuration",
    network.applyConfiguration()
    print "Done."

    print "Starting the network...",
    network.startSampling_noBeacon()
    print "Done."

    while True:
        sweeps = baseStation.getData(500)

        for sweep in sweeps:
            # print out information about the sweep
            print "Packet Received",
            print "Node", sweep.nodeAddress(),
            print "Timestamp", sweep.timestamp(),
            print "Tick", sweep.tick(),
            print "Sample Rate", sweep.sampleRate().prettyStr(),
            print "Base RSSI: ", sweep.baseRssi(),
            print "Node RSSI: ", sweep.nodeRssi(),

            print "DATA: ",
            # iterate over each point in the sweep
            for dataPoint in sweep.data():
                # just printing this out as a string. Other methods (ie. as_float, as_uint16) are also available.
                print dataPoint.channelName(), ":", dataPoint.as_string(),

            print ""
        print sweep.data()
except Exception, e:
    print "Error:", e `

Here, my trouble is I am unable to read acceleration data which is supposed to stored in each 500 millisecond. How can I get the acceleration data while program is running an infinite loop?

rwslord, 2016-06-09:
It looks like the Node might not actually be sampling.

You are using the `startSampling_noBeacon()` function from the `SyncSamplingNetwork` class.
When using this function, the beacon does not get enabled. The nodes will be "started", but waiting to hear from a beacon to actually start sending data.

In most cases, you will want to use the [`startSampling()`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/SyncSamplingNetwork.h:SyncSamplingNetwork.startSampling) function. This will start the nodes, and enable the beacon.

Alternatively, you can still use the `startSampling_noBeacon()` function, and then later enable the beacon via the `BaseStation` [`enableBeacon()`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/BaseStation.h:BaseStation.enableBeacon) function.

rwslord, 2016-06-09:
Just realized that the example Python code uses that same `startSampling_noBeacon` function when it was meant to just use the `startSampling` function. I will fix this to get rid of the confusion.

mahatos, 2016-06-13:
Thank you @rwslord. The data is streaming but I am unable find the acceleration value. What is the command to read acceleration.

mahatos, 2016-06-13:
at present I am getting like 

`Packet Received Node 31905 Timestamp 2016-06-13 05:44:28.001750000 Tick 0 Sample Rate 128Hz Base RSSI:  -16 Node RSSI:  -23 DATA:  ch3 : 1200 <mscl.ChannelData; proxy of <Swig Object of type 'std::vector< mscl::WirelessDataPoint > *' at 0x018700C8> >

Packet Received Node 31905 Timestamp 2016-06-13 05:44:28.009562500 Tick 1 Sample Rate 128Hz Base RSSI:  -16 Node RSSI:  -23 DATA:  ch3 : 1202 <mscl.ChannelData; proxy of <Swig Object of type 'std::vector< mscl::WirelessDataPoint > *' at 0x01925B18> >

Packet Received Node 31905 Timestamp 2016-06-13 05:44:28.017375000 Tick 3 Sample Rate 128Hz Base RSSI:  -16 Node RSSI:  -23 DATA:  ch3 : 1202 <mscl.ChannelData; proxy of <Swig Object of type 'std::vector< mscl::WirelessDataPoint > *' at 0x018700C8> >`

rwslord, 2016-06-13:
Ok, so you can see you are getting data for 'ch3'. I'm guessing this is the only channel you have enabled on the Node. I'm not sure what type of Node you have, but if this is a G-Link, ch3 is Acceleration - Z. Enabling ch1 and ch2 would get you Accel - X and Accel - Y respectively.

The data values are around 1200. This looks like it is in raw bits, so the Node is likely set up to send uint16, uncalibrated data. Setting the [DataFormat](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Configuration/WirelessNodeConfig.h:WirelessNodeConfig.dataFormat) to `dataFormat_4byte_float` will tell the Node to send calibrated data over the air. This will use whatever Calibration Coefficients are stored on the Node for each channel. (If this is a G-Link, it was calibrated here before you received it.)

mahatos, 2016-06-14:
Thank you @rwslord. I am using 'G-Link-LXRS' and wish to read only channel 3 data. I tried to follow your instruction but unable implement. Can you please tell me in which place I have to edit the code and what will be the command?

rwslord, 2016-06-14:
The [ConfigureNode.py example](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL_Examples/Wireless/Python/ConfigureNode.py) shows how to get and set configuration settings for a Node. Specifically for your case, look at the `setCurrentConfig` function in that example.

What you need to do is use the `WirelessNodeConfig` object and use the `dataFormat` function, passing in `mscl.WirelessTypes.dataFormat_4byte_float`. Then take that `WirelessNodeConfig` and apply it to the `WirelessNode`:

```python
    # create a WirelessNodeConfig which is used to set all node configuration options
    config = mscl.WirelessNodeConfig()

    # set the configuration options that we want to change
    config.dataFormat(mscl.WirelessTypes.dataFormat_4byte_float)

    # attempt to verify the configuration with the Node we want to apply it to
    #   Note that this step is not required before applying, however the apply will throw an
    #   Error_InvalidNodeConfig exception if the config fails to verify.
    issues = mscl.ConfigIssues()
    if not node.verifyConfig(config, issues):
        print "Failed to verify the configuration. The following issues were found:"

        # print out all of the issues that were found
        for issue in issues:
            print issue.description()

        print "Configuration will not be applied."

    else:
        # apply the configuration to the Node
        node.applyConfig(config)
```

Note that once this configuration setting is applied to the Node, the Node will remember this even on power cycle. So unless you want to change configuration settings for each run of your program, you can just do this once and remove it. Setting this value via Node Commander / SensorConnect would achieve the same result. Once this value is set, you should receive the calibrated G's over the air, instead of bits.

mahatos, 2016-06-15:
Thank you very much @rwslord. It's working.

rwslord, 2016-06-15:
That's great to hear :smiley:

closed 2016-06-15 by rwslord