LORD-MicroStrain/MSCL #13: WSDA Base unavailable after long use.

crosvera, 2016-06-08:
I'm using mscl to read data from wireless nodes (glink2, dvrt-link, etc.), these nodes are monitoring for long time. What I found is that sometimes the WSDA Base stops working, I got errors about reading the EPROM. I close my reading program and check the base and nodes from NodeCommander but it appears as unavailable and the only way to get it work is unplug and plug again.

Have you experienced that? Do you have some sort of protocol of how one should start reading and stop the nodes?  maybe some good practices handling the nodes?

Thanks,
Best regards.

rwslord, 2016-06-08:
Can you tell me the firmware version of the BaseStation that shows this issue?
And is it WSDA-Base-104 (USB), WSDA-Base-101 (Analog), or WSDA-Base-102 (RS232)?

crosvera, 2016-06-08:
I have several bases all WSDA-Base-104 (USB), I think that the older firmware is the 3.06v. I cannot access to all the bases right now.

Best regards.

rwslord, 2016-06-08:
I think we should try updating the firmware on one of your Base Stations and seeing if the issue you are seeing goes away.

The latest firmware can be downloaded here: http://updates.microstrain.com/MicroStrain_Wireless_Firmware_2_2_5.zhex

To upgrade your Base Station, in Node Commander:
1. Make sure your Base Station is plugged in and shows up (not greyed out) in the device tree.
2. Hold F1 and Right-Click on the Base Station.
3. In the Firmware Upgrade window, click the 'Browse' button.
4. Browse to and select the zhex file that you downloaded above.
5. Click the 'Write' button and wait for the upgrade to complete.

Before going through the process of updating all of your devices, I would just update 1 and run your tests again to verify the problem has been solved.

crosvera, 2016-06-09:
@rwslord I found out that some base stations were with firmware v3.34, these base stations also presented the same problem.

rwslord, 2016-06-09:
Ok. The firmware file I linked to should upgrade the Base Stations to 3.37.
Since it's still happening on 3.34, I'm not as hopeful it will solve the problem, but it's worth a shot.

Can you describe more of what's actually happening? You are collecting data from the Nodes and then all of a sudden you stop getting data? Is an error/exception thrown?

You mention read eeprom failures. Is this after you set the Nodes to Idle (they can't be communicated with while they are sampling).

crosvera, 2016-06-15:
Hello Richard,

To give a little more of context:

The problem that I mentioned you occurs mainly when I measure with the
nodes in time intervals cycles, for example:

  While True: #loop all the time
    Stop node
    Set node in 32Hz and LDC mode
    Measure for 5 seconds
    Stop node
    Set node in 1024Hz and DL mode
    Measure for 11 seconds
    Stop node
    Download DL
    Stop node
    Sleep node

    Wait for 3 Hours.
    Wake node

So when I try to stop the nodes I found these errors "Failed toread EEPROM
108", and some times I got "Failed to write EEPROM 72".

Do you have any protocol about how to handle nodes correctly?

Thanks and best regards.

rwslord, 2016-06-16:
Does stopping the nodes (setting to idle) succeed correctly? It would be interesting to know what the light on the Node is doing when you get these errors. I'm wondering if the Node is failing to be stopped. If that were the case, you wouldn't be able to talk to it.

A feature that might be of use to you is the [number of sweeps / sampling duration](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#CClass:WirelessNodeConfig:numSweeps(2)). This allows you to set how long you want the Node to sample for. At the end of the set duration, the Node will go back to idle mode automatically. You will need to turn off the [Unlimited Duration](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#CClass:WirelessNodeConfig:unlimitedDuration(2)) flag as well, which I'm guessing is enabled at this time.

It might be worth upgrading to the latest version of MSCL if you haven't already. I released [4.3.1](https://github.com/LORD-MicroStrain/MSCL/releases/tag/v4.3.1) about a week ago.

All of our protocol information can be found here: https://github.com/LORD-MicroStrain/Protocols

crosvera, 2016-08-15:
@rwslord How do I calculate de number of sweeps? and does it works for all sampling modes (LDC, DL, Sync)?

Best regards.

rwslord, 2016-08-15:
Setting the number of sweeps works for all sampling modes.

I'm not sure what you mean by how to calculate the number of sweeps. This is the number of samples that will be collected. So if you are sampling at 1000Hz and you set the number of sweeps to 5000, it will sample for 5 seconds.

crosvera, 2016-11-08:
@rwslord Thanks for the advice, setting the number of sweeps works great, in both modes Data Logging and Low Duty Cycle.

Best regards.

closed 2016-11-08 by crosvera