LORD-MicroStrain/MSCL #14: Compiling MSCL under Cygwin

DouglasAlden, 2016-06-27:
I am trying to compile MSCL under 32-bit version of Cygwin.  The following packages are installed:

- p7zip
- gcc-g++
- python
- rsync
- git

The MSCL setup is as follows:

1. Downloaded  boost_1_61_0 source into /libraries/boost_1_61_0 and compiled it.

2. Edited my .bashrc file and added the following lines:
`export BOOST_ROOT=/usr/local/build/boost/`
`export BOOST_BUILD_PATH=/libraries/boost_1_61_0/`

3. Created a user-config.jam file with the following line:
`using gcc : 5.4.0 : /usr/bin/g++ : <cxxflags>-I/usr/include/python2.7 ;`

4. Cloned the MSCL repository onto my machine.

5. From the top of the MSCL directory structure ran
`./BuildScripts/buildAll.sh`
which returned the following:

`========================================================`
`(buildAll.sh) STARTING MSCL BUILD`
`(buildAll.sh) Creating the build\swig-python directory`
`(buildAll.sh) Running BJAM (C++)`
`...patience...`
`...found 989 targets...`
`...updating 218 targets...`
`gcc.compile.c++ /cygdrive/d/Software/MSCL_source/MSCL/build/MSCL/gcc-5.4.0/release/source/stdafx.o`
`In file included from /usr/local/include/boost/asio/async_result.hpp:18:0,`
`                 from /usr/local/include/boost/asio.hpp:20,`
`                 from MSCL/source/stdafx.h:13,`
`                 from MSCL/source/stdafx.cpp:6:`
`/usr/local/include/boost/asio/detail/config.hpp:566:5: warning: #warning Please define _WIN32_WINNT or _WIN32_WINDOWS appropriately. [-Wcpp]`
` #   warning Please define _WIN32_WINNT or _WIN32_WINDOWS appropriately.`
`     ^`
`/usr/local/include/boost/asio/detail/config.hpp:567:5: warning: #warning For example, add -D_WIN32_WINNT=0x0501 to the compiler command line. [-Wcpp]`
` #   warning For example, add -D_WIN32_WINNT=0x0501 to the compiler command line.`
`     ^`
`/usr/local/include/boost/asio/detail/config.hpp:568:5: warning: #warning Assuming _WIN32_WINNT=0x0501 (i.e. Windows XP target). [-Wcpp]`
` #   warning Assuming _WIN32_WINNT=0x0501 (i.e. Windows XP target).`
`     ^`
`/usr/local/include/boost/asio/detail/config.hpp:592:5: error: #error You must add -D__USE_W32_SOCKETS to your compiler options.`
` #   error You must add -D__USE_W32_SOCKETS to your compiler options.`
...

Is this a Boost compilation issue or a MSCL compilation issue?

Thanks,

Douglas

rwslord, 2016-06-28:
From the [Boost.Asio documentation](http://www.boost.org/doc/libs/1_61_0/doc/html/boost_asio/using.html) it looks like `__USE_W32_SOCKETS` must be defined when using Cygwin. (see the 3rd bullet point in the link).

DouglasAlden, 2016-06-28:
I updated the boost bjam compilation setup with
`define=_WIN32_WINNT=0x0601 define=__USE_W32_SOCKETS`
and added the defines above to each of the bjam lines in buildAll.sh but I am still having issues with ASIO.

`========================================================`
`(buildAll.sh) STARTING MSCL BUILD`
`(buildAll.sh) Creating the build\swig-python directory`
`(buildAll.sh) Running BJAM (C++)`
`...patience...`
`...found 989 targets...`
`...updating 218 targets...`
`gcc.compile.c++ /cygdrive/d/Software/MSCL_source/MSCL/build/MSCL/gcc-5.4.0/release/source/stdafx.o`
`In file included from /usr/local/include/boost/asio/detail/impl/posix_tss_ptr.ipp:24:0,`
`                 from /usr/local/include/boost/asio/detail/posix_tss_ptr.hpp:76,`
`                 from /usr/local/include/boost/asio/detail/tss_ptr.hpp:27,`
`                 from /usr/local/include/boost/asio/detail/call_stack.hpp:20,`
`                 from /usr/local/include/boost/asio/impl/handler_alloc_hook.ipp:19,`
`                 from /usr/local/include/boost/asio/handler_alloc_hook.hpp:80,`
`                 from /usr/local/include/boost/asio/detail/handler_alloc_helpers.hpp:21,`
`                 from /usr/local/include/boost/asio/detail/bind_handler.hpp:19,`
`                 from /usr/local/include/boost/asio/detail/wrapped_handler.hpp:18,`
`                 from /usr/local/include/boost/asio/io_service.hpp:24,`
`                 from /usr/local/include/boost/asio/basic_io_object.hpp:19,`
`                 from /usr/local/include/boost/asio/basic_socket.hpp:20,`
`                 from /usr/local/include/boost/asio/basic_datagram_socket.hpp:20,`
`                 from /usr/local/include/boost/asio.hpp:21,`
`                 from MSCL/source/stdafx.h:13,`
`                 from MSCL/source/stdafx.cpp:6:`
`/usr/local/include/boost/asio/error.hpp:87:31: error: ‘__MSABI_LONG’ was not declared in this scope`
`       BOOST_ASIO_NATIVE_ERROR(ERROR_BROKEN_PIPE),`
`                               ^`
`/usr/local/include/boost/asio/error.hpp:140:31: error: ‘__MSABI_LONG’ was not declared in this scope`
`       BOOST_ASIO_NATIVE_ERROR(ERROR_OUTOFMEMORY),`
`                               ^`
`/usr/local/include/boost/asio/error.hpp:145:31: error: ‘__MSABI_LONG’ was not declared in this scope`
`       BOOST_ASIO_NATIVE_ERROR(ERROR_ACCESS_DENIED),
                               ^`
`/usr/local/include/boost/asio/error.hpp:153:31: error: ‘__MSABI_LONG’ was not declared in this scope`
`       BOOST_ASIO_NATIVE_ERROR(ERROR_BAD_UNIT),`
`                               ^`
`/usr/local/include/boost/asio/error.hpp:164:31: error: ‘__MSABI_LONG’ was not declared in this scope`
`       BOOST_ASIO_NATIVE_ERROR(ERROR_OPERATION_ABORTED),`
`                               ^`
`/usr/local/include/boost/asio/error.hpp:178:31: error: ‘__MSABI_LONG’ was not declared in this scope`
`       BOOST_ASIO_NATIVE_ERROR(ERROR_RETRY),`
`                               ^`
...

rwslord, 2016-06-28:
Unfortunately, I don't have much experience with Cygwin. If you're on Windows, is there a reason you don't want to just use the Windows version of MSCL?

But from looking online, it seems like other people with the [same problem](http://stackoverflow.com/questions/34884499/boost-asio-example-does-not-build) uninstalled boost (which they had previously installed manually), and then re-ran the Cygwin installer and allowed its installer to install boost, instead of manually installing it.

DouglasAlden, 2016-06-28:
The Windows version of MSCL does not work with the Python package in Cygwin.  I found a workaround by installing the Windows version of Python and calling my Python script with this version, however, any console output is buffered between the Windows version of Python and Cygwin so it is kind of a kludge.

I started using Cygwin because I wanted linux features: ssh server, screen, etc. and the MSCL compiled drivers are only available under Windows.

I initially tried the option you listed, however, there is no Cygwin package for bjam so I started trying to compile Boost from source.

At this point I think my best option is to setup a development system with Linux see if I can get MSCL running from there.

closed 2016-06-28 by DouglasAlden