LORD-MicroStrain/MSCL #15: Error compiling under Debian Linux

DouglasAlden, 2016-09-22:
I am trying the compile the MSCL libraries on a Debian Linux system.  I am using the following versions of various programs:

gcc version 4.9.2 (Debian 4.9.2-10)
Boost.Jam Version 2011.12.1  OS=LINUX.

BJAM (C++) completes.
BJAM (Python) has errors.

Error file is attached.

Thanks,

Douglas
[buildAll_output.txt](https://github.com/LORD-MicroStrain/MSCL/files/488292/buildAll_output.txt)

DouglasAlden, 2016-09-22:
I should also add I am using:
Python 2.7.9 (default, Aug 13 2016, 16:41:35)

rwslord, 2016-09-23:
What version of SWIG are you using? This requires SWIG 3.0.6+. I believe the issue you are seeing might be caused by running SWIG 2.X.X

rwslord, 2016-09-23:
Also, you are getting a lot of errors from the Unit Test project because you don't have Turtle installed. You can probably just remove the building of that unit test project from the build script.

DouglasAlden, 2016-09-23:
That resolved the compilation issues.  You were correct that I was running SWIG 2.X.X.  The is the version that gets installed when using apt-get.  Compiling SWIG from source also require pcre.  For future reference, I ran the following commands:

wget http://prdownloads.sourceforge.net/swig/swig-3.0.10.tar.gz
tar -xzf swig-3.0.10.tar.gz
cd swig-3.0.10
wget ftp://ftp.csx.cam.ac.uk/pub/software/programming/pcre/pcre-8.39.tar.gz
chmod u+x ./Tools/pcre-build.sh
./Tools/pcre-build.sh
./configure
./make
sudo make install

Now that swig was installed I could continue compilation of the MSCL library

closed 2016-09-23 by DouglasAlden