LORD-MicroStrain/MSCL #16: Error when starting Synchronized Sampling

DouglasAlden, 2016-09-30:
I had to run my SynchronizedSampling.py script several times before the network started.  Here is the output from one of the failures:

Adding node 49607 to the network...
Done.
Adding node 49608 to the network...
Done.
Adding node 52431 to the network...
Done.
Adding node 52432 to the network...
Done.
Adding node 52433 to the network...
Done.
Adding node 52435 to the network...
Done.
Adding node 52437 to the network...
Done.
Adding node 52377 to the network...
Done.
Adding node 52378 to the network...
Done.
Adding node 52379 to the network...
<type 'exceptions.Exception'> Failed to read EEPROM 108 from Node 52379
Done.
Adding node 52380 to the network...
Done.
Adding node 52381 to the network...
Done.
Adding node 52384 to the network...
<type 'exceptions.Exception'> Failed to read EEPROM 108 from Node 52384
Done.
Adding node 52386 to the network...
Done.
Adding node 52389 to the network...
Done.
Adding node 52390 to the network...
Done.
Adding node 52391 to the network...
Done.
Adding node 52392 to the network...
Done.
Adding node 52393 to the network...
Done.
Adding node 52394 to the network...
Done.
Adding node 52395 to the network...
Done.
Adding node 52397 to the network...
Done.
Adding node 52398 to the network...
Done.
Adding node 52399 to the network...
Done.
Adding node 52400 to the network...
Done.
Adding node 52401 to the network...
Done.
Adding node 52402 to the network...
<type 'exceptions.Exception'> Failed to read EEPROM 108 from Node 52402
Done.
Adding node 52404 to the network...
Done.
Adding node 52405 to the network...
Done.
Adding node 52406 to the network...
Done.
Adding node 52408 to the network...
Done.
Adding node 47675 to the network...
Done.
Network info:
Network OK:  True
Percent of Bandwidth:  25.0244312286
Lossless Enabled:  False
High Capacity Mode:  False
Applying network configuration to network...
<type 'exceptions.Exception'> Failed to write EEPROM 250 to Node 52402
Error: Unable to create BaseStation global name 'sys' is not defined

[SynchronizedSampling.zip](https://github.com/LORD-MicroStrain/MSCL/files/503487/SynchronizedSampling.zip)

Thoughts?

Douglas

rwslord, 2016-09-30:
You have a lot of Nodes, so it's definitely possible that the wireless communication is failing in some cases. But it looks like you have retries in there to handle that.

From the output, it seems like your retries must have worked when attempting to add Nodes to the network and had communication issues: 

> Adding node 52384 to the network...
> Failed to read EEPROM 108 from Node 52384
> Done.

But then when it fails at the `network.applyConfiguration()` line, the retry seems to not work correctly? I haven't seen that error `Error: Unable to create BaseStation global name 'sys' is not defined` before. I don't see why it would be trying to create a BaseStation global name at that point? Is the retry actually happening (maybe print out the retry count)?

Do you always get that same error at the same place, which causes it to break? 

One thing to note is that it looks like your `if Exception == mscl.Error` logic isn't working...I don't see those error strings printed out to the console.

DouglasAlden, 2017-06-01:
Found the source of the error in retry.py.  I forgot to import sys and had a typo.
  e = sys.exc.info()[1]
should have been
  e = sys.exc_info()[1]

I simplified retry.py.
[retry.zip](https://github.com/LORD-MicroStrain/MSCL/files/1045282/retry.zip)

I am finding that I need to set the retry attempts to 40 to successfully get all the nodes to add to the network, and, apply the network configuration.

You noted above that the retries are associated with communication issues (e.g. Failed to read EEPROM 108 from Node 52384), which I catch as an unlisted Exception.  Is there an error wrapper for communications issues?

rwslord, 2017-06-01:
The Node communication errors will be thrown as a mscl.Error_NodeCommunication exception. All exceptions in MSCL inherit from mscl.Error, so if you wanted to write a generic catch, you can use that as well.

closed 2018-03-08 by rwslord