LORD-MicroStrain/MSCL #17: Python _mscl.pyd file

pgarzon, 2016-10-28:
Trying to do a python script to collect the data from the WSDA aggregator. The file _mscl.pyd only runs in windows. Is there an equivalent file for linux or a work around? when I tried to import the mscl.py file it gives an error the following error: ImportError: No module named _mscl

rwslord, 2016-10-28:
Unfortunately, at this time there are no pre-built binaries available for linux. However, MSCL can be built for Linux. The [Build Readme](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme.md#linux) can help, as well as issue #11.

pgarzon, 2016-10-28:
thanks

Pedro Garzon
Principal EDS Design Engineer
M +1 408 705 3790
www.morphodetection.com<http://www.morphodetection.com/>

pgarzon, 2016-10-28:
Thanks my problem now is that I can’t get the mscl.so file? I downloaded mscl and its not there,
Thanks

Pedro Garzon
Principal EDS Design Engineer
M +1 408 705 3790
www.morphodetection.com<http://www.morphodetection.com/>

closed 2016-10-28 by pgarzon

rwslord, 2016-10-31:
_mscl.so will get generated by the build, along with mscl.py