LORD-MicroStrain/MSCL #18: Getting hardware Gain and bits resolution

crosvera, 2016-11-09:
Reading the CHANGELOG, I've realised that getHardwareGain() is not longer available. So I'm wondering how can I get the hardware Gain from a specific channel? and also, there is any way to get the bit resolution?

Thanks, and best regards.

rwslord, 2016-11-09:
Yea, Hardware Gain has been changed out in favor of Input Range. So you will see a [getInputRange()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessNode.h:WirelessNode.getInputRange) function. 

There is no current way to get the bit resolution, although that has come up in the past.
I will add it to our internal backlog for MSCL. In the meantime, do you have specific questions about bit resolution for certain products that I might be able to answer for you?

crosvera, 2016-11-09:
I will take a look into getInputRange(). About the bit resolution, I'm
looking a way to get the range resolution in bits, so I can get the actual
offset-scale value.

offset-scale = bit_resolution * autobalance percentage

Thanks,
Best regards.

rwslord, 2016-11-09:
Are you looking for the resolution of the offset voltage? That would be 10-bits for all current products.
If you're looking for the resolution of the actual product, that can be found on the product data sheets on our website. Typically, this is 12, 16, or 18-bit.

crosvera, 2016-11-09:
Yeah I know that I can check the datasheet to get the product resolution,
just I thought it could be a great idea to get it from something like
wirelessNode().getBitResolution(), because sometimes I have to deal with
several nodes and not always are the same.

Thanks.

rwslord, 2016-11-09:
Definitely a good idea. I've created a ticket to do this, but there is some stuff ahead of it, so it won't be an immediate add.

crosvera, 2016-11-15:
@rwslord is there anyway to transform the `getInputRange()` result into hardware Gain?

For example from 2.5mV that I get as InputRange from NodeCommander (I get 2mV as InputRange in MSCL) how can I transform that into 569 (the gain value that is shown in NodeCommander).

Best Regards.

rwslord, 2016-11-15:
Unfortunately, there is no explicit way at this time in MSCL to convert to a hardware gain value.

crosvera, 2016-11-15:
So there is no way to set or get the hardware gain?

rwslord, 2016-11-15:
Not the hardware gain value itself...but just to be clear, setting the input range _is_ adjusting the hardware gain. It's the same setting, just a different way of looking at it.

crosvera, 2016-11-15:
Well, I don't know how to look it then. 

Now I tried to autobalance a SG-Link node, but  `node.getInputRange()` returns "Invalid Input Range".

From NodeCommander I see the following difference:

**SG-Link Gain configuration:**
[image]
**V-Link Gain configuration:**
[image]

As you can see the input is different, one is accepting the hardware gain and the other the Input Range.

How should I configure the SGLink gain (input range) value?

Best Regards.

rwslord, 2016-11-15:
calling `WirelessNode.features.inputRanges()` will give you a list of input ranges for the `ChannelGroup` that you pass in to the function.

Also, it might be worth using [SensorConnect ](http://microstrain.com/software/sensorconnect) instead of Node Commander for comparing to, as SensorConnect directly uses MSCL for everything.

crosvera, 2016-11-16:
@rwslord, I've compared the input range options that SensorConnect gives me with a SG-Link node, which are:
`14mV, 10mV, 7mV, 4mV, 2mV, 1.5mV, 1mV, 812uV`

I have autobalanced the the node with each of these input-ranges, then I compared what NodeCommander returns as gain values, which are respectively:

`104, 147, 198, 372, 750, 997, 1505, 1800`

Now, we used to use 503 as gain value in SG-Link nodes, but this configuration throws me an "Invalid Input Range". So I wondering, how can I set a custom gain value? or I'm just missing something?

According to the [MicroStrain LXRS Node EEPROM Map](http://files.microstrain.com/MicroStrain+LXRS+Node+EEPROM+Map.pdf) I can use the following equation to calculate the gain values to be setted via EEPROM commands:

[image]

In this equation appears the `Bit value` variable, but what refers to? Because trying to solving the equation (for sg-link nodes) using the gain values:
`104, 997, 147, 198` I obtain the following `Bit values`: `14.006, 137.035, 19,987, 27.010`

Any toughs?

Best regards.

rwslord, 2016-11-17:
Yes, some Nodes in the past allowed for custom hardware gain values. We've gone away from this model and now just allow an Input Range from a list of predefined values. Is there not a range/gain that gets close enough for you?

crosvera, 2016-11-17:
@rwslord and what about write directly into the EEPROM using the equation from above?

Thanks.

rwslord, 2016-11-17:
Yes, that should still work.

crosvera, 2016-11-17:
@rwslord In that case, what refers to `Bit value` in the equation? 

Best regards

rwslord, 2016-11-17:
That's a value between 0 and 255 which is what goes into the eeprom location.

crosvera, 2016-11-18:
@rwslord  Worked perfect with SG-Link!!

For V-Link, NodeCommander and SensorConnect, gives the following gain options: `21, 30, 75, 147, 291, 569, 1214, 2222, 3799, 13074`. Reading the Eeprom, those gain values correspond with the following bits values: `5, 76, 184, 219, 237, 246, 251, 253, 254, 255` respectively.

In the python console I ran the following snippet:
```python
for i in [5, 76, 184, 219, 237, 246, 251, 253, 254, 255]:
    print i, 10.0 - (1020000.0 / ((20.0 *i)-5151.0))
```
What I'm trying to do here is to check calculate the gain values from the bits obtained before, I used the equation shown before, these are the results:
```
bits gain
5    211.940209859
76   290.914348664
184  703.405846363
219  1332.95719844
237  2491.75182482
246  4425.58441558
251  7796.25954198
253  11218.7912088
254  14376.1971831
255  20010.0
```

As you can see, I'm not getting correct gain values. Do you know if the equation it is ok? If not, can you share the good one?

Thanks,
Best regards.

rwslord, 2016-11-21:
Can you tell me the model number of your V-Link? We've had a few different versions, so I want to make sure I'm giving you the correct information.

crosvera, 2016-11-21:
@rwslord the V-Link model number is: 1000-37416.

Best regards.

rwslord, 2016-11-21:
Ok, I'm guessing it is a 6312-1000? That first number tells me the most, but I believe that is what you have. Please correct me if not.

The formula that I actually have for that Node is:

```cpp
double HardwareGain::bitsToGain_vLink(uint16 bits)
{
    return 21.0 * (99609.375 / (160.0 + (( (255.0 - bits) / 256.0) * 100000.0)));
}
```

The EEPROM map that you are referencing is our legacy version. The new one can be found on our [Protocols](https://github.com/LORD-MicroStrain/Protocols) page, although the hardware gain formulas are no longer in it. I will add a request to get them added to that document.

rwslord, 2016-11-21:
Sorry, I should have clarified. The link you want on the Protocols page is [LORD MicroStrain Wireless Configuration](https://github.com/LORD-MicroStrain/Protocols/blob/master/Wireless/LORD%20MicroStrain%20Wireless%20Configuration.xlsx?raw=true)

crosvera, 2016-11-21:
@rwslord Thanks for the equation and yes it was that model number :dancer: . About the link, I knew the existence of that file, but it doesn't  contains any equation or value for the hardware gain configuration.

Best regards.

rwslord, 2016-11-21:
Gotcha. Yea, I will work on getting details of hardware gain / input range conversion publicly accessible.

crosvera, 2016-11-28:
@rwslord I'm doing some tests with SG-Link OEM nodes, and I'm wondering if you have another equation to calculate the Gain, like the one shown above, but exclusively for sg-link oem.

Best regards.

rwslord, 2016-11-28:
Yes, here is the conversion for the SG-Link-OEM:

```cpp
double HardwareGain::bitsToGain_sgLinkOem(uint16 bits)
{
    return std::pow(2.0, bits + 1) * 10.0;
}
```

rwslord, 2018-06-22:
Hey @crosvera. Just wanted to let you know that `NodeFeatures::inputRanges()` now returns a vector of `InputRangeEntry` objects. 

An `InputRangeEntry` has the `WirelessTypes::InputRange` enumeration, as well as the `gain` associated with that input range, if applicable.

crosvera, 2018-06-28:
@rwslord excellent news, I will try it!

crosvera, 2019-01-17:
@rwslord I'm trying to use `inputRanges()` method but I'm stuck. I'm using a SG-Link-200-OEM node:

```
features = node.features()
chngroups = features.channelGroups()
for c in chngroups:
    mask = c.channels()
    try:
        print(features.inputRanges(mask))
    except:
        print('error', mask.name())
```
I get the following:
```
('error', 'Differential (ch1)')
('error', 'Single-ended (ch2)')
('error', 'Pulse Frequency (ch5)')
('error', 'Total Pulses (ch6)')
('error', 'Differential (ch1)')
('error', 'Single-ended (ch2)')
('error', 'Pulse Input (ch5-ch6)')
```
What I'm doing wrong here?

By the way, as previously we talked, is there any convertion formula for this node model (sn: 6305-4200-85393)?

Best regards.

rwslord, 2019-01-17:
In your `except`, you should be able to catch the `mscl.Error` and print out the error message that you are getting. This might help you for further debugging.

It looks like there are a couple issues in your code that I can spot. The SG-Link-200-OEM actually requires you to use a different version of the `inputRanges()` function, because it supports configuration of excitation voltage. There is an overloaded version that you should be using which is documented [here](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Features/NodeFeatures.h:NodeFeatures.inputRanges(2)). Essentially, you have to pass the excitation voltage that you want to get the list of input ranges for, as a second parameter. You can check if you need to use this overloaded version of inputRanges by calling [`features.supportsExcitationVoltageConfig()`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Features/NodeFeatures.h:NodeFeatures.supportsExcitationVoltageConfig).

Another issue I see is that you are iterating over all the channel groups and assuming that each group supports the input range functionality. Before you call `features.inputRanges(mask)`, you should check `c.hasSetting(mscl.WirelessTypes.chSetting_inputRange)` is true.

Once you get this working, the `inputRanges()` function will return a container of [`InputRangeEntry`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Configuration/InputRange.h:InputRangeEntry.InputRangeEntry) objects, which should have `gain()` as an option.

mglord, 2021-06-09:
A bit late, but just wanted to add some new info in case anyone is looking for it -

The output bit resolutions for many of our current products are now available for lookup. The following properties were added to the `WirelessChannel` objects returned by `NodeFeatures::channels()`:
- `adcResolution()` - the bit resolution for the channel
- `adcMaxValue()` - the maximum uncalibrated value this channel is able to output

I'm a little unclear about what you were originally using this value for (offset-scale = bit_resolution * autobalance percentage) but if it was related to determining desired calibration values we have added a few new tools to SensorConnect to assist with this. I'll look into publishing the underlying formulas used in calculating these calibration values if that would be helpful. If I've misunderstood let me know and if you're interested in clarifying and I can look into making relevant info/formulas generally available!

Thanks!

closed 2024-10-24 by github-actions[bot]