LORD-MicroStrain/MSCL #19: Autobalance in MSCL differs with NodeCommander

crosvera, 2016-11-10:
I have been testing the `WirelessNode.autobalance()` function, and I realised that I'm having different results comparing with NodeCommander autobalance.

Specifically, trying to autobalance at mid-scale a channel without nothing connected (no-sensor). In node commander I got this:
[image]

As you can see, I'm getting a offset-scale of 65535 bits and the hardware offset is 177, also the percentage achieved is 9%.

Now, I made the same autobalance over the same channel via MSCL's  `WirelessNode().autobalance()`, and I got this:

`+ Percentage channel was balanced: 42.93`
`+ New offset-scale: 28134.0`
`+ Input Range (Gain): +- 2 mV`
`+ New hardware offset: 153`

the `errorCode` that I get is 1 or 2. But as you can see what I got is very different from what NodeCommander returns. And now I have some question about it:

1. How can the percentage vary from 9% to 42%, do you do some extra math?
2. The offset scale I calculated by `offset_scale = bits_resolution * percentageAchieved() / 100` is that ok? because I got 42% it looks like I have a good autobalance, despite the fact that I have no sensor connected.
3. From NodeCommander window, I can see that the hardware Gain is setted at `+- 2.5 mV`, but MSCL `WirelessType.InputRange` enum says it is `+- 2mV`. Is it just a typo in the code?
4. The hardware offset also differs, do you do some extra math?

Thanks
Best regards.

rwslord, 2016-11-11:
Can you tell the type of Node and the firmware version that it is using?

crosvera, 2016-11-14:
@rwslord The Node is a V-Link and its firmware version is 3.31.

rwslord, 2016-11-14:
The UI in Node Commander is showing you the resulting _hardware offset_ percentage after the autobalance. This is not the same percentage as the percentage you get from MSCL, which is how close to the autobalance came to balancing the channel (based on taking a single sample and comparing it to full scale). 

To get the hardware offset percent that Node Commander is displaying, you can take the hardwareOffset that MSCL gives you in the `AutoBalanceResult` and divide it by 1024

crosvera, 2016-11-15:
> To get the hardware offset percent that Node Commander is displaying, you can take the hardwareOffset that MSCL gives you in the AutoBalanceResult and divide it by 1024

Sorry @rwslord but I don't get it. If I divide the `hardwareOffset()` value that I get from `AutoBalanceResult` by `1024` I get:

`153 / 1024 = 0.1494`

rwslord, 2016-11-15:
Multiply that `0.1494` by 100 to get `14.9%`.
This percentage would be the same one that Node Commander displays in its Hardware Offset window.

Each time you run it (at least with a floating channel), you will probably get a different result, which is why it doesn't match Node Commander exactly. If you _are_ getting the same result every time, then there may be something else going on that I'm not seeing.

crosvera, 2016-11-15:
Thanks @rwslord that worked very well. And yes, I'm getting different results every run.

rwslord, 2016-11-15:
Glad that worked :)

closed 2016-11-15 by rwslord