LORD-MicroStrain/MSCL #20: Getting errors while building MSCL Libraries

Ansar006, 2017-01-31:
Hello,

I'm trying to build the package under Rasbian, I ran the "bash buildAll.sh" script but I get some errors and I don't know about them

pi@raspberrypi:~/MSCL-master $ sudo bash buildAll.sh
========================================================
(buildAll.sh) STARTING MSCL BUILD
(buildAll.sh) Creating the build\swig-python directory
(buildAll.sh) Running BJAM (C++)
warning: No toolsets are configured.
warning: Configuring default toolset "gcc".
warning: If the default is wrong, your build may not work correctly.
warning: Use the "toolset=xxxxx" option to override our guess.
warning: For more configuration options, please consult
warning: http://boost.org/boost-build2/doc/html/bbv2/advanced/configuration.html
...patience...
...found 1021 targets...
(buildAll.sh) Running BJAM (Python)
warning: No toolsets are configured.
warning: Configuring default toolset "gcc".
warning: If the default is wrong, your build may not work correctly.
warning: Use the "toolset=xxxxx" option to override our guess.
warning: For more configuration options, please consult
warning: http://boost.org/boost-build2/doc/html/bbv2/advanced/configuration.html
...patience...
...found 1109 targets...
...updating 3 targets...
gcc.compile.c++ /home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.o
/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp:173:21: fatal error: Python.h: No such file or directory
 # include <Python.h>
                     ^
compilation terminated.

    "g++"  -ftemplate-depth-128 -O3 -finline-functions -Wno-inline -Wall -fPIC -std=c++11 -DNDEBUG -DUNIX_BUILD  -I"MSCL/source" -I"MSCL/source/mscl/Wrapper" -c -o "/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.o" "/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp"

...failed gcc.compile.c++ /home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release>_mscl.so for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release>source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/MSCL-master/Output/Python>_mscl.so for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release>_mscl.so...
...failed updating 1 target...
...skipped 2 targets...
(buildAll.sh) Running BJAM (Python) (2)
warning: No toolsets are configured.
warning: Configuring default toolset "gcc".
warning: If the default is wrong, your build may not work correctly.
warning: Use the "toolset=xxxxx" option to override our guess.
warning: For more configuration options, please consult
warning: http://boost.org/boost-build2/doc/html/bbv2/advanced/configuration.html
...patience...
...found 1109 targets...
...updating 3 targets...
gcc.compile.c++ /home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.o
/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp:173:21: fatal error: Python.h: No such file or directory
 # include <Python.h>
                     ^
compilation terminated.

    "g++"  -ftemplate-depth-128 -O3 -finline-functions -Wno-inline -Wall -fPIC -std=c++11 -DNDEBUG -DUNIX_BUILD  -I"MSCL/source" -I"MSCL/source/mscl/Wrapper" -c -o "/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.o" "/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp"

...failed gcc.compile.c++ /home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release>_mscl.so for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release>source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/MSCL-master/Output/Python>_mscl.so for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-4.9.2/release>_mscl.so...
...failed updating 1 target...
...skipped 2 targets...

Can anyone guide me that how can I remove these errors. I have installed Swig and Boost latest versions on Rasbian but still I am failing to build MSCL libraries for Python.

crosvera, 2017-01-31:
> configuration.html

Hello there,

Seems you have to install the pyhon-dev package, because pyhon.h cannot be
found.
Also, in the closed issues there are some similars topics, you can check
for them.
https://github.com/LORD-MicroStrain/MSCL/issues?q=is%3Aissue+is%3Aclosed
https://github.com/LORD-MicroStrain/MSCL/issues/1

Best regards.

Ansar006, 2017-01-31:
Hello, 
I have installed all the dependencies including python-dev package on my system as written in readme file. The list of all dependencies installed on my system is given in the file attached.
Best Regards,Ansar Abdul Jabbar 

 

https://github.com/LORD-MicroStrain/MSCL/issues?q=is%3Aissue+is%3Aclosed
https://github.com/LORD-MicroStrain/MSCL/issues/1

rwslord, 2017-01-31:
I believe you may need to add a [site-config.jam or user-config.jam](http://www.boost.org/build/doc/html/bbv2/overview/configuration.html)

The MSCL documentation [shows this for the Windows Python build](http://www.boost.org/build/doc/html/bbv2/overview/configuration.html), but looks like it was missed for the Linux Python build. This is likely still required.

Can you try adding this and see if it solves the problem?

Ansar006, 2017-01-31:
I have added these two files  site-config.jam and user-config.jam in the root folder of MSCL. In user-conf.jam file I write using gcc; whereas in the other file I put path of boost-build but still I am getting the same error.

rwslord, 2017-01-31:
If you check out that first [link ](http://www.boost.org/build/doc/html/bbv2/overview/configuration.html)I added, it tells you where it needs the site-config.jam or user-config.jam files. You should only need to add one. site-config.jam is usually global for the system, where as user-config.jam is usually per user. Which one you use is up to your use case.

But anyway, the location of the files matters. Try putting the user-config.jam file in your `$HOME` path.
And try using this in the file:

```
using gcc
:	host
:	g++
:	<cxxflags>-I/usr/include/python2.7
;
```

Ansar006, 2017-02-01:
I put user-config.jam file in $HOME path but still I am getting the same errors in compiling. I have seen issue #3 because my problem is very similar to this and I try to apply these things discussed there but still I am getting the same errors.

pi@raspberrypi:~/MSCL-master $ sudo bash buildAll.sh
========================================================
(buildAll.sh) STARTING MSCL BUILD
(buildAll.sh) Creating the build\swig-python directory
(buildAll.sh) Running BJAM (C++)
...patience...
...found 1021 targets...
(buildAll.sh) Running BJAM (Python)
...patience...
...found 1027 targets...
...updating 4 targets...
swig.swig /home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp
swig error : Unrecognized option -outdir/home/pi/MSCL-master/build/swig-python
Use 'swig -help' for available options.

	swig -c++ -DUNIX_BUILD -outdir"/home/pi/MSCL-master/build/swig-python" -python -o "/home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp" "MSCL/source/mscl/Wrapper/MSCL_Main_Interface.i"

...failed swig.swig /home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp...
...skipped <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>source/mscl/Wrapper/MSCL_Main_Interface.o for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-host/release-object(c-scanner)@1197>source/mscl/Wrapper/MSCL_Main_Interface.cpp...
...skipped <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>_mscl.so for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/MSCL-master/Output/Python>_mscl.so for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>_mscl.so...
...failed updating 1 target...
...skipped 3 targets...
(buildAll.sh) Running BJAM (Python) (2)
...patience...
...found 1027 targets...
...updating 4 targets...
swig.swig /home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp
swig error : Unrecognized option -outdir/home/pi/MSCL-master/build/swig-python
Use 'swig -help' for available options.

	swig -c++ -DUNIX_BUILD -outdir"/home/pi/MSCL-master/build/swig-python" -python -o "/home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp" "MSCL/source/mscl/Wrapper/MSCL_Main_Interface.i"

...failed swig.swig /home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp...
...skipped <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>source/mscl/Wrapper/MSCL_Main_Interface.o for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-host/release-object(c-scanner)@1197>source/mscl/Wrapper/MSCL_Main_Interface.cpp...
...skipped <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>_mscl.so for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/MSCL-master/Output/Python>_mscl.so for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>_mscl.so...
...failed updating 1 target...
...skipped 3 targets...

Any suggestions to remove these errors ?

rwslord, 2017-02-01:
Ok, it looks like you are getting a different error now, so I think that solved your original one.

Can you check what version of SWIG you have? MSCL requires SWIG 3.0.6 or above. Other SWIG 3.X.X versions might work ok, but SWIG 2.X.X will not.

Ansar006, 2017-02-01:
I have installed SWIG version 3.0.12 which is the latest one I think so I don't think so that this error is due to swig old version.

rwslord, 2017-02-01:
Can you try putting a space between -outdir and the path? So like this:

`swig -c++ -DUNIX_BUILD -outdir "/home/pi/MSCL-master/build/swig-python" -python -o "/home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp" "MSCL/source/mscl/Wrapper/MSCL_Main_Interface.i"`

Ansar006, 2017-02-01:
after putting space now I am getting these errors

pi@raspberrypi:~/MSCL-master $ sudo bash buildAll.sh
========================================================
(buildAll.sh) STARTING MSCL BUILD
(buildAll.sh) Creating the build\swig-python directory
(buildAll.sh) Running BJAM (C++)
...patience...
...found 1021 targets...
(buildAll.sh) Running BJAM (Python)
...patience...
...found 1109 targets...
...updating 3 targets...
gcc.compile.c++ /home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.o
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.

    "g++"  -ftemplate-depth-128 -I/usr/include/python2.7 -O3 -finline-functions -Wno-inline -Wall -fPIC -std=c++11 -DNDEBUG -DUNIX_BUILD  -I"MSCL/source" -I"MSCL/source/mscl/Wrapper" -c -o "/home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.o" "/home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp"

...failed gcc.compile.c++ /home/pi/MSCL-master/build/MSCL/gcc-host/release/source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>_mscl.so for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/MSCL-master/Output/Python>_mscl.so for lack of <p/home/pi/MSCL-master/build/MSCL/gcc-host/release>_mscl.so...
...failed updating 1 target...
...skipped 2 targets...

rwslord, 2017-02-01:
From googling the `g++: internal compiler error: Killed (program cc1plus)` error, it looks like it may have ran out of memory when compiling? See this link: https://github.com/purinda/grooveshark-server/issues/4
It may be that you can add more swap space and solve the issue (if you are compiling on the Raspberry Pi itself). Another option is to look in to cross compiling, which would speed up the builds as well, but might be a lot of work.

Ansar006, 2017-02-06:
Yeah I think there was some problem with the cache memory so I add more swap space and now its working. Thanks for your help. I appreciate it

rwslord, 2017-02-06:
No problem! Glad you got it working.

closed 2017-02-06 by rwslord