LORD-MicroStrain/MSCL #21: getting angles from Inertial Node

crosvera, 2017-02-06:
I'm using MSCL with an Inertial node 3DM-GX3 -15-OEM, which contains an accelerometer and a gyroscope. From what I understood reading the [MIP Data Communications Protocol](http://files.microstrain.com/3DM-GX3-15,-25%20MIP%20Data%20Communications%20Protocol.pdf) to get the device angle I have to use the [InertialType's ChannelField](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20Documentation/index.html#File:MicroStrain/Inertial/InertialTypes.h:InertialTypes.ChannelField) CH_FIELD_SENSOR_DELTA_THETA_VEC. 

[Here](https://gist.github.com/crosvera/190f0e6b0b84e21793b3ef41f35afa95) I paste test code to get the accelerations and angles. But I get the following error:

```Traceback (most recent call last):
  File "testangle.py", line 23, in <module>
    node.setActiveChannelFields(mscl.InertialTypes.CATEGORY_SENSOR, chs)
  File "/home/pi/mscl/lin_mscl.py", line 2738, in setActiveChannelFields
    def setActiveChannelFields(self, *args): return _mscl.InertialNode_setActiveChannelFields(self, *args)
mscl.lin_mscl.Error_InertialCmdFailed: The Sensor Message Format command has failed.
```

I realize that this error disappears when I change the sample rate for the theta_vector channel to 64Hz. But in my application I to sample at 1Hz.

What I'm missing?

Best regards.

rwslord, 2017-02-06:
There is a setting on the device that controls the Orientation Calculation Rate. It is part of the **IMU/AHRS Signal Conditioning Settings (0x0C, 0x35)** command in the [Data Communications Protocol](http://files.microstrain.com/3DM-GX3-15,-25%20MIP%20Data%20Communications%20Protocol.pdf).

This calculation rate needs to also be set to the same 1Hz that you are attempting to set the Delta Theta vector to. Unfortunately, MSCL does not have support for that feature at this time. Do you have access to MIP Monitor? It may be easiest (if you aren't changing rates all the time in your software) to set this value in MIP Monitor, and then save the startup settings so that resetting the device keeps your setting. If you have MIP Monitor available, I can attach a guide on how to change this setting.

On a side note, I am surprised that 64Hz worked, as I believe that is not a supported sample rate for that channel...

crosvera, 2017-02-06:
Thanks for the soon response. Unfortunately I have not access to MIP Monitor. But trying to pass raw command via EEPROM writing should works right? Can you tell me which parameters/values should I use?

Best regards.

rwslord, 2017-02-06:
You should be able to send the raw command using the [Connection.write](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:Communication/Connection.h:Connection.write) function.

The bytes should look like this:
`75 65 0C 10 10 35 01 03 E8 00 03 0E 11 00 0A 00 0A 01 00 00 5E 46`

crosvera, 2017-02-06:
ok, I'm using the following to write these bytes:

```
b = bytearray("75650C1010350103E800030E11000A000A0100005E46")
connection = mscl.Connection.Serial(tty, 921600)
node = mscl.InertialNode(connection)
connection.write(b)
```
I got: `TypeError: in method 'Connection_write', argument 2 of type 'mscl::Bytes const &'`.

And I'm able to write these bytes using:

```
ba = "75 65 0C 10 10 35 01 03 E8 00 03 0E 11 00 0A 00 0A 01 00 00 5E 46"
connection = mscl.Connection.Serial(tty, 921600)
node = mscl.InertialNode(connection)
for b in ba.split(" "):
    connection.writeStr(b)
```

But then, when I try to set the delta_theta channel I got:
```
mscl.Error_InertialCmdFailed: The Sensor Message Format command has failed.
```

Best regards.

rwslord, 2017-02-06:
The Connect.write() method takes an mscl.Bytes object as a parameter.
Try making one like this:

```
b = mscl.Bytes([0x75, 0x65, 0x0C, 0x10, 0x10, 0x35, 0x01, 0x03, 0xE8, 0x00, 0x03, 0x0E, 0x11, 0x00, 0x0A, 0x00, 0x0A, 0x01, 0x00, 0x00, 0x5E, 0x46])

connection.write(b)
```

crosvera, 2017-02-06:
@rwslord Than you very much, that worked!

Best regards

closed 2017-02-06 by crosvera