LORD-MicroStrain/MSCL #22: Error create a WirelessNode

johanode, 2017-02-26:
Just started to work with MSCL. Running:

connection = mscl.Connection.Serial(COM_PORT)
baseStation = mscl.BaseStation(connection)
node = mscl.WirelessNode(NODE_ADDRESS, baseStation)

Setting the node, I only get the following:
Error: in method 'new_WirelessNode', argument 1 of type 'mscl::uint16'

Any advise would be grateful.

Thanks,
Johan

rwslord, 2017-02-27:
I'm assuming this is the Python version of MSCL you are using, correct? Can I see your whole code snippet?

johanode, 2017-02-27:
Yes, I'm using Python. Enclosed please find the code.

# MSCL Example

# import the mscl library
import sys
sys.path.append('C:\Python27\lib\mscl14')
import mscl

COM_PORT = 'COM3'
NODE_ADDRESS = 921600

try:
    # create a Serial Connection with the specified COM Port, default baud rate of 921600
    connection = mscl.Connection.Serial(COM_PORT)

    # create a BaseStation with the connection
    baseStation = mscl.BaseStation(connection)
    
    # create a WirelessNode with the BaseStation we created
    node = mscl.WirelessNode(NODE_ADDRESS, baseStation)

    print "Attempting to ping the Node..."

    # ping the Node
    response = node.ping()

    # if the ping response was a success
    if response.success():
        # get some details from the response
        print "Successfully pinged Node", NODE_ADDRESS
        print "Base Station RSSI:", response.baseRssi()
        print "Node RSSI:", response.nodeRssi()
    else:
        print "Failed to ping Node", NODE_ADDRESS

except Exception, e:
    print "Error:", e

rwslord, 2017-02-27:
It looks like you are using 921600 as the Node Address, which is not supported. That is likely a typo and should be representing the Baud Rate. If you change this value to the Node Address for your Wireless Node, I think you should be good.

johanode, 2017-02-27:
Thanks for the comment. As I understand now, connection = mscl.Connection.Serial(COM_PORT) sets up the serial connection using the baud rate of 921600. The node address is something else. I found it running the SensorConnect software. Now everything is working.

rwslord, 2017-02-27:
Yes, the `mscl.Connection.Serial(COM_PORT)` function uses a default value of 921600 for the baud rate. You can also specify this in its second parameter, like so: `mscl.Connection.Serial(COM_PORT, 115200)`

Glad you are up and running!

closed 2017-02-27 by rwslord