LORD-MicroStrain/MSCL #23: building mscl.py

johanode, 2017-03-15:
Problem building mscl.py, any suggestions?

$ bjam MSCL//stage_python_x86 release
warning: No toolsets are configured.
warning: Configuring default toolset "gcc".
warning: If the default is wrong, your build may not work correctly.
warning: Use the "toolset=xxxxx" option to override our guess.
warning: For more configuration options, please consult
warning: http://boost.org/boost-build2/doc/html/bbv2/advanced/configuration.html
...patience...
...found 1060 targets...
...updating 4 targets...
swig.swig /home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp
MSCL/source/mscl/Wrapper/../Bin.h:28: Error: Syntax error in input(3).

	swig -c++ -DUNIX_BUILD -modern -outdir "/home/pi/Downloads/MSCL-master/build/swig-python" -python -o "/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp" "MSCL/source/mscl/Wrapper/MSCL_Main_Interface.i"

...failed swig.swig /home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp...
...skipped <p/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release>source/mscl/Wrapper/MSCL_Main_Interface.o for lack of <p/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release-object(c-scanner)@1227>source/mscl/Wrapper/MSCL_Main_Interface.cpp...
...skipped <p/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release>_mscl.so for lack of <p/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release>source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/Downloads/MSCL-master/Output/Python/x86>_mscl.so for lack of <p/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release>_mscl.so...
...failed updating 1 target...
...skipped 3 targets...

rwslord, 2017-03-15:
Can you check the version of SWIG you are using? MSCL requires SWIG 3.0.6+

johanode, 2017-03-15:
Seems i use version 2, although I have installed version 3 
$ sudo apt-get install swig3.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
swig3.0 is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

$ swig -version

SWIG Version 2.0.12

Compiled with g++ [armv7l-unknown-linux-gnueabihf]

Configured options: +pcre

Please see http://www.swig.org for reporting bugs and further information

crosvera, 2017-03-15:
Uninstall swig 2:
$ sudo apt-get remove swig

Then create a symbolic link to swig3
ln -s /usr/bin/swig3.0 /usr/bin/swig

the check the version by:
$ swig -version

Best regards.

johanode, 2017-03-15:
Thanks a lot.
Now I got this far and mscl.py was created.

BR,
Johan

$ bjam MSCL//stage_python_x86 release
warning: No toolsets are configured.
warning: Configuring default toolset "gcc".
warning: If the default is wrong, your build may not work correctly.
warning: Use the "toolset=xxxxx" option to override our guess.
warning: For more configuration options, please consult
warning: http://boost.org/boost-build2/doc/html/bbv2/advanced/configuration.html
...patience...
...found 1143 targets...
...updating 3 targets...
gcc.compile.c++ /home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.o
/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp:154:21: fatal error: Python.h: No such file or directory
 # include <Python.h>
                     ^
compilation terminated.

    "g++"  -ftemplate-depth-128 -O3 -finline-functions -Wno-inline -Wall -fPIC -std=c++11 -DNDEBUG -DUNIX_BUILD  -I"MSCL/source" -I"MSCL/source/mscl/Wrapper" -c -o "/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.o" "/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.cpp"

...failed gcc.compile.c++ /home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release/source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release>_mscl.so for lack of <p/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release>source/mscl/Wrapper/MSCL_Main_Interface.o...
...skipped <p/home/pi/Downloads/MSCL-master/Output/Python/x86>_mscl.so for lack of <p/home/pi/Downloads/MSCL-master/build/MSCL/gcc-4.9.2/release>_mscl.so...
...failed updating 1 target...
...skipped 2 targets...

rwslord, 2017-03-15:
Make sure you have installed python, and python-dev.
Then you need to tell bjam where to find some things using site-config.jam or user-config.jam.

I just noticed that I didn't include the site-config.jam details for Linux in the build documentation. I've updated that, so go take a look: https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme.md#linux

johanode, 2017-03-19:
Thanks. mscl imported. 
After a while I maneged to understand that the port setting should be '/dev/ttyUSB2'. Everything seems allright, however when trying to adding the node to the baseStattion I get

Error: Failed to read EEPROM 124 from the BaseStation. 

Below please find the code.
# MSCL init
#   This code init Microstrain WSDA and nodes

# import the mscl library
import sys
sys.path.append('/home/pi/Downloads/mscl14')
import mscl

# Setup
COM_PORT = "/dev/ttyUSB2"
NODE_ADDRESS = 469

try:
    # create a SerialConnection with the specified COM Port, default baud rate of 921600
    connection = mscl.Connection.Serial(COM_PORT)

    # create a BaseStation with the connection
    baseStation = mscl.BaseStation(connection)

    # create a WirelessNode with the BaseStation we created
    node = mscl.WirelessNode(NODE_ADDRESS, baseStation)

    response = node.ping()
    if response.success()==False:
        status = node.setToIdle()    

    # create a WirelessNodeConfig which is used to set all node configuration options
    config = mscl.WirelessNodeConfig()

    # set the configuration options that we want to change
    #config.bootMode(mscl.WirelessTypes.bootMode_normal)
    config.inactivityTimeout(7200)
    config.samplingMode(mscl.WirelessTypes.samplingMode_sync)
    config.sampleRate(mscl.WirelessTypes.sampleRate_256Hz)
    config.unlimitedDuration(True)

    # apply the configuration to the Node
    node.applyConfig(config)
        
    # create a ArmedDataloggingNetwork object, giving it the BaseStation that
    # will be the master BaseStation for the network
    #network = mscl.ArmedDataloggingNetwork(baseStation)
    network = mscl.SyncSamplingNetwork(baseStation)
        
    # add a WirelessNode to the network
    #    Note: The Node must already be configured for Armed Datalogging before adding
    #    to the network, or else Error_InvalidConfig will be thrown.
    # TODO: Repeat this for all WirelessNodes that you want in the network
    print "Adding node to the network...",
    network.addNode(node)    
    print "Done."
          
    network.lossless(True)
    network.applyConfiguration()

    status = network.ok()
    if status:
        print "Network is ok"
    else:
        print "Network is not ok"
        
    
except Exception, e:
    print "Error:", e

johanode, 2017-03-19:
Problem solved.

closed 2017-03-19 by johanode

rwslord, 2017-03-20:
Glad everything worked out 👍