LORD-MicroStrain/MSCL #24: Visual Studio Pro 2015 Throws Error Unable to load DLL 'mscl'

tkreyche, 2017-04-12:
Trying to run your C# example code VS 2015 throws the error at mscl.Connection connection = mscl.Connection.Serial(COM_PORT); Same error happened when I built my own project.

The dlls are loaded in to the dependencies directory and the object and methods show up on the VS Object Browser and look correct, as are the Solution References.

I've build many projects with my own and 3rd party libraries and haven't seen this problem before. Typically if the references are not set up correctly you can't browse them and the solution will say they're missing etc.

System.TypeInitializationException was unhandled
  Message=The type initializer for 'mscl.msclPINVOKE' threw an exception.
  Source=MSCL_Managed
  TypeName=mscl.msclPINVOKE
  StackTrace:
       at mscl.msclPINVOKE.Connection_Serial__SWIG_1(String jarg1)
       at mscl.Connection.Serial(String port)
       at ParseData.Program.Main(String[] args) in C:\Users\tkreyche\Source\Repos\EVOX Microstrain Monitor\Examples\Wireless\CSharp\MSCL_Examples_CSharp\ParseData\ParseData.cs:line 27
       at System.AppDomain._nExecuteAssembly(Assembly assembly, String[] args)
       at System.AppDomain.ExecuteAssembly(String assemblyFile, Evidence assemblySecurity, String[] args)
       at Microsoft.VisualStudio.HostingProcess.HostProc.RunUsersAssembly()
       at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
       at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
       at System.Threading.ThreadHelper.ThreadStart()
  InnerException: 
       Message=The type initializer for 'SWIGExceptionHelper' threw an exception.
       Source=MSCL_Managed
       TypeName=SWIGExceptionHelper
       StackTrace:
            at mscl.msclPINVOKE.SWIGExceptionHelper..ctor()
            at mscl.msclPINVOKE..cctor()
       InnerException: 
            Message=Unable to load DLL 'mscl': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
            Source=MSCL_Managed
            TypeName=""
            StackTrace:
                 at mscl.msclPINVOKE.SWIGExceptionHelper.SWIGRegisterExceptionCallbacks_mscl(ExceptionDelegate applicationDelegate, ExceptionDelegate arithmeticDelegate, ExceptionDelegate divideByZeroDelegate, ExceptionDelegate indexOutOfRangeDelegate, ExceptionDelegate invalidCastDelegate, ExceptionDelegate invalidOperationDelegate, ExceptionDelegate ioDelegate, ExceptionDelegate nullReferenceDelegate, ExceptionDelegate outOfMemoryDelegate, ExceptionDelegate overflowDelegate, ExceptionDelegate systemExceptionDelegate)
                 at mscl.msclPINVOKE.SWIGExceptionHelper..cctor()
            InnerException:

rwslord, 2017-04-13:
Can you verify the following:
- `MSCL_Managed.dll` is added as a Reference to the C# project
- `MSCL.dll` is sitting next to the generated executable
- The Platform Target is not set to x86 or x64 (not AnyCPU).
- If the platform target is x86, you loaded the x86 MSCL dlls. If the platform target is x64, you loaded hte x64 MSCL dlls.

tkreyche, 2017-04-15:
I'm using your C# example Solution, and specifically the ParseData Project but the same error occurs on others.
It's running on a x86 machine.
The reference is set to the x86 version at C:\Users\tkreyche\Source\Repos\EVOX Microstrain Monitor\MSCL\Examples\dependencies\DotNet\x86\MSCL_Managed.dll
The Object Browser can see and browse the objects in the dll, as previously mentioned.
The same directory holds the x86 version of MSCL.dll

I don't get it....have used numerous other 3rd party libraries and haven't seen this type of problem. Next week I might try on a x64 machine. It looks like the managed version of MSCL is not loading the unmanaged code. Maybe a SWIG problem? I haven't used SWIG.

tkreyche, 2017-04-17:
I have the problem partially figured out. It has to do with the MSCL.dll not getting copied by the program into the appropriate directory. Not clear to me yet why it doesn't work when setting the reference to dependencies directory.

rwslord, 2017-04-17:
Strange. I'm not a C# expert, so I'm not sure why that isn't working either. But it sounds like you have found the root of the issue. I'm not sure that is an MSCL issue, but if you feel that it is or would like me to consider changing the example project in someway, please let me know.

tkreyche, 2017-04-23:
It's a little unusual in this environment to have a managed DLL with a dependent unmanaged DLL. Or at least for me anyway. This is not a problem but it may be necessary to take any extra step to copy the unmanaged DLL into the output directory since it won't happen automatically by default. I admit I haven't fully looked into the best configuration for this type of project, nor have I worked with SWIG, so this is just a quick and easy work around.

closed 2017-04-23 by tkreyche

rwslord, 2017-04-24:
Gotcha. I _think_ it's just the nature of the beast with how the project works through SWIG. Glad you got it working though.