LORD-MicroStrain/MSCL #25: Separating triggers during armed datalogging

skurwa, 2017-06-14:
Hi! When downloading data, multiple triggers are being grouped in the output from the DatalogDownloader object. How can I separate out each trigger into individual sessions?

Here's my code:

`# SET ARMED DATALOGGING CONFIGURATION

config = mscl.WirelessNodeConfig()
config.samplingMode(mscl.WirelessTypes.samplingMode_armedDatalog)
config.inactivityTimeout(7200)
config.sampleRate(mscl.WirelessTypes.sampleRate_1024Hz)
config.unlimitedDuration(False)
config.numSweeps(2100)

network = mscl.ArmedDataloggingNetwork(basestation)

for node in nodes:
    node.applyConfig(config)
    network.addNode(node)

# START DATALOGGING

network.startSampling()

# SET NODES TO IDLE

for node in nodes:
    status = node.setToIdle()
    while not status.complete(500):
        print('.')
    result = status.result()
    if result == mscl.SetToIdleStatus.setToIdleResult_success:
    # completed successfully
     print('Successfully idled Node', node.name())
    elif result == mscl.SetToIdleStatus.setToIdleResult_canceled:
    # canceled by the user
     print('Idle command canceled by user for Node', node.name())
    else:
    # Failed to perform the operation
     print('Failed to idle Node', node.name())

# DOWNLOAD DATA

for node in nodes:
    dwn = mscl.DatalogDownloader(node)
    name = node.name()
    while not dwn.complete():
        sweep = dwn.getNextData()
        timestamp = sweep.timestamp()
        data = [p.as_float() for p in sweep.data()]
        node_list[nodes.index(node)].append([name, timestamp, data[0], data[1], data[2]])`

rwslord, 2017-06-14:
after your `sweep = dwn.getNextData()` line, you can add a `dwn.startOfSession()` call. This should return true for each session that you performed. This means all of the data in your `sweep` object was collected in a new session. It will then return false until the end of that session.

skurwa, 2017-06-14:
Got it. Thanks!

closed 2017-06-15 by rwslord