LORD-MicroStrain/MSCL #26: Making Sense of Acquired Data

skurwa, 2017-06-16:
Here's my code:

```python
# SET ARMED DATALOGGING CONFIGURATION

config = mscl.WirelessNodeConfig()

# create an ArmedDataloggingNetwork object, giving it the BaseStation that will be the master BaseStation for the network
network = mscl.ArmedDataloggingNetwork(basestation)

# set the configuration options that we want to change
config.samplingMode(mscl.WirelessTypes.samplingMode_armedDatalog)
config.inactivityTimeout(7200)
config.sampleRate(mscl.WirelessTypes.sampleRate_1024Hz)
config.dataFormat(mscl.WirelessTypes.dataFormat_cal_float)
config.unlimitedDuration(False)
config.numSweeps(2100)
for node in nodes:
    chgroups = node.features().channelGroups()
    for group in chgroups:
        config.unit(group.channels(), mscl.WirelessTypes.unit_accel_g)
        print(node.getUnit(group.channels()))
    node.applyConfig(config)
    network.addNode(node)    
```
Output:

    4
    4
    4
    4

Therefore, the channels must be in units of G's.

```python
# SEE DATA

for node in nodes:
    dwn = mscl.DatalogDownloader(node)
    print(dwn.calsApplied())
    na = node.nodeAddress()
    while not dwn.complete():
        sweep = dwn.getNextData()
        ts = sweep.timestamp()
        tick = sweep.tick()
        pnts = [p.as_float() for p in sweep.data()]
        print(tick, ts, pnts)
```

Output snippet:

    True
    0 2017-06-16 16:34:35.001298340 [1929.0, 1822.0, 1373.0]
    1 2017-06-16 16:34:35.002274902 [1930.0, 1818.0, 1373.0]
    2 2017-06-16 16:34:35.003251464 [1933.0, 1822.0, 1377.0]
    3 2017-06-16 16:34:35.004228026 [1931.0, 1818.0, 1374.0]
    4 2017-06-16 16:34:35.005204588 [1932.0, 1824.0, 1377.0]
    5 2017-06-16 16:34:35.006181150 [1929.0, 1820.0, 1376.0]

Therefore, with calibration coefficients applied and units of Gs, the data points in each channel are on the order of magnitude of 1000 which doesn't make a whole lot of sense. The nodes are placed on a stationary table when I took this data, so there shouldn't be much acceleration at all. I tried collecting data using the Node Commander software and found results that were more reasonable. Is there some sort of post-processing on the data that I'm neglecting?

rwslord, 2017-06-16:
So the unit type doesn't actually do much except get treated as a label for your use. The real magic happens from the slope and offset that are applied to the Node for each channel. You can see what's applied using
```
eq = node.getLinearEquation( group.channels() )
print(eq.slope())
print(eq.offset())
```
However, since you are going in to Node Commander (which, by the way, has been replaced by [SensorConnect ](http://www.microstrain.com/software/sensorconnect), in case that's of interest) and seeing the results as you would expect, I'm guessing the slopes and offsets _are_ actually set correctly.

My guess is that `dwn.calsApplied()` might actually be false, meaning you need to apply the cal coeffiecients (which can be retrieved using `dwn.calCoefficients`) to the data as it is being downloaded. The `dwn.calsApplied()` can change after calling `dwn.getNextData()` whenever `dwn.startOfSession()` returns true. Can you print out the result of calsApplied each time and see if it returns false?

skurwa, 2017-06-16:
Nice catch. It does return false. Is there a method to apply cal coefficients or do I need to apply the slope and offset to each channel by using the `node.getLinearEquation(group.channels())` as you described above?

I'll take a look into SensorConnect. I can confirm that you're right about the slopes and offsets in Node Commander though.

rwslord, 2017-06-20:
You will get the slope and offset for each channel using the method in your last post, then apply it like so:

value = slope * bitValue + offset

rwslord, 2017-06-20:
Actually, sorry, I misspoke a bit in my last post. You can get the cal coefficients from the node like that as long as you never change them. But they are also available from the DatalogDownloader object itself: `dwn.calCoefficients` which will be specific for the each session in case they changed.

rwslord, 2017-06-21:
No problem, glad you are up and running!

closed 2017-06-21 by rwslord

skurwa, 2017-07-11:
Had to step away from this project for awhile, but I'm not sure I understand how to use `dwn.calCoefficients`. How do you get the slope and offset from the ChannelCalMap object?

rwslord, 2017-07-11:
You should be able to iterate over the ChannelCalMap (its a map of `Wirelesschannel::ChannelId`s to `CalCoefficients`. 

Then on each `CalCoefficients` object, you can call `cal.unit()`, `cal.linearEq().slope()`, `cal.linearEq().offset()`

skurwa, 2017-07-11:
Got it. Ran the following script after generating some data:

```
for node in nodes:
    dwn = mscl.DatalogDownloader(node)
    test = dwn.calCoefficients()
    print(test)
    print(test.asdict())
    for cal in test:
        print(cal)
        print(cal.linearEq().slope())
```

My output:

```
<mscl.ChannelCalMap; proxy of <Swig Object of type 'std::map< mscl::WirelessChannel::ChannelId,mscl::CalCoefficients > *' at 0x000000000529DBD0> >
{}
<mscl.ChannelCalMap; proxy of <Swig Object of type 'std::map< mscl::WirelessChannel::ChannelId,mscl::CalCoefficients > *' at 0x000000000529DCF0> >
{}
```

Note that there are 2 wireless nodes. In the output, the CalCoefficients and the slopes are not printing - probably because the ChannelCalMap is empty (when printed out as a dictionary). Any reason why this might be the case?

rwslord, 2017-07-11:
The `DatalogDownloader` object has a function called `startOfSession()` which is true when a sweep is the start of a session (which indicates cal coefficients have potentially changed), and false for all other sweeps.

So what you want to do is call `dwn.getNextData()` which will give you a `LoggedDataSweep`. After calling getNextData() each time, you will want to then call `dwn.startOfSession()`. If this returns true, you then want to update your calibration coefficients by calling `dwn.calCoefficients()`.

skurwa, 2017-07-11:
Makes sense. I moved it to run if at the start of session, but now cal is a Swig Object of type 'unknown' rather than a CalCoefficients object, so it does not have the unit(), linearEquation(), etc attributes.

```
for node in nodes:
    dwn = mscl.DatalogDownloader(node)      
    
    name = node.name()
    sweep = dwn.getNextData()
    pnts = [p.as_float() for p in sweep.data()]
    while not dwn.complete():
        if dwn.startOfSession():
            test = dwn.calCoefficients()
            print(test)
            for cal in test:
                print(cal)
                print(type(cal))
                print(cal.unit())
```

Output:
```
<mscl.ChannelCalMap; proxy of <Swig Object of type 'std::map< mscl::WirelessChannel::ChannelId,mscl::CalCoefficients > *' at 0x0000000001F54300> >
<Swig Object of type 'unknown' at 0x000000000529DCF0>
<class 'SwigPyObject'>
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-173-04f533b9a83e> in <module>()
     15                 print(type(cal))
---> 16                 print(cal.unit())
     17             startTime = datetime.datetime.strptime(str(sweep.timestamp())[:-3], "%Y-%m-%d %H:%M:%S.%f")

AttributeError: 'SwigPyObject' object has no attribute 'unit'

The above exception was the direct cause of the following exception:

SystemError                               Traceback (most recent call last)
SystemError: <built-in function delete_SwigPyIterator> returned a result with an error set
```

rwslord, 2017-07-12:
[`ChannelCalMap`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/CalCoefficients.h:CalCoefficients.ChannelCalMap) is actually a map (dictionary in Python) where the keys are [`WirelessChannel.ChannelId`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessChannel.h:WirelessChannel.ChannelId)s and the values are the [`CalCoefficients`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/CalCoefficients.h:CalCoefficients.CalCoefficients) for that channel.

Can you try using it like a dictionary?

skurwa, 2017-07-12:
Whoops, yep. Getting the slopes and offsets now. Thanks!

rwslord, 2017-07-12:
Perfect! Happy to help!