LORD-MicroStrain/MSCL #27: trouble getting datapoint timestamp

rielt12, 2017-06-27:
Hello,

I am trying to get the timestamp of a data point but it is throwing an exception. I declared t_meas as a Timestamp object but I get the Error_BadDataType exception.
`const mscl::Timestamp t_meas = dataPoint.as_Timestamp();`

Am I missing something?

rwslord, 2017-06-27:
Can you show me where `dataPoint` comes from? Are you writing code for our Wireless devices or our Inertial devices?

rielt12, 2017-06-27:
I am writing code for inertial devices and I am getting datapoint from here
http://lordmicrostrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#CClass:DataPoint

rwslord, 2017-06-27:
Ok, so I'm guessing you are calling `InertialNode.getDataPackets()` which gives you a container of [`InertialDataPacket`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Inertial/Packets/InertialDataPacket.h:InertialDataPacket) objects?

On each `InertialDataPacket`, there is a `data` function which gets you all the points in the packet, as well as a `timestamp` function which gives you the [`Timestamp`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:Timestamp.h:Timestamp) of all the points that are in that packet.

One thing to note, this Timestamp is added by MSCL when it is parsed. So this isn't the same as the GPS Time that can be enabled to be sent as data from the device.

rielt12, 2017-06-27:
Is think this is what you mean:
` mscl::InertialDataPackets packets = node.getDataPackets(500);

      for(mscl::InertialDataPacket packet : packets)
      {
        

        //get the data in the packet
        mscl::InertialDataPoints data = packet.data();
        mscl::InertialDataPoint dataPoint;
		
		//get the time stamp_info
		mscl::Timestamp t_meas = packet.timestamp();
		std:: cout << t_meas.seconds() << std:: endl;

        //loop through all the data points in the packet
        for(unsigned int itr = 0; itr < data.size(); itr++)
        {
          dataPoint = data[itr];

          //print out the channel data
          
		  
          cout << dataPoint.as_float() << std::endl; //Just printing this out as a string. Other methods (ie. as_float, as_uint16, as_Vector) are also available.
          IMU_data.push_back((double)dataPoint.as_float());
		  
          //if the dataPoint is invalid
          if(!dataPoint.valid())
          {
            cout << "[Invalid] ";
          }
        }
        cout << endl;
	
        }
		`

rwslord, 2017-06-27:
right, so that gives you the timestamp for the data points like you wanted, right?

rielt12, 2017-06-27:
Yes thank you that solved it. Just to confirm though: in each packet there is one data point and one timestamp correct? Assuming that everything on the channel is read on the same frequency.

rwslord, 2017-06-27:
Actually, in each packet, there is 1 timestamp, and 1-many data points. For example, a packet might have roll, pitch, yaw, accel-x, accel-y, and accel-x all in it. If all of those were in the same packet, they would each have the same timestamp.

closed 2017-06-29 by rwslord