LORD-MicroStrain/MSCL #30: Error: InertialChannel (33286) is not in the Sensor descriptor set

rielt12, 2017-08-16:
Hello,

When i do:

`chs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_ESTFILTER_ESTIMATED_GYRO_BIAS, mscl::SampleRate::Hertz(1/intvl))); chs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_ESTFILTER_ESTIMATED_ACCEL_BIAS, mscl::SampleRate::Hertz(1/intvl)));`
I get `Error: InertialChannel (33286) is not in the Sensor descriptor set`
Is there a way to fix this?
Thanks.

rwslord, 2017-08-16:
I think you should only get that error when calling `InertialNode.setActiveChannelFields()`, is that when you see the error?

If so, that command takes an `InertialTypes::InertialCategory` as its first parameter, which I think you are passing in `CATEGORY_SENSOR` when you should be passing in `CATEGORY_ESTFILTER` for those channels which you set (note the `ESTFILTER` part of the `CH_FIELD_ESTFILTER_XXXXXXXX`)

rielt12, 2017-08-16:
Ok, I see, So is it possible to have a node have an active channel be a CATEGORY_SENSOR and a CATEGORY_ESTFILTER at the same time?

rwslord, 2017-08-16:
Yes, that is exactly how you do it, where `chs` would only have the Sensor channels for the first line, and only have the Estimation Filter channels for the second line.

rielt12, 2017-08-17:
Hmm So I do:

```
mscl::InertialNode node(connection);
		

	mscl::InertialChannels chs;

    chs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_SENSOR_SCALED_ACCEL_VEC, mscl::SampleRate::Hertz(1/intvl)));
    chs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_SENSOR_SCALED_GYRO_VEC, mscl::SampleRate::Hertz(1/intvl)));
	chs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_SENSOR_SCALED_MAG_VEC, mscl::SampleRate::Hertz(1/intvl)));
 	chs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_ESTFILTER_ESTIMATED_GYRO_BIAS, mscl::SampleRate::Hertz(1/intvl)));
	chs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_ESTFILTER_ESTIMATED_ACCEL_BIAS, mscl::SampleRate::Hertz(1/intvl)));

 
	

   //set the active channels for the Sensor category on the Node
    node.setActiveChannelFields(mscl::InertialTypes::CATEGORY_SENSOR, chs);
	node.setActiveChannelFields(mscl::InertialTypes::CATEGORY_ESTFILTER, chs);

    
	node.enableDataStream(mscl::InertialTypes::CATEGORY_SENSOR);
	node.enableDataStream(mscl::InertialTypes::CATEGORY_ESTFILTER);
```
And I still get the error:
`Error: InertialChannel (33286) is not in the Sensor descriptor set
`

rwslord, 2017-08-17:
You need to pass 2 separate containers like this:

```
mscl::InertialChannels sensorChs;
mscl::InertialChannels estChs;

sensorChs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_SENSOR_SCALED_ACCEL_VEC, mscl::SampleRate::Hertz(1/intvl)));
sensorChs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_SENSOR_SCALED_GYRO_VEC, mscl::SampleRate::Hertz(1/intvl)));
sensorChs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_SENSOR_SCALED_MAG_VEC, mscl::SampleRate::Hertz(1/intvl)));

estChs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_ESTFILTER_ESTIMATED_GYRO_BIAS, mscl::SampleRate::Hertz(1/intvl)));
estChs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_ESTFILTER_ESTIMATED_ACCEL_BIAS, mscl::SampleRate::Hertz(1/intvl)));

node.setActiveChannelFields(mscl::InertialTypes::CATEGORY_SENSOR, sensorChs);
node.setActiveChannelFields(mscl::InertialTypes::CATEGORY_ESTFILTER, estChs);

node.enableDataStream(mscl::InertialTypes::CATEGORY_SENSOR);
node.enableDataStream(mscl::InertialTypes::CATEGORY_ESTFILTER);
```

rielt12, 2017-08-17:
with that I get:
`Error: The Estimation Filter Message Format command has failed.`

rwslord, 2017-08-17:
Hmm...do you get that every time? What are the actual sample rates that you are passing in?

rielt12, 2017-08-17:
I am sending in 100hz

rielt12, 2017-08-17:
This is the whole piece of code
`#include <iostream>
#include <iomanip>

#include "mscl/Communication/Connection.h"
#include "mscl/MicroStrain/Inertial/InertialNode.h"
#include "mscl/Exceptions.h"
#include "Interfiles.h"

using namespace std;

class IMU_read {

public:

int IMU_reader(double intvl)
{
	IMU_meas imu_meas;
	std::vector<float> IMU_data;
	
	try{

	//create a SerialConnection with the COM port
	mscl::Connection connection = mscl::Connection::Serial("/dev/serial/by-id/usb-Lord_Microstrain_Lord_Inertial_Sensor_0000__6234.55966-if00", 9600);
    

   //create an InertialNode with the connection
	mscl::InertialNode node(connection);
		

	mscl::InertialChannels sensorChs;
    mscl::InertialChannels estChs;

    sensorChs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_SENSOR_SCALED_ACCEL_VEC, mscl::SampleRate::Hertz(1/intvl)));
    sensorChs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_SENSOR_SCALED_GYRO_VEC, mscl::SampleRate::Hertz(1/intvl)));
    sensorChs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_SENSOR_SCALED_MAG_VEC, mscl::SampleRate::Hertz(1/intvl)));

    estChs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_ESTFILTER_ESTIMATED_GYRO_BIAS, mscl::SampleRate::Hertz(1/intvl)));
    estChs.push_back(mscl::InertialChannel(mscl::InertialTypes::CH_FIELD_ESTFILTER_ESTIMATED_ACCEL_BIAS, mscl::SampleRate::Hertz(1/intvl)));

    node.setActiveChannelFields(mscl::InertialTypes::CATEGORY_SENSOR, sensorChs);
    node.setActiveChannelFields(mscl::InertialTypes::CATEGORY_ESTFILTER, estChs);

     node.enableDataStream(mscl::InertialTypes::CATEGORY_SENSOR);
     node.enableDataStream(mscl::InertialTypes::CATEGORY_ESTFILTER);

		//endless loop of reading in data
		while(true)
		{
      //get all the data packets from the node, with a timeout of 500 milliseconds
      mscl::InertialDataPackets packets = node.getDataPackets(500);

      for(mscl::InertialDataPacket packet : packets)
      {
        
		
		// each packet has a timestamp and accel_xyz and gyro_xyz
		
		mscl::Timestamp t_ts = packet.timestamp();
		struct timespec t_meas;
		t_meas.tv_sec = 0;
		t_meas.tv_nsec =t_ts.nanoseconds();
		imu_meas.t_meas = t_meas;

        //get the data points in the packet
        mscl::InertialDataPoints data = packet.data();
        mscl::InertialDataPoint dataPoint;
		
		

        //loop through all the data points in the packet
        for(unsigned int itr = 0; itr < data.size(); itr++)
        {
          dataPoint = data[itr];

          //print out the channel data
          /*
		  mutex_fd.lock();
		  if (itr == 9 )
	      myfile<<"roll: " << dataPoint.as_float() << std::endl; //Just printing this out as a string. Other methods (ie. as_float, as_uint16, as_Vector) are also available.
		  if (itr ==10 )
 	      myfile<<"pitch: " << dataPoint.as_float() << std::endl; //Just printing this out as a string. Other methods (ie. as_float, as_uint16, as_Vector) are also available.
	      if (itr ==11)
		  myfile<<"yaw: " << dataPoint.as_float() << std::endl; //Just printing this out as a string. Other methods (ie. as_float, as_uint16, as_Vector) are also available.
	      mutex_fd.unlock();
          */
     	  //myfile << dataPoint.as_float() << std::endl;

		  
		IMU_data.push_back(dataPoint.as_float());

          //if the dataPoint is invalid
          if(!dataPoint.valid())
          {
            cout << "[Invalid] ";
          }
        }
        //cout << endl;
		
		imu_meas.accel_x = IMU_data.at(0)* 9.80665; // convert to m/s^2
		imu_meas.accel_y = IMU_data.at(1)* 9.80665; // convert to m/s^2
		imu_meas.accel_z = IMU_data.at(2)* 9.80665; // convert to m/s^2
		imu_meas.w_x = IMU_data.at(3);
		imu_meas.w_y = IMU_data.at(4);
		imu_meas.w_z = IMU_data.at(5);
		imu_meas.mag_x = IMU_data.at(6);
		imu_meas.mag_y = IMU_data.at(7);
		imu_meas.mag_z = IMU_data.at(8);
		imu_meas.bias_gyro_x = IMU_data.at(9);
		imu_meas.bias_gyro_y = IMU_data.at(10);
		imu_meas.bias_gyro_z = IMU_data.at(11);
		imu_meas.bias_accel_x = IMU_data.at(12)* 9.80665; // convert to m/s^2;
		imu_meas.bias_accel_y = IMU_data.at(13)* 9.80665; // convert to m/s^2;
		imu_meas.bias_accel_z = IMU_data.at(14)* 9.80665; // convert to m/s^2;
		
		

		

		
		/*
		std:: cout <<"accelx= " <<imu_meas.accel_x << std:: endl;
		std:: cout<<"accely= " << imu_meas.accel_y << std:: endl;
		std:: cout<<"accelz= " << imu_meas.accel_z << std:: endl;
		std:: cout<<"w_x= " << imu_meas.w_x << std:: endl;
		std:: cout<<"w_y= " << imu_meas.w_y << std:: endl;
		std:: cout<<"w_z= " << imu_meas.w_z << std:: endl;
		std:: cout<<"mag_x= " << imu_meas.mag_x << std:: endl;
		std:: cout<<"mag_y= " << imu_meas.mag_y << std:: endl;
	    std:: cout<<"mag_z= " << imu_meas.mag_x << std:: endl;

		*/

		

		
		
		IMU_data.clear();
		
		//we push into the queue
		
		mutex_IMU_meas.lock();
		//IMU_meas_q->push_front(imu_meas);
		IMU_meas_q ->emplace_front(imu_meas);
		mutex_IMU_meas.unlock();
		
	
        }
		
		}
}
catch(mscl::Error& e)
	{
		cout << "Error: " << e.what() << endl;
	}

}

};`

rwslord, 2017-08-17:
What device do you have? Do you get that error every time?

rielt12, 2017-08-17:
I have a 3DM-GX4-25. Yah I get that error every time I run the code.

rwslord, 2017-08-17:
Hmm...everything seems ok.

Can you catch the `Error_InertialCmdFailed` exception and print out exception.code(). 

The error codes are as follows:
MIP_ACK_NACK_ERROR_NONE                                         = 0,
MIP_ACK_NACK_ERROR_UNKNOWN_COMMAND            = 1,
MIP_ACK_NACK_ERROR_CHECKSUM_INVALID            = 2,
MIP_ACK_NACK_ERROR_PARAMETER_INVALID        = 3,
MIP_ACK_NACK_ERROR_COMMAND_FAILED            = 4,
MIP_ACK_NACK_ERROR_COMMAND_TIMEOUT            = 5,
MIP_ACK_NACK_ERROR_UNKNOWN_DESCRIPTOR_SET    = 6

rielt12, 2017-08-17:
Do you mean:
```
catch(mscl::Error& e)
	{
		cout << "Error: " << e.code() << endl;
	}

```
Because that doesn't work.

Also how would I create the` Error_InertialCmdFailed` exception?

rwslord, 2017-08-17:
catch(mscl::Error_InertialCmdFailed& e)
	{
		cout << "Error: " << e.code() << endl;
	}

rielt12, 2017-08-17:
It ouputs:

Error: 4

rwslord, 2017-08-17:
Can you try changing your baud rate from 9600 to 115200 or 921600 in your `Connection::Serial` code?

rielt12, 2017-08-17:
I tried and I still get the `Error: The Estimation Filter Message Format command has failed.` for both baudrates

rwslord, 2017-08-17:
The 3DM-GX4-25 does not support the `CH_FIELD_ESTFILTER_ESTIMATED_ACCEL_BIAS` field. 

You should be able to check what is supported by using the [`node.features().supportedChannelFields`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Inertial/Features/InertialNodeFeatures.h:InertialNodeFeatures.supportedChannelFields) command if you want to do it via code.