LORD-MicroStrain/MSCL #32: vtable error

rielt12, 2017-08-24:
Hello,
I recently tried recompiling and I am getting this error
```
main.cpp:(.text._ZN4mscl12InertialNodeD2Ev[_ZN4mscl12InertialNodeD5Ev]+0xc): undefined reference to `vtable for mscl::InertialNode'
main.cpp:(.text._ZN4mscl12InertialNodeD2Ev[_ZN4mscl12InertialNodeD5Ev]+0x10): undefined reference to `vtable for mscl::InertialNode'
```
I am stumped.

rwslord, 2017-08-24:
Can you explain what you changed? 

Were you successful in compiling an older version of MSCL, and now compiling the latest does this? 

When do you get these errors?

rielt12, 2017-08-24:
I deleted my project folder. I then recently pulled my project off of the git server. I tried to recompile but I got that error.

rwslord, 2017-08-24:
Do you get that error when you try to compile MSCL? Or does it compile successfully and you get that error when you go to use MSCL?

There has been an update recently to the MSCL project. I'm looking into this, but in the meantime, you might want to try grabbing the version before the most recent update.

rielt12, 2017-08-28:
Hello , 

It turned out the library was not compiled correctly.
So I recompiled it and it worked.
Thanks for your efforts

closed 2017-08-28 by rielt12