LORD-MicroStrain/MSCL #33: Visual Studio 2010 Throws IPINVOKE Error

dghirsch, 2017-10-17:
I get the typical IPINVOKE issue in Visual Studio 2010, when using Visual Basic. However, when I run the code in C# everything works fine. Unfortunately, VB is not documented in the help library so I had to translate some of the code into VB, which shouldn't be hard, but might be the source of the error.

Currently I'm just trying to translate the introduction examples into VB: http://lord-microstrain.github.io/MSCL/Documentation/Getting%20Started/index.html

Does anyone have a working example? That would at least exclude the possibility that this is the "typical" IPINVOKE error that was discussed here several times.

Below is the exception I got back from trying to call mscl.LibVersion.asString() 

System.TypeInitializationException was unhandled
  HResult=-2146233036
  Message=The type initializer for 'mscl.msclPINVOKE' threw an exception.
  Source=MSCL_Managed
  TypeName=mscl.msclPINVOKE
  StackTrace:
       at mscl.msclPINVOKE.LibVersion_asString()
       at mscl.LibVersion.asString()
       at LORD_MicroStrain.Main.btnInit_Click(Object sender, EventArgs e) in 
       at System.Windows.Forms.Control.OnClick(EventArgs e)
       at System.Windows.Forms.Button.OnClick(EventArgs e)
       at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
       at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
       at System.Windows.Forms.Control.WndProc(Message& m)
       at System.Windows.Forms.ButtonBase.WndProc(Message& m)
       at System.Windows.Forms.Button.WndProc(Message& m)
       at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
       at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
       at System.Windows.Forms.NativeWindow.DebuggableCallback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
       at System.Windows.Forms.UnsafeNativeMethods.DispatchMessageW(MSG& msg)
       at System.Windows.Forms.Application.ComponentManager.System.Windows.Forms.UnsafeNativeMethods.IMsoComponentManager.FPushMessageLoop(IntPtr dwComponentID, Int32 reason, Int32 pvLoopData)
       at System.Windows.Forms.Application.ThreadContext.RunMessageLoopInner(Int32 reason, ApplicationContext context)
       at System.Windows.Forms.Application.ThreadContext.RunMessageLoop(Int32 reason, ApplicationContext context)
       at Microsoft.VisualBasic.ApplicationServices.WindowsFormsApplicationBase.OnRun()
       at Microsoft.VisualBasic.ApplicationServices.WindowsFormsApplicationBase.DoApplicationModel()
       at Microsoft.VisualBasic.ApplicationServices.WindowsFormsApplicationBase.Run(String[] commandLine)
       at LORD_MicroStrain.My.MyApplication.Main(String[] Args) in 17d14f5c-a337-4978-8281-53493378c1071.vb:line 81
       at System.AppDomain._nExecuteAssembly(RuntimeAssembly assembly, String[] args)
       at System.AppDomain.ExecuteAssembly(String assemblyFile, Evidence assemblySecurity, String[] args)
       at Microsoft.VisualStudio.HostingProcess.HostProc.RunUsersAssembly()
       at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
       at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
       at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
       at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
       at System.Threading.ThreadHelper.ThreadStart()
  InnerException: System.TypeInitializationException
       HResult=-2146233036
       Message=The type initializer for 'SWIGExceptionHelper' threw an exception.
       Source=MSCL_Managed
       TypeName=SWIGExceptionHelper
       StackTrace:
            at mscl.msclPINVOKE.SWIGExceptionHelper..ctor()
            at mscl.msclPINVOKE..cctor()
       InnerException: System.BadImageFormatException
            HResult=-2147024885
            Message=An attempt was made to load a program with an incorrect format. (Exception from HRESULT: 0x8007000B)
            Source=MSCL_Managed
            StackTrace:
                 at mscl.msclPINVOKE.SWIGExceptionHelper.SWIGRegisterExceptionCallbacks_mscl(ExceptionDelegate applicationDelegate, ExceptionDelegate arithmeticDelegate, ExceptionDelegate divideByZeroDelegate, ExceptionDelegate indexOutOfRangeDelegate, ExceptionDelegate invalidCastDelegate, ExceptionDelegate invalidOperationDelegate, ExceptionDelegate ioDelegate, ExceptionDelegate nullReferenceDelegate, ExceptionDelegate outOfMemoryDelegate, ExceptionDelegate overflowDelegate, ExceptionDelegate systemExceptionDelegate)
                 at mscl.msclPINVOKE.SWIGExceptionHelper..cctor()
            InnerException:

dghirsch, 2017-10-17:
Update: I just figured out that when putting the configuration manager to Any CPU it works. Seems to be different than some of the other languages.

rwslord, 2017-10-17:
whoa, that's unexpected. I'll have to take a look at that at some point. In C#, its always the opposite, where you have to set it to either x86 or x64 _instead_ of Any CPU. Good find

closed 2024-10-24 by dacuster