LORD-MicroStrain/MSCL #34: High CPU usage when reading data

KateBlueSky, 2017-11-06:
Hi,

We're reading from 2 sensors using the python 2.7 swig wrapper api. On both sensors we're sampling at 512Hz with a synchronization network on all 3 channels. One issue we're seeing is a high CPU utilization rate when looping through the sweeps and then the data points for each sensor.  

Attached is the code we're using to perform the reading of the sensors. The readRobotVibrationsData in the LordMicroStrainReader.txt is the method we're calling. Also attached is the cProfile log, it was created by running the attached Test.txt and calling the readRobotVibrationsData in the LordMicroStrainReader.txt. In the Test we're calling the readRobotVibrationsData only once but for the application we're creating we're calling the  readRobotVibrationsData every 500 milliseconds.  

A couple of questions:
1.) Is there a way to get the data without having to loop through each data point?
2.) Also, when looking at the cProfile I noticed the call to get the channel Name using the dataPoint.channelName() seems expensive. Are we guaranteed the dataPoints to be sequential for each sweep, so the first datapoint would be Ch1, second datapoint Ch2, and third data point Ch3? The reason I ask so we wouldn't need to call the dataPoint.channelName().  
3.) If we very the size of the internal circular buffer would we see any performance improvement? 
So by changing  the MAX_DATA_BUFFER_SIZE to something smaller.
4.) When api is reading the data from the sensor base station to the internal buffer is that a blocking call? So, if I'm calling getData() on the BaseStation object is that blocked while the internal circularbuffer is being written to? 

Thanks,
Kate

[cProfile.txt](https://github.com/LORD-MicroStrain/MSCL/files/1447418/cProfile.txt)
[Test.txt](https://github.com/LORD-MicroStrain/MSCL/files/1447422/Test.txt)
[LordMicroStrainReader.txt](https://github.com/LORD-MicroStrain/MSCL/files/1447406/LordMicroStrainReader.txt)

rwslord, 2017-11-06:
What CPU utilization are you seeing? Also, how many Nodes are sampling, and at what sample rate(s)?
 Our PC software, [SensorConnect](http://www.microstrain.com/software/sensorconnect), and our Linux-based [WSDA-1500](http://www.microstrain.com/wireless/WSDA-1500) both use MSCL for data parsing. We loop over the data continuously, as you are doing, and don't notice high CPU usage. 

1. Unfortunately, there is no other way to get the data without looping over each point.

2. You might want to try using [ChannelId](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessDataPoint.h:WirelessDataPoint.channelId) instead of the channel name.

3. I'm not sure. This is something that you can play with, but as we haven't seen the high CPU usage issue, this isn't something we've needed to try.

4. Yes, there is a mutex that gets locked when writing to and reading from the buffer container.

pa2492, 2019-02-25:
Hello, 
I am reading the G-Link 200 data in labview using the example "MSCL Aquire data.VI " as it is,  and I am seeing very high CPU usage that is locking the VI after 5 mins. Any idea how to fix this issue? 
Thanks

rwslord, 2019-02-25:
going to pick up this conversation in your other issue (#74)

mglord, 2021-06-09:
Closing as resolved (see issue #74) - if you continue to have trouble feel free to re-open this or create another issue. Thanks!

closed 2021-06-09 by mglord