LORD-MicroStrain/MSCL #35: synchronized stopping armed Nodes

Nasmom, 2017-12-30:
Hi,
 I am trying to stop the logging after certain time, but when I set nodes to idle(in order to stop them) they don't stop at the same point. So I got different sizes of logged data from each node.

How can I force all nodes to stop the armed data logging at the same time?

Thanks

rwslord, 2018-01-02:
@Nasmom, 

The main configuration option that you want is the [numSweeps](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Configuration/WirelessNodeConfig.h:WirelessNodeConfig.numSweeps) option. This determines the number of sweeps that the Node will sample for before it automatically returns to idle. However, this setting is ignored if [unlimitedDuration](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Configuration/WirelessNodeConfig.h:WirelessNodeConfig.unlimitedDuration) is enabled (which it is by default), so you will need to disable that as well.

Please note that Armed Datalogging is a legacy mode that is being deprecated soon. Newer products do not support it, and future versions of MSCL will have it removed. Instead, we recommend using Synchronized Sampling, or Non-Sync sampling, with the [dataCollectionMethod](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Configuration/WirelessNodeConfig.h:WirelessNodeConfig.dataCollectionMethod) set to logging.

Nasmom, 2018-01-03:
that solved my problem.
Thank you

rwslord, 2018-01-03:
excellent!

closed 2018-01-03 by rwslord