LORD-MicroStrain/MSCL #38: undefined symbol: _ZN4mscl10Connection9WebSocketERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEt

crosvera, 2018-01-18:
Hey @rwslord,

I was compiling the last version of mscl with the following parameters:

    bjam MSCL//stage_c++ release --without-ssl --without-websockets -j 2
    bjam MSCL//stage_python_x86 release --without-ssl --without-websockets

And everything is ok, but when I try to import the python module I got the following:

    ImportError: /home/dm_status/DM/mscl/_mscl.so: undefined symbol: _ZN4mscl10Connection9WebSocketERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEt

Any idea about what could be happening?

Best regards.

rwslord, 2018-01-18:
That's strange...we do the same thing here and haven't seen that problem. 

Is it possible you have multiple versions of python on your machine (32-bit vs 64-bit, 2.7 vs 3.6)? I'm wondering if it got compiled with a version of python that was different than the one you are trying to import it with.

crosvera, 2018-01-18:
Now it's working, I just deleted the `-j 2` parameter, seems like MSCL doesn't support parallel compilling. :smile:

closed 2018-01-18 by crosvera

rwslord, 2018-01-18:
Hmm...isn't the parameter `-j2` instead of `-j 2` ? Maybe that would fix it as well.