LORD-MicroStrain/MSCL #39: Python import error MSCL_32.8.3

triceraflops, 2018-01-25:
I'm running Fedora 27, with Python 3.6.4. When I try to import the mscl library I get the error below. It works without issues on Windows 7 using the same version of python.

Can anyone reproduce this? Is it a swig problem?

import mscl

ModuleNotFoundError                       Traceback (most recent call last)
~/Downloads/Python3.6/x64/mscl.py in swig_import_helper()
     13         try:
---> 14             return importlib.import_module(mname)
     15         except ImportError:

/usr/lib64/python3.6/importlib/__init__.py in import_module(name, package)
    125             level += 1
--> 126     return _bootstrap._gcd_import(name[level:], package, level)
    127 

/usr/lib64/python3.6/importlib/_bootstrap.py in _gcd_import(name, package, level)

/usr/lib64/python3.6/importlib/_bootstrap.py in _find_and_load(name, import_)

/usr/lib64/python3.6/importlib/_bootstrap.py in _find_and_load_unlocked(name, import_)

ModuleNotFoundError: No module named '_mscl'

During handling of the above exception, another exception occurred:

ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-3b6a8703300a> in <module>()
----> 1 import mscl

~/Downloads/Python3.6/x64/mscl.py in <module>()
     15         except ImportError:
     16             return importlib.import_module('_mscl')
---> 17     _mscl = swig_import_helper()
     18     del swig_import_helper
     19 elif _swig_python_version_info >= (2, 6, 0):

~/Downloads/Python3.6/x64/mscl.py in swig_import_helper()
     14             return importlib.import_module(mname)
     15         except ImportError:
---> 16             return importlib.import_module('_mscl')
     17     _mscl = swig_import_helper()
     18     del swig_import_helper

/usr/lib64/python3.6/importlib/__init__.py in import_module(name, package)
    124                 break
    125             level += 1
--> 126     return _bootstrap._gcd_import(name[level:], package, level)
    127 
    128 

ModuleNotFoundError: No module named '_mscl'

rwslord, 2018-01-26:
@triceraflops - First of all...loving the name 👍 

The version of MSCL that is built and ready to run, will only work on Windows. Did you build MSCL for Python on Linux? Build instructions can be found [here](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Linux.md).

We are hoping to get pre-built versions available for Linux as well in the future.

triceraflops, 2018-01-28:
Thanks! Sorry, I should have read the docs first.

closed 2018-01-28 by triceraflops