LORD-MicroStrain/MSCL #42: Issue with setActiveChannelFields

ntaylorBMT, 2018-03-06:
Hi, I am pretty new to python and i am trying to set up a 3DM-GX5-15 to constantly stream values, in this case euler angles. I can successfully connect and ping the device but i am haing troubble with the setActiveChannelFields function. I am getting the error shown below

[image]

It looks like the error says the function needs 3 arguments but in the [documentation](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#CClass:InertialNode:setActiveChannelFields) it only shows 2 

Do you have any ideas on how to solve this? my code is below. 
Also how would i set the device to stream two values ex. euler angles and linear accelerations?

Thank you for the help,

`# -*- coding: utf-8 -*-
"""
Spyder Editor

This is a temporary script file.
"""

import mscl
import misc

try:
   
    
        connection = mscl.Connection.TcpIp("10.0.1.53",10001)
        #create the InertialNode, passing in the connection
        node = mscl.InertialNode(connection)
        pingSuccess=node.ping()#test is already connected to imu
        if pingSuccess==True:# only reconnect if not already connected
             print misc.GREEN + "ping success" + misc.ENDC
        else:
            print misc.RED + "ping failed" + misc.ENDC
            
                     
        node.setToIdle()#set the accelerometer to idle so it can be configured
        
        

        sampleRate1=mscl.SampleRate(1,100)# mode 1=hertz, 100 hertz
      
        MipChannel1=mscl.MipChannel(0x8205,sampleRate1)#euler angles for channel 1
              

       node.setActiveChannelFields(0x80,MipChannel1)#need to fix this
             
except mscl.Error, e:
    
        print(e)

    

`

rwslord, 2018-03-06:
The `setActiveChannelFields` actually takes a `MipChannels` (note the 's') as a second parameter. You are passing it a single `MipChannel`. 

You want to do something like the following:

```python

mipChannel1 = mscl.MipChannel(mscl.MipTypes.CH_FIELD_ESTFILTER_ESTIMATED_ORIENT_EULER, mscl.SampleRate.Hertz(100))

chs = mscl.MipChannels()

chs.append(mipChannel1)
#append more 'mscl.MipChannel' objects to 'chs' as necessary
```

One other thing to note (that I'd like to abstract and make better in the future), is that the `setActiveChannelFields` function needs to be called once per `mscl.MipTypes.DataClass` that you want to change. For instance, the function will error out if you try to set Estimation Filter channel fields for the `mscl.MipTypes.CLASS_AHRS_IMU` Data Class, which is what it looks like your code was doing as well.

ntaylorBMT, 2018-03-09:
It worked. Thank you for the help.

closed 2018-03-09 by ntaylorBMT