LORD-MicroStrain/MSCL #43: Error codes with simple calls

tjtolon, 2018-03-08:
Hello,
We are having problems with some basic interacting with the 3DM-GX5-25.
We have taken the sample code from the quick start guide, but the code fails at the point where we call
`node.setActiveChannelFields` . This call throws an `mscl::Error_MipCmdFailed`, with an error code 4. We only took the code relevant to the AHRS component.

Then we tried to see more of the state of the device, and we made a call to `node.getBasicDeviceStatus()` right after initiating the node with a connection that succeeds when pinging. This getBasicDeviceStatus call throws an execption, again of type Error_MipCmdFailed, but with an error code 3.

Code we use is available in this gist https://gist.github.com/tjtolon/d7f25c9a911f37a9c738c02eeb1ae117

Best Regards,
Tatu

rwslord, 2018-03-08:
Strange. Is it only ping that is succeeding at this point? Can you try performing multiple pings, back to back? Do you always get a good ping response?

Are you connecting with a USB cable? Or direct to a serial port?

Do you have a Windows machine that you can try this with? Or try using SensorConnect or MIP Monitor? This would help us make sure the device is functioning properly.

tjtolon, 2018-03-09:
Yes, ping is the only thing that returns a success. Multiple ping calls as well.

Sensor is connected with a USB cable. We tested that it works on WIndows with MIP monitor, with all the needed data going in and out.

One interesting point is that when looking at dmesg, when connecting the device to ports, only one of the ports succeeded with message:

>[169711.903378] usb 1-7.1: new full-speed USB device number 57 using xhci_hcd
>[169711.975404] usb 1-7.1: device descriptor read/64, error -32
>[169712.415399] usb 1-7.1: new full-speed USB device number 58 using xhci_hcd
>[169712.505275] usb 1-7.1: New USB device found, idVendor=0483, idProduct=5740
>[169712.505300] usb 1-7.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
>[169712.505316] usb 1-7.1: Product: Lord Inertial Sensor
>[169712.505330] usb 1-7.1: Manufacturer: Lord Microstrain
>[169712.505343] usb 1-7.1: SerialNumber: 0000__6253.71532
>[169712.505983] usb 1-7.1: ep 0x82 - rounding interval to 1024 microframes, ep desc says 2040 microframes
>[169712.506656] cdc_acm 1-7.1:1.0: ttyACM0: USB ACM device

The failing ports gave out this kind of dmesg:

>[169953.150442] usb 1-4: new full-speed USB device number 59 using xhci_hcd
>[169953.262479] usb 1-4: device descriptor read/64, error -71
>[169953.482523] usb 1-4: device descriptor read/64, error -71
>[169953.702496] usb 1-4: new full-speed USB device number 60 using xhci_hcd
>[169953.814530] usb 1-4: device descriptor read/64, error -71
>[169954.034520] usb 1-4: device descriptor read/64, error -71
>[169954.254543] usb 1-4: new full-speed USB device number 61 using xhci_hcd
>[169954.254713] usb 1-4: Device not responding to setup address.
>[169954.458706] usb 1-4: Device not responding to setup address.
>[169954.662581] usb 1-4: device not accepting address 61, error -71
>[169954.834598] usb 1-4: new full-speed USB device number 62 using xhci_hcd
>[169954.834764] usb 1-4: Device not responding to setup address.
>[169955.038751] usb 1-4: Device not responding to setup address.
>[169955.242650] usb 1-4: device not accepting address 62, error -71
>[169955.249570] usb usb1-port4: unable to enumerate USB device

Now these test were done with Linux with Ubilinux distro with `uname -a`
>Linux up4 4.4.0-ubi4-amd64 #1 SMP Debian 4.4.35-1 (2016-11-29) x86_64 GNU/Linux

I tested with another linux board, which has a different kernel inside
>Linux up8 4.7.0.intel.r5.0 #1 SMP Tue Feb 20 15:17:41 EET 2018 x86_64 x86_64 x86_64 GNU/Linux

And that board also had a port where the sensor failed to connect to. It also had a port where the connection is successful, and that has a slightly different dmesg printout, as there is no mention of _rounding interval_:

>[604751.641488] usb 1-2: new full-speed USB device number 10 using xhci_hcd
>[604751.753526] usb 1-2: device descriptor read/64, error -71
>[604752.185554] usb 1-2: new full-speed USB device number 11 using xhci_hcd
>[604752.315776] usb 1-2: New USB device found, idVendor=0483, idProduct=5740
>[604752.315790] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
>[604752.315797] usb 1-2: Product: Lord Inertial Sensor
>[604752.315803] usb 1-2: Manufacturer: Lord Microstrain
>[604752.315809] usb 1-2: SerialNumber: 0000__6253.71532
>[604752.316573] cdc_acm 1-2:1.0: ttyACM0: USB ACM device

Now on this second board, the test program behaves identically to the first target: pings successfully, but all other commands come to Error_MipCmdFailed.

So it might be possible, that both of my host computers might be the reason for the problem, but both have couple of different setups. 

The MSCL documentation mentions to look for the error codes for their meaning. Where can I find these error code descriptions?

My host computers had the following setups:
Host 1:
- Distro: ubilinux, based on Debian, kernel 4.4.0, with ubilinux patches for Up board
- Compiler g++ 4.9.2-10, and g++ 7.3.0 both gave the same result
- Boost version 1.55.0
- MSCL compiled with options --without-ssl and --without-websockets

Host 2:
- Distro: ubuntu 16.04, kernel 4.7.0 with patches for intel patches for OpenCL
- Compiler g++ 5.4.0
- Boost version 1.58.0
- MSCL compiled with options --without-ssl and --without-websockets

From the next week forward I am unable to get my hands to the sensor, so this issue can be closed.

rwslord, 2018-03-09:
Thanks for the input. I'm going to forward this info on to the hardware/firmware engineers to check if they've seen this before.

The error codes can be found [here](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/MIP/Packets/MipPacket.h:MipPacket.MIP_ACK_NACK_ERROR_CHECKSUM_INVALID).

rwslord, 2018-03-09:
Just to clarify, are there some ports that are fully working? 

Or do you have this issue no matter which Linux machine you use, port you use, etc?

rwslord, 2018-03-09:
also, it would be interesting to know if using MSCL on Windows (which is already pre-compiled, so should be fast to get up and running) works with the same hardware.

contagon, 2020-12-10:
Hi, 

@tjtolon Did you ever find a solution to this? I have the 3DM-GX5-45 and am having identical issues with very similar code. 

It seems to be related to the channel that you're trying to enable. I can enable `CH_FIELD_SENSOR_SCALED_ACCEL_VEC`, but not `CH_FIELD_SENSOR_RAW_ACCEL_VEC` for example. I can actually pull data and everything from `CH_FIELD_SENSOR_SCALED_ACCEL_VEC` with no issues. I can continue to investigate which channels do/don't work if needed.

mglord, 2020-12-10:
Hi @contagon

Thanks for sharing! I suspect this may be what's causing the error @tjtolon was seeing on setActiveChannelFields() - in the sample code `CH_FIELD_SENSOR_RAW_ACCEL_VEC` is what is being sent. Each model supports different channel fields and raw accel data is not supported on either the GX5-45 or -25.

The supported channel fields for each device are indicated in the device's Data Communications Protocol in section 3.2 Command and Data Summary: Data
[GX5-45 DCP](https://www.microstrain.com/sites/default/files/applications/files/3dm-gx5-45_dcp_manual_8500-0064_rev_m_002.pdf)
[GX5-25 DCP](https://www.microstrain.com/sites/default/files/3dm-gx5-25_dcp_manual_8500-0065_reference_document.pdf)

The MIP data field ID to MSCL ChannelField enum mapping can be found here: [MipTypes::ChannelField](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/MIP/MipTypes.h:MipTypes.ChannelField)

You can also programatically check if a ChannelField is supported on your device prior to setting it to the active channels by checking if it's in the list of ChannelField returned by `node.features().supportedChannelFields()`.

I would hope that the device would send an invalid parameter error (error code 3) instead of a generic failure (error code 4) for unsupported ChannelFields, but I have seen inconsistent error code reporting at times. I will test on both the -45 and -25 and report the issue to our firmware team.

Hope this helps!

@tjtolon - regarding the error on `node.getBasicDeviceStatus()`: we have made significant improvements to this functionality and added support for additional devices since this issue was opened. If I remember correctly, this function was originally added with support for only the GX5-45. Please try with the latest version and let me know if you run into any issues!

contagon, 2020-12-11:
That was a perfect answer, thank you! And thanks for the quick response.

While I've got your attention, I have one more question. Could you point me to how to enable the internal EKFs in the GX5-45 in C++? I've done some digging, but can't found an obvious answer anywhere in the documentation.

mglord, 2020-12-11:
If you have access to a Windows PC our desktop application [SensorConnect](https://www.microstrain.com/software/sensorconnect) has an Initialize/Reset Estimation Filter screen that may provide some guidance for what functions/configurations are available and/or required for filter initialization. There are also additional filter settings on the Configure screen, but the Init/Reset screen should have everything that significantly impacts initialization.

Otherwise, or if you have further questions about filter initialization and/or the right configuration for your application I would encourage you to reach out to our inertial products expert, Barry Trutor ([barry.trutor@parker.com](barry.trutor@parker.com)) - he'll be able to offer much better guidance than I can.

Once you know the required commands/configuration/functionality we can certainly help with MSCL implementation if you run into any issues!

mglord, 2021-06-09:
Closing as resolved - if you continue to have trouble feel free to re-open this or create another issue. Thanks!

closed 2021-06-09 by mglord