LORD-MicroStrain/MSCL #45: Your library doesn't compile correctly under linux.

PandaHugMonster, 2018-03-11:
How the archive size can be less than 1 byte?!
Your library does not possible to compile. Please fix it as soon as possible!

Dump:
1 folder, 0 files, 0 bytes
Creating archive: ../Output/Examples.zip
Items to compress: 1
Files read from disk: 0
Archive size: 152 bytes (1 KiB)
Everything is Ok
(zipOutput.sh) Removing temp directory that was zipped
(zipOutput.sh) DONE
========================================================
(buildAll.sh) DONE
========================================================

crosvera, 2018-03-11:
Hello There,

Your message doesn't give much information. Did you follow the build guide?
What is your setup environment? can you pastebin the hole compiling log?

Best regards

PandaHugMonster, 2018-03-11:
I tried everything from your guide.
C++ .so file is generated, and it works (But your guide was wrong in so many places, or due to version updates, or because of something else).
And as much as I tried generating of mcsl.py is happening, but generating of _mscl.so (for python) is not happening. And nothing in FAQ about that...

So problems are in your installation scripts.
With all my respect, but I don't have time to post everything that I tired... So I'm just notifying you that you have problems in your guide or/and in your code.
I showed you as well that archive is generating wrong. It puts only empty folders in it.

The only way I could use more/less your library (in C++ only...) through that repository: https://github.com/AbundantRobotics/MSCL 
Because there was added cmake files.
I think if you could just prebuilt all the binaries for linux on the row as you did for windows - that would be much easier to use than now with problems in documentation and in the code.

rwslord, 2018-03-12:
Thank you for your feedback @PandaHugMonster. Providing prebuilt binaries for Linux is definitely on our todo list. 

Sorry you are having difficulties. The build scripts definitely work for me, and other customers. Usually the trouble comes in getting the system set up with dependencies, telling bjam/python where everything is, etc. If you are still having trouble and would like some more assistance, please let me know. Unfortunately, there isn't much troubleshooting I can do with the information that you provided.

closed 2018-05-29 by rwslord