LORD-MicroStrain/MSCL #46: python wrapper can't build in v35.1.2

pulux, 2018-03-21:
see the build output
```
bjam MSCL//stage_python_x86 release --without-ssl --without-websockets | grep error
MSCL/source/mscl/Wrapper/../Utils.h:521:16: error: 'function' in namespace 'std' does not name a template type
MSCL/source/mscl/Wrapper/../Utils.h:499:13: error: class 'mscl::Utils::Lazy<T>' does not have any field named 'm_loadFn'
MSCL/source/mscl/Wrapper/../Utils.h:504:13: error: class 'mscl::Utils::Lazy<T>' does not have any field named 'm_loadFn'
MSCL/source/mscl/Wrapper/../Utils.h:514:27: error: there are no arguments to 'm_loadFn' that depend on a template parameter, so a declaration of 'm_loadFn' must be available [-fpermissive]

```
```
$ git grep m_loadFn
MSCL/source/mscl/Utils.h:            m_loadFn(loadFn),
MSCL/source/mscl/Utils.h:            m_loadFn(nullptr),
MSCL/source/mscl/Utils.h:                m_value = m_loadFn();
MSCL/source/mscl/Utils.h:          std::function<T()> m_loadFn;
```

rwslord, 2018-03-21:
Can you tell me what version of g++/gcc you are using?

pulux, 2018-03-22:
This test was done with gcc-7.3.0 and boost-1.65.1 under voidlinux with glibc 2.26

A Test under alpinelinux 3.6, gcc-6.3.0, boost-1.64, musl-libc 1.1.16, and voidlinux with musl-libc 1.1.19 brings other errors. It fails to build the c++ lib with:
```
MSCL/source/mscl/LibVersion.h:31:23: error: expected unqualified-id before 'unsigned'
MSCL/source/mscl/LibVersion.h:31:23: error: expected ')' before 'unsigned'
MSCL/source/mscl/LibVersion.h:31:23: error: expected ')' before 'unsigned'
MSCL/source/mscl/LibVersion.h:38:23: error: expected unqualified-id before 'unsigned'
MSCL/source/mscl/LibVersion.h:38:23: error: expected ')' before 'unsigned'
MSCL/source/mscl/LibVersion.h:38:23: error: expected ')' before 'unsigned'
MSCL/source/mscl/LibVersion.cpp:13:24: error: expected unqualified-id before '(' token
MSCL/source/mscl/LibVersion.cpp:18:24: error: expected unqualified-id before '(' token
MSCL/source/mscl/MicroStrain/Wireless/NodeMemory_v1.h:47:29: error: expected unqualified-id before numeric constant
MSCL/source/mscl/MicroStrain/Wireless/NodeMemory_v1.h:47:29: error: expected unqualified-id before numeric constant
MSCL/source/mscl/MicroStrain/Wireless/NodeMemory_v1.h:47:29: error: expected unqualified-id before numeric constant
```
Is seems here the functions `major()` and `minor()` and the const `PAGE_SIZE` are the problems. A rename to e.g. _major(), _minor() and DPAGE_SIZE fixed it.

If this is fixed the lib and the wrapper is build under alpinelinux (gcc-6.3.0) but not under voidlinux (gcc-7.3.0, boost-1.65.1).

rwslord, 2018-03-22:
I've seen warnings about the major/minor functions like the errors you are seeing. My patch is slightly different, but it'll be available in the next release of MSCL on github.

As far as the build with gcc-7.3.0 failing, I have a hunch that the error is coming from `#include <functional>` being missing from `Utils.h.` I'm betting some previous header files included <functional>, which was making it work. It could be worth adding that and trying to build again. I'll add that in my next MSCL release as well.

mglord, 2021-06-09:
Confirmed that `#include<functional>` is now specified in `Utils.h`. Closing as resolved - if you continue to have trouble feel free to re-open this or create another issue. Thanks!

closed 2021-06-09 by mglord