LORD-MicroStrain/MSCL #48: Error importing mscl.py

jfg-creare, 2018-04-18:
`Python 2.7.9 (default, Sep 17 2016, 20:26:04)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mscl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mscl.py", line 17, in <module>
    _mscl = swig_import_helper()
  File "mscl.py", line 16, in swig_import_helper
    return importlib.import_module('_mscl')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: libboost_chrono.so.1.64.0: cannot open shared object file: No such file or directory
>>>
`

I think my problem relates to this, from the Linux install guide:
"Note: The default Python build does not link in dependencies (such as Boost) directly. You will need to link the dependencies in when using mscl.py in your project."

I don't understand what, how, or where I'm supposed to link.

rwslord, 2018-04-18:
Most likely your LD_LIBRARY_PATH isn't set up correctly. 

When building MSCL, you likely typed a line like this (with the path to where you installed boost):
``export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/jenkins/boost/boost_1.64.0_installed/lib``

Can you try doing this, and then running your python code?

There are more permanent ways to set up environment variables as well. These links might help:
https://unix.stackexchange.com/questions/117467/how-to-permanently-set-environmental-variables

https://stackoverflow.com/questions/1641477/how-to-set-environment-variable-for-everyone-under-my-linux-system

jfg-creare, 2018-04-18:
That did the trick. Thank you!

closed 2018-04-18 by jfg-creare