LORD-MicroStrain/MSCL #49: Does not connect to 3DM-Gx2

rookiebeotch, 2018-04-26:
using python version with 3DM-GX2 over serial and any call via the  InertialNode object returns Failed to communicate with device. i.e. call node.SetToIdel().  

 Using python example as a starting point.

The serial port works as i can read bytes from a gps.
The 3DM-Gx2 works on windows with microstrain software gui.

is this compatible with the 3DM-GX2 devices?

rwslord, 2018-04-27:
Unfortunately, the 3DM-GX2 is not supported by MSCL.

rookiebeotch, 2018-04-27:
Oh :(  thanks for the reply.
I was able to use Wine on ubuntu and run the Inertial Tool so i was able to
solve my issue.
thanks

jorge

rwslord, 2018-04-30:
I'm glad you were able to solve your issue. Sorry about that...the GX2 line is an older product that communicates differently than our GX3, GX4, and GX5 lines of 3DM products.

closed 2018-04-30 by rwslord

rookiebeotch, 2018-04-30:
Thanks for reply.  Yes im not surprised.  We had them sitting on the shelf
for almost 10 years, lol.   They still work though which is good.

jorge

rwslord, 2018-04-30:
haha, awesome 👍 
Let me know if you want to purchase a new one...the new model is pretty amazing, and works with MSCL : )