LORD-MicroStrain/MSCL #51: Boost  No such file or directory

HTLife, 2018-05-14:
After following the instructions on README page, I got this error message when building MSCL.
```
MSCL/source/mscl/ScopeHelper.cpp:6:
MSCL/source/mscl/Communication/BoostCommunication.h:14:27: fatal error: boost/beast.hpp: No such file or directory
```

My installation steps:
```
tar --bzip2 -xf /path/to/boost_1_64_0.tar.bz2
```
Add user-config.jam to /home/username/boost_1_64_0
```
using gcc
:	host
:	g++
:	<cxxflags>-I/usr/include/python2.7
	<cxxflags>-I/home/username/boost/boost_1.64.0_installed/include
	<cxxflags>-I/usr/include/openssl
	<linkflags>-L/home/username/boost/boost_1.64.0_installed/lib
;
```
```
./bootstrap.sh --prefix=/home/username/boost
./b2 install
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/username/boost/lib
bjam MSCL//stage_c++ release
```

rwslord, 2018-05-14:
I'm assuming [this](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Linux.md) is the page you followed the instructions from, right? Did you download the Beast library and put it with the other headers in the Beast library?

Many users don't actually need the WebSocket functionality that is part of MSCL. If you don't you can skip the beast and openssl parts by passing `--without-ssl` and `--without-websockets` to the bjam command. So it would look like this:
`bjam MSCL//stage_c++ release --without-ssl --without-websockets`

HTLife, 2018-05-15:
@rwslord Thank you!  The final command works for me.

closed 2018-05-15 by HTLife

HTLife, 2018-05-15:
@rwslord Did LORD provide any Ubuntu 1604 driver for IMU (3DM-GX5-25)?

rwslord, 2018-05-15:
@HTLife I believe the 3DM-GX5-25 should just work plug-and-play with the 3DM-GX5-25. Does it show up as a tty device when you plug it in?

HTLife, 2018-05-16:
@rwslord I saw the device by typing `lsusb` (`Bus 001 Device 016: ID 0483:5740 STMicroelectronics STM32F407`).

How can I find the correct port name that was used by [ParseData.cpp](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL_Examples/Inertial/C%2B%2B/MSCL_Inertial_Example_C%2B%2B/ParseData/ParseData.cpp)
```cpp
const string COM_PORT = "COM15";
```

HTLife, 2018-05-16:
I got the following error message by executing ParseData.cpp (com port set to : "/dev/ttyS0")
```
$ sudo ./hello 
Error: Failed to communicate with the device.
sh: 1: pause: not found
```

TTY status
```
$ sudo setserial -g /dev/ttyS[0123]
/dev/ttyS0, UART: 16550A, Port: 0x03f8, IRQ: 4
/dev/ttyS1, UART: unknown, Port: 0x02f8, IRQ: 3
/dev/ttyS2, UART: unknown, Port: 0x03e8, IRQ: 4
/dev/ttyS3, UART: unknown, Port: 0x02e8, IRQ: 3

```

My code modify from ParseData.cpp:
```cpp
//MSCL Example: ParseData
//  This example shows how to parse incoming data from an Inertial Device.
//  This example does not start a Node sampling. To receive data, a Node
//  must be put into a sampling mode.
//
//Updated: 12/20/2017

#include <iostream>
using namespace std;

#include "mscl/Types.h"

#include "mscl/Communication/Connection.h"
#include "mscl/MicroStrain/Inertial/InertialNode.h"
#include "mscl/Exceptions.h"
#include "mscl/MicroStrain/MIP/MipTypes.h"

int main(int argc, char **argv)
{
	//TODO: change these constants to match your setup
	const string COM_PORT = "/dev/ttyS0";

	try
	{
		//create a SerialConnection with the COM port
		mscl::Connection connection = mscl::Connection::Serial(COM_PORT);

		//create an InertialNode with the connection
		mscl::InertialNode node(connection);

        //Put the Inertial Node into its idle state
        //  (This is not required but reduces the parsing
        //  burden during initialization and makes visual
        //  confirmation of the commands easier.)
        node.setToIdle();

        //build up the channels to set
        //mscl::InertialChannels sensorChs;
        mscl::MipChannels sensorChs;
        
        sensorChs.push_back(mscl::MipChannel(mscl::MipTypes::CH_FIELD_SENSOR_SCALED_ACCEL_VEC, mscl::SampleRate::Hertz(100)));

        sensorChs.push_back(mscl::MipChannel(mscl::MipTypes::CH_FIELD_SENSOR_SCALED_GYRO_VEC, mscl::SampleRate::Hertz(100)));

        //set the active channels for the Sensor category on the Node 
        node.setActiveChannelFields(mscl::MipTypes::CLASS_AHRS_IMU, sensorChs);
        
        //start sampling on the Sensor category of the Node
        node.enableDataStream(mscl::MipTypes::CLASS_AHRS_IMU);
        

		//endless loop of reading in data
		while(true)
		{
            mscl::MipDataPoints data;
            //get all the data packets from the node, with a timeout of 500 milliseconds
            mscl::MipDataPackets packets = node.getDataPackets(500);

            for(mscl::MipDataPacket packet : packets)
            {
                data = packet.data();
                printf("%f", data[0].as_double());
            }
		}
	}
	catch(mscl::Error& e)
	{
		cout << "Error: " << e.what() << endl;
	}

	system("pause");
	return 0;
}

```

rwslord, 2018-05-16:
@HTLife How are you connecting the device? Is it via USB? I believe the GX5-25 will show up as a ttyACM port. Try comparing these when the device is attached vs when it is unattached to get an idea of which device it is.

HTLife, 2018-05-16:
@rwslord It works!  Now I can read the data correctly!

The correct port name is `/dev/ttyACM0`.