LORD-MicroStrain/MSCL #53: Incorrect sampling frequency

hooksj, 2018-05-18:
Hello,

I am using a 3DM-GX4-25 to read both Euler angles and angular rates. I would like to do this at the highest possible frequency (which from the documentation on the adaptive Kalman Filter appears to be 500 Hz). Currently I am trying to sample at 500Hz and I have written a test piece of code to see if it is actually running at 500Hz. The two files shared below are imu.py and run_imu.py. The file imu.py is a class that communicates with the 3DM-GX4-25. The file run_imu.py uses that class to continuously read from the imu. 

In the test code you will see that the IMU is set to sample at 500Hz but my code reads from the IMU at 400Hz. This would imply that every time I read from the IMU I should have a new piece of data if the IMU is actually sampling at 500Hz. I have written code to compare current data to previous data and I am finding that I am getting a substantial amount of repeated data. On average I get 3 repeated values in a row which implies the IMU is sampling at a frequency of 133Hz. 

imu.py - https://gist.github.com/hooksj/0322ea03251ffd3567497d2f52e4b98c

run_imu.py - https://gist.github.com/hooksj/0791d6eae0975cc677db907f75ab0146

Please let me know if I have set something up incorrectly. My ideal setup is that each time I read from the IMU I get the latest estimation. It would be great if the buffer was only the size of 1 and never got filled up but I was unable to find a setting that would allow me to do that. Thank you in advance!

hooksj, 2018-05-18:
After playing with this even more I realized the reason I was getting so much repeated data was due to the fact that I had my timeout set to 1 millisecond.

packets = self.node.getDataPackets(1)

Once I raised this to 10 milliseconds then I no longer got repeating data but that means I am only able to read from the IMU at 100Hz. This makes me think I am not setting up the streaming correctly because that seems far to slow. How can I communicate as fast as possible?

rwslord, 2018-05-18:
When you say you are seeing duplicate data, do you mean the values for a single channel are the same? Is the `packet.collectedTimestamp()` for the MipDataPacket the same for the duplicate points?

The way that data collection works in MSCL is that once you create the `InertialNode` object, any data that is streaming in will be placed in a large circular data buffer for that device (whether you are asking for the data or not). When you call `InertialNode.getDataPackets()` it just pulls data out of this circular buffer and gives it to you. 

The first parameter in getDataPackets() is the timeout. This is how long (in milliseconds) the function will block if there is no data in the circular buffer. Once data arrives, or if there is already data in the buffer, the function will return immediately. So this _shouldn't_ have any affect on getting duplicate data. It shouldn't change the data at all. If you leave this at 0, it will just retrieve all the packets that are immediately available, and return immediately if there are none.

The second parameter in getDataPackets() is the maxPackets. This can be used if you want to cap the number of packets returned from the function. However, this is just going to pull from the circular buffer. So if you use this parameter, and your Node is sampling faster than you are asking for data, you will quickly start to fall behind (compare the packet.collectedTimestamp vs the PC time). By default, this parameter is 0, which will return all packets that have been collected.

From what I can tell, the way you are setting up the channels looks correct. The device should be sending data at 500Hz.

hooksj, 2018-05-18:
rwslord thank you for your quick response. I took your advice and I started investigating the issue using `packet.collectedTimestamp()`. As I am sure you suspected any duplicate data that I was receiving had the same timestamp so it was indeed the same exact data.

After discovering this fact I decided to back out what the sampling frequency of the node was by taking the difference between two consecutive different timestamps. This test revealed that the max sampling frequency of the 3DM-GX4-25 is 166Hz and this is very concerning. 

If I set the sampling frequency to 1Hz then I back out the actual frequency to be between 0.998Hz - 1.004hz. If I set the sampling frequency to 10Hz then the actual is between 9.802Hz - 10.418Hz, which seems reasonable. However, when I set the sampling frequency to 100Hz I get 83.582Hz - 165.533Hz, that seems very strange. And if I set the sampling frequency to anything above 166Hz then the actual frequency maxes out at 166Hz.

This tests tells me that I am indeed able to set the sampling frequency of the node correctly through MSCL but the AKF cannot actually performing sampling at 500Hz. I have tested this on two different 3DM-GX4-25 IMU's. Is there a setting that I need to change to be able to sample at 500Hz? Below are links to the updated code that I used to do the time testing. Thank you for looking into this issue.

imu.py - https://gist.github.com/hooksj/0322ea03251ffd3567497d2f52e4b98c

run_imu.py - https://gist.github.com/hooksj/0791d6eae0975cc677db907f75ab0146

rwslord, 2018-05-18:
There should never be duplicate data received when calling getDataPackets. 

Is it possible that you are calling `getDataPackets()` often enough that it sometimes returns no data? If this happened in your `update` function, the for loop within there wouldn't run, which would leave the old timestamp and data channels in your object. Your other file which is using the object would just assume it is new. 

Can you check if this is happening?

hooksj, 2018-05-18:
Sorry you are correct that is exactly what is happening. I now have updated it to check for when I actually have a packet. There was never any duplicate information being sent. 

I still however can only get a sampling frequency of 166Hz. Is there a setting I need to change in MIP Monitor?

rwslord, 2018-05-18:
There is no setting that controls this. The device internally samples much faster than 500Hz, so there should be no problem here. 

Another issue I see in your code is that your `update` function is throwing away all packets except the last one it gets. For instance, when you call `getDataPackets()` lets say it returns 100 packets. Your for loop iterates over all of these, but if there were points from the same channel name, it is overwriting them every time. 

This is likely causing you to think you have far less data than you actually do.

hooksj, 2018-05-18:
All right so I cut out everything and all I am doing is getting data packets as fast as I can. I see something weird as I increase the sampling frequency. If I set the sampling frequency to 50Hz then the post packets in `getDataPackets()` will be 1 and I will be sampling at 50Hz. However, if I set the sampling frequency to 500 Hz then when I do `getDataPackets()` there will be times that there are 3 packets which is strange because I am asking for data at a much much faster rate than 500Hz. When I look at the timestamps I see that the 3 packets will be sampled at very high frequencies(> 20000Hz). However, the next `getDataPackets()` that returns actual data will be 166Hz apart. So in essence the data I am receiving and is useful to me is coming in at a frequency of 166Hz.   Maybe I am still doing something incorrect, please take a look at the simplified code below. 

```
import mscl
import numpy as np
import pdb

sample_rate = 500

try:
    COM_PORT = "/dev/ttyACM1"
    connection = mscl.Connection.Serial(COM_PORT, 921600)
    node = mscl.InertialNode(connection)

    chs_filter = mscl.MipChannels()
    chs_filter.append(mscl.MipChannel(mscl.MipTypes.CH_FIELD_ESTFILTER_ESTIMATED_ORIENT_EULER, mscl.SampleRate.Hertz(sample_rate)))
    chs_filter.append(mscl.MipChannel(mscl.MipTypes.CH_FIELD_ESTFILTER_ESTIMATED_ANGULAR_RATE, mscl.SampleRate.Hertz(sample_rate)))

    node.setActiveChannelFields(mscl.MipTypes.CLASS_ESTFILTER, chs_filter)
    node.enableDataStream(mscl.MipTypes.CLASS_ESTFILTER)
    node.resume()
except mscl.Error, e:
    print "Error:", e

last_time = np.int64(0.0)
while True:
    packets = node.getDataPackets()
    first = True
    for packet in packets:
        if first:
            print ""
        first = False
        points = packet.data()
        time_stamp = packet.collectedTimestamp()
        current_time = time_stamp.nanoseconds()
        diff = (current_time - last_time)/1000000000.0
        if not diff == 0.0:
            print 1.0/diff
            print time_stamp
        last_time = current_time
```

hooksj, 2018-05-18:
Here is what the console output looks like if I run the test code above.

```
167.913170085
2018-05-18 20:44:25.041534719

51826.8981601
2018-05-18 20:44:25.041554014
165316.581253
2018-05-18 20:44:25.041560063

83.4417102079
2018-05-18 20:44:25.053544477

51671.5754663
2018-05-18 20:44:25.053563830
174034.110686
2018-05-18 20:44:25.053569576
37788.6105128
2018-05-18 20:44:25.053596039

20838.5429691
2018-05-18 20:44:25.053644027
33974.3154175
2018-05-18 20:44:25.053673461

171.431255552
2018-05-18 20:44:25.059506703

121832.358674
2018-05-18 20:44:25.059514911
252206.809584
2018-05-18 20:44:25.059518876

165.774715808
2018-05-18 20:44:25.065551159
```

rwslord, 2018-05-19:
There will be some buffering that occurs, especially in the USB to Serial driver. This might be what you are seeing. Because of this, it usually isn't good enough to depend on the PC collected timestamp if you need precise time. If you need better time-stamping, you will need to enable one of the timestamp channels and use that when parsing the data.

mglord, 2021-06-09:
As mentioned previously, this is most likely caused by inherent buffering over USB. To get the actual data sampled timestamp for the EKF please add the GPS Timestamp field to the message format (`MipTypes::ChannelField::CH_FIELD_ESTFILTER_GPS_TIMESTAMP` in MSCL). To read the data sampled timestamp of a `MipDataPacket` object use the `deviceTimestamp()` function - this can be used in post-processing to accurately synchronize and space your data. If you need data to come in faster in real-time you will need to switch to connecting via serial port.

Closing as resolved, but if you continue to have trouble feel free to re-open this or create another issue. Thanks!

closed 2021-06-09 by mglord