LORD-MicroStrain/MSCL #56: Problem with  system shunt calibration of sensors using Vlink 200 via MSCL

ProteinPotato, 2018-05-30:
I am facing the following issues:
1- While using autoBalance to set the target percentage, I am using : 
    mscl.WirelessNode.autoBalance(mask, target percentage value), however, I am getting the following error: 
    autoBalance() missing 1 required positional argument: 'targetPercentage'
2- What is the hardware offset that we need to input, and how to convert it into uint16 type?
3- How to use autoShuntCal(const ChannelMask &mask,const ShuntCalCmdInfo &commandInfo)? That is how do I set the shuntCalCmdInfo?

rwslord, 2018-05-30:
1. Can you show me your code for auto balance?

2. I'm not sure what context you mean here. Where are you trying to use hardware offset? There is a [WirelessNode::getHardwareOffset](getHardwareOffset) function that you can use to get the current offset. You will need to pass in a channel mask that supports the channel group setting of  [mscl::WirelessTypes::ChannelGroupSetting::chSetting_hardwareOffset](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessTypes.h:WirelessTypes.ChannelGroupSetting)

3. You will want to first create a default [ShuntCalCmdInfo](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Commands/AutoCalInfo.h:ShuntCalCmdInfo) object. Then set everything how you would like it on that object (# of active gauges, gauge resistance, shunt resistance, etc). Then pass this in with the autoShuntCal command.

All this being said, the majority of customers don't need to use the configuration side of MSCL, as they usually just set up their Nodes once in SensorConnect, and then use MSCL for starting/stopping sampling, and collecting data. The settings will stay applied to the Nodes even after a power cycle. If you need to change settings in the application that you are writing, then you are on the right track. Just want to make sure you aren't adding extra work for yourself.

mglord, 2021-06-15:
I'm going to go ahead and close this but if you continue to run into issues feel free to re-open. When re-opening please provide as much additional information as possible.

As mentioned previously, most users are able to configure nodes via SensorConnect and then user MSCL only for data collection but if you need to run a shunt cal through your application there are some additional details and instructions [here](https://github.com/LORD-MicroStrain/MSCL/issues/162#issuecomment-859777447). Some of those instructions are specific to the SG-Link-200-OEM, but if anything is unclear or not working as expected just let us know and we'd be happy to clarify!

Thanks!

closed 2021-06-15 by mglord