LORD-MicroStrain/MSCL #58: 3DM-GX5-25 Estimation filter randomly fails

Martikain, 2018-06-21:
I have been testing the Lord 3DM-GX5-25 AHRS sensor for some while now. The problem is that the sensor only randomly produces valid estimation filter values.  Most of the time all of the measurements show as invalid. The raw sensor data is shown correctly. Is there some sort of estimator setup that needs to be run before the estimator is used?

rwslord, 2018-06-21:
Yes, the 3DM-GX5-25 Attitude and Heading Reference System (AHRS) does require the user to apply various settings to groom the device for use in their particular application. These settings insure that the Estimation Filter properly initializes and is maintained during operation. Please consult the user manual or your LORD support engineer at sensing_support@lord.com.

gilberko1, 2020-10-13:
I have some issues with raw data with this sensor (magnetic problems), can you please explain which settings I should "groom" and how?

mglord, 2020-10-13:
I would encourage you to reach out to our product support engineers at [sensing_support@lord.com](sensing_support@lord.com). The MSCL developers do not have the product-specific knowledge to assist with this issue. When you reach out please include as much detail as possible, including the data results you expect as well as the actual behavior.

Once the desired configuration is determined, I can assist with any implementation issues you may run into when trying to configure the device through MSCL.

closed 2021-06-15 by mglord