LORD-MicroStrain/MSCL #59: upgrade firmware

crosvera, 2018-07-10:
Hello there,
Is there any method to upgrade basestation or wireless node firmware via MSCL?
It would be great to have a method, that way we could automate some stuff.

Best regards.

rwslord, 2018-07-10:
Unfortunately, we opted to keep this out of MSCL, and only put it in our higher-level software (SensorConnect).

crosvera, 2018-07-10:
Is there any raw recipe to upload the firmware? Maybe sending raw bytes via
serial or something?

Best regards.

rwslord, 2018-07-11:
There are a few internal (undocumented) commands that need to be used, and it differs a lot between BaseStations and Nodes, as well as between the "-LXRS" series products and the "-200" series products. Our main worry was that MSCL users could end up bricking their devices by loading the wrong firmware, canceling the process at the wrong time, etc.

crosvera, 2018-07-11:
Interesting, I understand the risk doing a firmware upgrade and I thinks is
the same risk upgrading any firmware in any device, you can brick your
device.

Currently we are working with:
- WSDA-Base-104 LXRS
- SG-Link OEM LXRS
- SG-Link LXRS (wireless 2 channel analog input sensor node)

It would be great if you can share the commands to upgrade the firmware in
those devices.

Best regards

rwslord, 2018-07-11:
The process to upgrade is rather complex, and it isn't something that can be easily given out. Currently, we only support firmware upgrades using SensorConnect. We could entertain making this feature available outside our software, however this would become a business decision and require funding to gain priority.

If this is important to your company, I would definitely recommend contacting our sales team.

mglord, 2021-06-15:
I'm afraid this is still true and we have no current plans to expose this functionality outside of SensorConnect. I'm going to go ahead and close this, but as mentioned feel free to reach out to our [sales team](https://www.microstrain.com/contact-sales) if this is an application requirement to talk through options.

Sorry for the inconvenience!

closed 2021-06-15 by mglord