LORD-MicroStrain/MSCL #61: Can ping but not pull data on inertial node

smbryan, 2018-08-08:
I'm having trouble pulling data from a `3DM-GX3-25`. I am able to ping the device, but not pull data at the moment. 

Dev Environment:
- Ubuntu 16.04
- MSCL:
  - Built with Boost 1.67.
  - `--without-ssl` & `--without-websockets` flags used.
- Device is connected on `/dev/ttyACM0`.
- User is added to the `dialout` group.

Using this modified `ParseData.cpp`:

```
#include <iostream>
using namespace std;

#include "mscl/Communication/Connection.h"
#include "mscl/Exceptions.h"
#include "mscl/MicroStrain/Inertial/InertialNode.h"

int main(int argc, char **argv) {
  // DONE: change these constants to match your setup
  const string port = "/dev/ttyACM0";
  mscl::uint32 baudrate = 921600; // Also tried 115200
  try {
    // create a SerialConnection with the COM port
    mscl::Connection connection = mscl::Connection::Serial(port, baudrate);
    // mscl::Connection connection = mscl::Connection::Serial(COM_PORT);

    // create an InertialNode with the connection
    // mscl::InertialNode node(connection);
    mscl::InertialNode node = mscl::InertialNode(connection);
    bool success = node.ping();
    if (success) {
      std::cout << "Ping!" << std::endl;
    } else {
      std::cout << "[crickets]" << std::endl;
    }
    // endless loop of reading in data
    while (true) {
      // get all the data packets from the node, with a timeout of 500
      // milliseconds
      mscl::MipDataPackets packets = node.getDataPackets(500);

      for (mscl::MipDataPacket packet : packets) {
        // print out the data
        cout << "Packet Received: ";

        // get the data in the packet
        mscl::MipDataPoints data = packet.data();
        mscl::MipDataPoint dataPoint;

        // loop through all the data points in the packet
        for (unsigned int itr = 0; itr < data.size(); itr++) {
          dataPoint = data[itr];

          // print out the channel data
          cout << dataPoint.as_string()
               << " "; // Just printing this out as a string. Other methods (ie.
                       // as_float, as_uint16, as_Vector) are also available.

          // if the dataPoint is invalid
          if (!dataPoint.valid()) {
            cout << "[Invalid] ";
          }
        }
        cout << endl;
      }
    }
  } catch (mscl::Error &e) {
    cout << "Error: " << e.what() << endl;
  }

  system("pause");
  return 0;
}
```

I have a successful "Ping!" response and the program continues to run. However. no additional data is received. 

What troubleshooting steps would you recommend?

rwslord, 2018-08-08:
My first guess would be that the device isn't actively sampling any of its data fields. You can programatically  tell it start sampling specific fields in MSCL (the [Quick Start Guide](http://lord-microstrain.github.io/MSCL/Documentation/Getting%20Started/index.html?cpp#configuring-a-node11) has a good example of this), or you can use our PC software (SensorConnect or MIP Monitor) to set the active fields and enable them on startup so that the device starts sampling whenever it is powered on.

smbryan, 2018-08-08:
That was it! I overlooked the line:

> This example does not start a Node sampling. To receive data, a Node must be put into a sampling mode. 

For others, I updated the code to the following and got results:

```
// MSCL Example: ParseData
//  This example shows how to parse incoming data from an Inertial Device.
//  This example does not start a Node sampling. To receive data, a Node
//  must be put into a sampling mode.
//
// Updated: 12/20/2017

#include <iostream>
using namespace std;

#include "mscl/Communication/Connection.h"
#include "mscl/Exceptions.h"
#include "mscl/MicroStrain/Inertial/InertialNode.h"

int main(int argc, char **argv) {
  // DONE: change these constants to match your setup
  const string port = "/dev/ttyACM0";
  mscl::uint32 baudrate = 921600;
  try {
    // create a SerialConnection with the COM port
    mscl::Connection connection = mscl::Connection::Serial(port, baudrate);
    // mscl::Connection connection = mscl::Connection::Serial(COM_PORT);

    // create an InertialNode with the connection
    // mscl::InertialNode node(connection);
    mscl::InertialNode node = mscl::InertialNode(connection);
    bool success = node.ping();
    if (success) {
      std::cout << "Ping!" << std::endl;
    } else {
      std::cout << "[crickets]" << std::endl;
    }
    mscl::MipChannels ahrsImuChs;
    ahrsImuChs.push_back(
        mscl::MipChannel(mscl::MipTypes::CH_FIELD_SENSOR_RAW_ACCEL_VEC,
                         mscl::SampleRate::Hertz(500)));
    ahrsImuChs.push_back(
        mscl::MipChannel(mscl::MipTypes::CH_FIELD_SENSOR_RAW_GYRO_VEC,
                         mscl::SampleRate::Hertz(100)));
    // set the active channels for the different data classes on the Node
    node.setActiveChannelFields(mscl::MipTypes::CLASS_AHRS_IMU, ahrsImuChs);
		//start sampling the active channels on the AHRS/IMU class of the Node
		node.enableDataStream(mscl::MipTypes::CLASS_AHRS_IMU);
    // endless loop of reading in data
    while (true) {
      // get all the data packets from the node, with a timeout of 500
      // milliseconds
      mscl::MipDataPackets packets = node.getDataPackets(500);

      for (mscl::MipDataPacket packet : packets) {
        // print out the data
        cout << "Packet Received: ";

        // get the data in the packet
        mscl::MipDataPoints data = packet.data();
        mscl::MipDataPoint dataPoint;

        // loop through all the data points in the packet
        for (unsigned int itr = 0; itr < data.size(); itr++) {
          dataPoint = data[itr];

          // print out the channel data
          cout << dataPoint.as_string()
               << " "; // Just printing this out as a string. Other methods (ie.
                       // as_float, as_uint16, as_Vector) are also available.

          // if the dataPoint is invalid
          if (!dataPoint.valid()) {
            cout << "[Invalid] ";
          }
        }
        cout << endl;
      }
    }
  } catch (mscl::Error &e) {
    cout << "Error: " << e.what() << endl;
  }

  system("pause");
  return 0;
}
```

Thanks again.

rwslord, 2018-08-08:
Awesome! No problem!

closed 2018-08-08 by rwslord