LORD-MicroStrain/MSCL #62: G-Link Stop DataArmed Sampling

BaldeepSinghAtMack, 2018-08-21:
Hi,
we are using the G-Link sensors for moving objects at 15kmh. What is the fastest way to stop sampling apart from 
C#...SetToIdle();
It takes a lot of time.
please help
Thx and regards from Germany
Baldeep

BaldeepSinghAtMack, 2018-08-21:
We use DataArmedLogging Sampling. The Software looks like

 if (Node.Active)
            {
                for (int i = 1; i <= SearchTry; i++)
                {
                    textBox1.Text = "Search operation Nr : " + i.ToString() + " . Node "+ Node.Address.ToString() + " is being tried to set to Idle Mode."; textBox1.Refresh();
                    try
                    {
                         
                        Node.MSCL_MyNode_status = Node.MSCL_MyNode.setToIdle();
                        Console.WriteLine(i.ToString() + ". Node " + Node.Address.ToString() + " is being tried to set to Idle Mode.");
                        DateTime start = DateTime.Now;
                        while (!Node.MSCL_MyNode_status.complete(300))
                        {
                            if (progressBar2.Value >= 100)
                            {
                                progressBar2.Value = 0;
                            }
                            progressBar2.Value = progressBar2.Value+1; progressBar2.Refresh();
                            textBox1.Text = textBox1.Text + "."; textBox1.Refresh();
                            Console.Write(".");
                            TimeSpan timeDiff = DateTime.Now - start;
                            if (timeDiff.Seconds >= 10)
                            {
                                Node.MSCL_MyNode_status.cancel();
                            }
                        }
                        switch (Node.MSCL_MyNode_status.result())
                        {
                        //completed successfully
                            case mscl.SetToIdleStatus.SetToIdleResult.setToIdleResult_success:
                                textBox1.Text = "Node "+ Node.Address.ToString() + " is successfully set to Idle Mode."; textBox1.Refresh();
                                Console.WriteLine("Node " + Node.Address.ToString() + " is successfully set to Idle Mode.");
                                i = SearchTry + 1;
                                progressBar2.Value = 100;
                                break;

                        //canceled by the user
                            case mscl.SetToIdleStatus.SetToIdleResult.setToIdleResult_canceled:
                                textBox1.Text = "Node " + Node.Address.ToString() + " Idle routine was canceled after timeout!"; textBox1.Refresh();
                                Console.WriteLine("Node " + Node.Address.ToString() + " Idle routine was canceled after timeout!");
                                break;

                       //Failed to perform the operation
                           case mscl.SetToIdleStatus.SetToIdleResult.setToIdleResult_failed:
                               default:

                                Console.WriteLine("Node " + Node.Address.ToString() + " Idle routine was failed!");
                                textBox1.Text = "Node " + Node.Address.ToString() + " Idle routine was failed!"; textBox1.Refresh();
                                break;
                        }
                    }
                    catch (mscl.Error StopNodeError)
                    {
                    Console.WriteLine(Environment.NewLine + "Node Stop try is failed with Code : " + StopNodeError.Message);
                        i = i - 1;
                    }
                }

BaldeepSinghAtMack, 2018-08-21:
some Notes to times:

baseStation.timeout(300);
config.samplingMode(mscl.WirelessTypes.SamplingMode.samplingMode_armedDatalog);
config.defaultMode(mscl.WirelessTypes.DefaultMode.defaultMode_idle);
config.sampleRate(mscl.WirelessTypes.WirelessSampleRate.sampleRate_256Hz);
config.dataCollectionMethod(mscl.WirelessTypes.DataCollectionMethod.collectionMethod_logOnly);
config.inactivityTimeout(65535);

BaldeepSinghAtMack, 2018-08-21:
ist there something as Network.startSampling().....Network.StopSampling();

rwslord, 2018-08-21:
Our wireless nodes can't continuously listen for the set to idle command, as that would draw too much power. You can adjust the "check radio interval", which controls how often the node listens for these set to idle commands, but this will affect your battery life. To adjust this, use [config.checkRadioInterval](WirelessNodeConfig.h:WirelessNodeConfig.checkRadioInterval).

Another option is to set the [numSweeps](WirelessNodeConfig.h:WirelessNodeConfig.numSweeps) and disable the unlimitedSampling. The node will then automatically stop sampling and return to idle mode after the set number of sweeps.

BaldeepSinghAtMack, 2018-09-10:
Thanks,
i tried with config.checkRadioInterval (1) and then apply config but as i am using three nodes and it still takes a lot of time to stop Data log sampling.

according to the defination of SetToIdle -> **The Set to Idle operation uses all of the BaseStation’s bandwidth to attempt to stop/wake the Node.** 
_can we just use to Frequency 15 instead of trying all the frequencies? if yes then how odes the Syntax look like?_

and second optimisation i seek ist to get data download faster.
right now i use following logic

                   ```
                    while (!Node.MSCL_DataDown.complete())
                    {
                        try
                        {
                           //new mscl.LoggedDataSweep();
                            mscl.LoggedDataSweep sweep = Node.MSCL_DataDown.getNextData();
                          ........
                        }
```
i would like to get more sweeps instead single. is it possible, if yes then how ?
Thanks for your help!
Regards,
Baldeep

rwslord, 2018-09-10:
How long is it taking you to set a single node to idle?

Have you tried disabling unlimited sampling and setting a specific number of sweeps/duration that you want to sample?

I'm not sure what you mean by "can we just use to Frequency 15 instead of trying all the frequencies." Could you elaborate on that? I don't see a spot in your code that is attempting to run set to idle on multiple frequencies. If that is the case, you definitely don't need to do this, unless you are unaware of the frequency a node is on.

There is an additional command you could use to potentially speed things up: [BaseStation.broadcastSetToIdle](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/BaseStation.h:BaseStation.broadcastSetToIdle) will perform the Set to Idle operation that can be heard by **ALL** nodes on the same frequency as the BaseStation. The caveat is that this broadcast operation doesn't ever "complete," as it isn't targeting any specific node. So if you use this one, you will need to run it for a bit, cancel it, then attempt to [ping](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessNode.h:WirelessNode.ping) each node that you were trying to set to idle to see if it was successful. You will then need to repeat this process until all the nodes are set to idle. Also, be mindful that all nodes on your frequency will be set to idle, so if you have multiple groups of nodes within the same area, you will potentially be affecting those as well.

BaldeepSinghAtMack, 2018-09-10:
If i use three nodes (G-Link), all configured with the command config.checkRadioInterval (1) and the whole system is configured with Bandwidth Freq_15 (2.425 GHz). Everything works nice apart from 
1. the Stopping the Sampling. For this purpose i used following code
`        private void IdleThisNode(ref obj_Lord_Node Node)
        {
            progressBar2.Value = 0;
            if (Node.Active)
            {
                for (int i = 1; i <= SearchTry; i++)
                {
                    textBox1.Text = i.ToString() + ". Node " + Node.Address.ToString() + " is being tried to set to Idle Mode."; textBox1.Refresh();
                    try
                    {
                        Node.MSCL_MyNode_status = Node.MSCL_MyNode.setToIdle();
                        Console.WriteLine(i.ToString() + ". Node " + Node.Address.ToString() + " is being tried to set to Idle Mode.");
                        DateTime start = DateTime.Now;
                        while (!Node.MSCL_MyNode_status.complete(1000))
                        {
                            if (progressBar2.Value >= 100)
                            {
                                progressBar2.Value = 0;
                            }
                            progressBar2.Value = progressBar2.Value + 1; progressBar2.Refresh();
                            textBox1.Text = textBox1.Text + "."; textBox1.Refresh();
                            Console.Write(".");
                            TimeSpan timeDiff = DateTime.Now - start;
                            if (timeDiff.Seconds >= 10)
                            {
                                Node.MSCL_MyNode_status.cancel();
                            }
                        }
                        switch (Node.MSCL_MyNode_status.result())
                        {
                            //completed successfully
                            case mscl.SetToIdleStatus.SetToIdleResult.setToIdleResult_success:
                                textBox1.Text = "Node " + Node.Address.ToString() + " is successfully set to Idle Mode."; textBox1.Refresh();
                                Console.WriteLine("Node " + Node.Address.ToString() + " is successfully set to Idle Mode.");
                                i = SearchTry + 1;
                                progressBar2.Value = 100;
                                break;

                            //canceled by the user
                            case mscl.SetToIdleStatus.SetToIdleResult.setToIdleResult_canceled:
                                textBox1.Text = "Node " + Node.Address.ToString() + " Idle routine was canceled after timeout!"; textBox1.Refresh();
                                Console.WriteLine("Node " + Node.Address.ToString() + " Idle routine was canceled after timeout!");
                                break;

                            //Failed to perform the operation
                            case mscl.SetToIdleStatus.SetToIdleResult.setToIdleResult_failed:
                            default:

                                Console.WriteLine("Node " + Node.Address.ToString() + " Idle routine was failed!");
                                textBox1.Text = "Node " + Node.Address.ToString() + " Idle routine was failed!"; textBox1.Refresh();
                                break;
                        }
                    }
                    catch (mscl.Error IdleNodeError)
                    {
                        Console.WriteLine(Environment.NewLine + "Node Stop try is failed with Code : " + IdleNodeError.Message);
                        i = i - 1;
                    }
                }
            }
        }`

rwslord, 2018-09-10:
ok, how long is it taking to set nodes to idle?

BaldeepSinghAtMack, 2018-09-10:
roughly 30 seconds 
lets say with stopwatch
node 1 stops at time 8 seconds, then i say set to idle node 2
node 2 stops at 18 sec., then i say set to idle node 3
node 3 stops at 28 sec.

BaldeepSinghAtMack, 2018-09-10:
regardsing downloading it takes 40 minutes to download all the nodes if they are all 100% full.
i used 
mscl.LoggedDataSweep sweep = Node.MSCL_DataDown.getNextData();
to get sweeps after sweeps and then i write them in file sweep wise.
i would have fetched say 100 sweeps in buffer and then wrote them in file. and then again hundred.....

BaldeepSinghAtMack, 2018-09-10:
regardsing downloading it takes 40 minutes to download all the nodes if they are all 100% full.
i used
mscl.LoggedDataSweep sweep = Node.MSCL_DataDown.getNextData();
to get sweeps after sweeps and then i write them in file sweep wise.
i would have fetched say 100 sweeps in buffer and then wrote them in file. and then again hundred....

rwslord, 2018-09-10:
Can you tell me the model number of the G-Link you are using? It sounds like it might be one of our legacy products. These products actually didn't use the Check Radio Interval setting for anything more than Sleep Mode, so it isn't helping you at all to have that set lower. I think your best bet would (if possible for your setup) would be setting a specific number of sweeps that you want to sample for so that the Nodes automatically go back to idle every time.

As far as the the 40 minutes download goes, I would again need to know the model number of your device to know if this is an expected time to wait or not. Also, are you using a USB/Serial connected BaseStation, or a ethernet connected BaseStation to do the download?

BaldeepSinghAtMack, 2018-09-12:
[image]

rwslord, 2018-09-12:
Thanks. So this is definitely our legacy hardware, which means the Check Radio Interval setting doesn't affect the sampling mode, so that won't help you set to idle any faster during sampling. I believe specifying the duration of sampling will be your best option, or you could try using the broadcast set to idle function, with the caveats that I indicated previously.

As far as the download speed, could you take one of the Nodes, when 100% full, and attempt to download the data in SensorConnect, or Node Commander? 40 minutes to download from 3 of these G-Links is unexpected.

BaldeepSinghAtMack, 2018-09-13:
`do
                        {
                            baseStation.broadcastSetToIdle();
                            Thread.Sleep(100);
                            baseStation.broadcastSetToIdle().cancel();
                            Thread.Sleep(150);
                            Node.MSCL_MyNodeResponse = Node.MSCL_MyNode.ping();
                        }
                        while (!Node.MSCL_MyNodeResponse.success());`

BaldeepSinghAtMack, 2018-09-13:
**Thanks for your help, i appreciate that!! **
_The idea of sending the whole Nodes Bunch SETtoIDLE is actually what i seek._
i am posting the code which i tried,
Result : 
sometimes my hardware goes in Idle line in 4-5 seconds and sometimes it takes 40 seconds. 
Is there anything i can do with the Times

rwslord, 2018-09-13:
The only other thing I can suggest is letting the broadcastSetToIdle run for a bit longer. It's only running for 100ms at a time in your code snippet. You might want to bump that up some and see what affect it has.

If you want the best case scenario, you should specify the number of samples you want, and avoid the set to idle operation altogether.

mglord, 2021-06-15:
It sounds like the `broadcastSetToIdle()` function is your best bet if you're not able to set a pre-determined sample time. Just to reiterate: this will potentially set all nodes within communication range on the base station's frequency to idle. Use with caution.

Regarding the download speeds, I would try the following:

- Attempt downloading in SensorConnect to see how long this takes
- Remove any data storage from your download functionality and see if that improves the speed - if there is significant improvement it is likely your data storage logic is causing delay between `getNextData()` calls.

I hope you were able to get this working for your application! I'm going to go ahead and close this, but if you continue to run into issues please feel free to re-open this or enter another issue. Thanks!

closed 2021-06-15 by mglord