LORD-MicroStrain/MSCL #63: Connecting To WSDA-2000 (I did read FAQ's already)

kgreen, 2018-08-30:
I am building an app that needs to communicate with the WSDA-2000 in C#, running on WIndows 10.

I am connecting via USB(cannot use ethernet). When I connect the USB with no software installed, the device shows as a Virtual COM port. Once I install SensorConnect it shows as a Network Adapter(No more COM port).

After looking at the FAQ's and all through the example code and documentation there is nothing about this step in the process(other than expecting you to "just know" what the port is.
I have a lot of experience integrating with usb/serial devices.  It seems I am missing something that should be obvious.

How do I know what the COM port is once the drivers are installed?
It seems Sensor Connect can discover the device automatically. Some insight into how this is being done would be helpful as well.

Thanks
Kurt

jpherbst, 2018-08-30:
Hi Kurt,

The WSDA-2000 over usb is a virtual Ethernet port, not a virtual serial port.  I’m not sure why it showed up as a COM port without drivers, but Windows sometimes does weird things when it doesn’t have the right drivers.

MSCL can discover basestations, including WSDA-2000s, using the mscl.Devices.listBaseStations() function, documented here:
http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#CClass:Devices:listBaseStations

Each WSDA has a static IP that is used to talk to it over USB, that IP is on the sticker on the side of your WSDA, USB IP, and is generated from the serial number.  MSCL uses a TCP connection to talk to a WSDA-2000.

-          Jon

kgreen, 2018-08-31:
Thanks for your help. I didn't see or read about the IP address on the tag.  Must have overlooked teh listBaseStations() function.
Maybe noting this in the getting started docs would be helpful?
Thanks again!
Kurt

closed 2018-08-31 by kgreen