LORD-MicroStrain/MSCL #64: Other attached usb device causes listBaseStations() to return WSDA-2000 as COM Device

kgreen, 2018-09-17:
In our application we are using a WSDA-2000 base station. We also have a USB device that is a shows as a virtual COM port. Here are details on the device:
[image]

When I call Devices.listBaseStations() the station returns, but BaseStationInfo.Key is "COM4", so when I call mscl.Connection.TcpIp(actualMicrostrainObject.BaseStationInfo.Key, 5000); it Throws an exception.

If I unplug the other device everything works as expected.

kgreen, 2018-09-17:
No, I only have 1 base station, the wsda-2000. The other device is a serial-to usb converter connected to a Card Mag Stripe reader.

rwslord, 2018-09-17:
Ah, ok. Thanks for clearing that up.

Unfortunately, some of our older USB devices did not use a custom VID/PID, so MSCL has to just assume that SiLabs drivers are one of our products (which will error out when you try to send any of our commands to it).

There is, however, a way to check if the device info should be used to create a TCP/IP connection, or a Serial connection. In the list that you get back from the [listBaseStations()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:Communication/Devices.h:Devices.listBaseStations) call, use the [DeviceInfo::connectionType()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:Communication/Devices.h:DeviceInfo.connectionType) function to check if the type is [DeviceInfo::ConnectionType::connectionType_tcp](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:Communication/Devices.h:DeviceInfo.ConnectionType).

kgreen, 2018-09-17:
Ok that should work. Thanks

rwslord, 2018-09-17:
No problem. Feel free to reopen this if it doesn't solve your issue.

closed 2018-09-17 by rwslord