LORD-MicroStrain/MSCL #65: MSCL for 32-bit raspian

johanode, 2018-10-10:
Is there a pre-built python package for 32-bit linux like raspbian?
I have tried to build MSCL for Python on a raspberry pi (raspbian os) using the instructions https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Linux.md. However, the raspberry pi freeze on common.copy /usr/share/[package_name]/programs/lib/libboost_regex.a 

Best,
Johan

rwslord, 2018-10-10:
We do not currently have any pre-built packages for raspbian. Does it always fail at the same spot?

johanode, 2018-10-10:
Running it again, it fails at different spots. It seems the rpi sooner or later runs out of memory. 
[image]

johanode, 2018-10-10:
Leaving out the -j2 option for the b2 install will reduce memory by half (but it will take twice the time). For now it seems to work.

rwslord, 2018-10-10:
Ah, sorry about that. We will try to make that setting more obvious/configurable in the future. Cross-compiling might also be a good option.

johanode, 2018-10-10:
No worries. The boost is installed ok. However, during mscl build it got stuck again. Seems 1024 MB is not enough, or is there a way to solve this memory issue?

johanode, 2018-10-11:
E.g. is there a way to run bjam that uses less memory?

rwslord, 2018-10-11:
That's strange. We were able to use the same process you are using to build MSCL for the Raspberry Pi 3 model B, which has 1GB of RAM. When it gets stuck, how long do you wait before pulling the plug?

johanode, 2018-10-11:
Reducing GPU and other interfaces/activities, I manage to build MSCL. Thank you for the support.

rwslord, 2018-10-11:
awesome! glad you got it working

closed 2018-10-11 by rwslord