LORD-MicroStrain/MSCL #66: Discovering base stations on the network

kgreen, 2018-10-11:
I would like to find the base station connected to a switch on the network. Is there a way to do this through MSCL or do I need find any devices that are communicating over port 5000 on my network and assume they are base stations? I'm curious as to why Devices.listBaseStations() doesn't have an option to search the network....

rwslord, 2018-10-11:
If you are on Windows, you can use the [WsdaFinder](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:Communication/WsdaFinder.h) class to find WSDA-Pro (-1000, -1500, and -2000) devices that were found with upnp.

To use this class, you simply create an instance of the `WsdaFinder` object, and then call `.found()`.

kgreen, 2018-10-11:
perfect thanks!

closed 2018-10-11 by rwslord

kgreen, 2018-10-11:
Using SensorConnect, I can click "Add Device", and the IP address of the base station shows in a list of available devices.
[image]

In my app using WsdaFinder, found returns 0 devices:
[image]

rwslord, 2018-10-11:
SensorConnect is definitely using the same MSCL code for that screen. Try calling `.found` repeatedly for a while. Sometimes it takes a bit to find after creating the `WsdaFinder` object.

kgreen, 2018-10-11:
That did the trick. Thanks again.

rwslord, 2018-10-11:
no problem!