LORD-MicroStrain/MSCL #67: RuntimeLibrary Error

sejiny3, 2018-10-26:
Hi,

I am including the prebuilt MSCL library "mscl_44.0.2_Windows_C++" on Windows 10 with Visual Studio 2015.
After setting the include, lib folders and Linker's input and debugging in my own project,  I got the error : 

LNK2038 mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MDd_DynamicDebug' in PCL_Viewer.obj

Please give me help, thank you.

rwslord, 2018-10-26:
It looks like you are trying to link the static MSCL library against your program which you are trying to build with the Multithreaded Debug DLL setting. You should be able to go into the properties in your Visual Studio project and change the Run-Time library setting (`C/C++` / `Code Generation` / `Runtime Library`) to MultiThreadedDebug (/MTd).

It might also be helpful to download the (MSCL Example Code)[https://github.com/LORD-MicroStrain/MSCL/tree/master/MSCL_Examples] and try out the C++ projects there.

sejiny3, 2018-10-29:
Thank you for replying,
I changed the Run-Time library setting to /MTd.
However, the following message was shown;
'MDd_DynamicDebug' doesn't match value 'MTd_StaticDebug' in PCL_Viewer.obj

If possible, please help me again.

rwslord, 2018-10-29:
Could you grab the example code and try out the C++ project there? If you can get that to compile, you should be able to look at the project properties and mirror those settings.

closed 2021-06-15 by mglord