LORD-MicroStrain/MSCL #68: Can't stop nodes G-Link 200 8g OEM and WSDA-200-USB

mt2h, 2018-11-27:
Dear developers
I have a WSDA-200-USB with a node G-Link 200 8g OEM and this doesn't respond to stop command but if the commands of cycle power and show information. I installed the Linux driver as the README indicates in the repository, I  also tried with a baud rate of 3000000 but I can't to stop and to ping. Finally I also update the firmware to WSDA-200-USB and it doesn't work.

rwslord, 2018-11-28:
Which programming language are you using with mscl?

How did you go about building mscl for Linux?

Could you show the code you are using and indicate which parts work, and which are failing?

cmonasterio, 2018-11-28:
I'm working with MSCL for Linux

My code using (example part)

`from mscl import mscl

try:
    connection = mscl.Connection.Serial("/dev/ttyUSB0", 3000000)
    baseStation = mscl.BaseStation(connection)
    node = mscl.WirelessNode(14736, baseStation)
    status = node.setToIdle()

    print "Setting Node to Idle",

    while not status.complete(300):
        print ".",

    result = status.result()
    if result == mscl.SetToIdleStatus.setToIdleResult_success:
        print "Successfully set to idle!"
    elif result == mscl.SetToIdleStatus.setToIdleResult_canceled:
        print "Set to Idle was canceled!"
    else:
        print "Set to Idle has failed!"

except Exception, e:
    print "Error:", e`

In this case it stays stuck in the attempt and doesn't respond to the stop command

My code (production part)

`        while nodes and (time.time() - t) <= timeout :
            print(".", end="")
            sys.stdout.flush()
            tries = total_tries.pop(0)
            tries += 1
            status = node.setToIdle()
            if status.complete(1000): # and node.quickPing():
                print("")
                self.log.info("Node %d stopped at %d tries" % (node.nodeAddress(), tries))`

In this case also it stays stuck in the attempt and doesn't respond to the stop command

mt2h, 2018-11-28:
I'm working with MSCL for Linux

My code using (example part)

`from mscl import mscl

try:
    connection = mscl.Connection.Serial("/dev/ttyUSB0", 3000000)
    baseStation = mscl.BaseStation(connection)
    node = mscl.WirelessNode(14736, baseStation)
    status = node.setToIdle()

    print "Setting Node to Idle",

    while not status.complete(300):
        print ".",

    result = status.result()
    if result == mscl.SetToIdleStatus.setToIdleResult_success:
        print "Successfully set to idle!"
    elif result == mscl.SetToIdleStatus.setToIdleResult_canceled:
        print "Set to Idle was canceled!"
    else:
        print "Set to Idle has failed!"

except Exception, e:
    print "Error:", e`

In this case it stays stuck in the attempt and doesn't respond to the stop command

My code (production part)

`    def stop_nodes(self, timeout=125):
        nodes = self.wirelessNodes[::]
        total_tries = [0] * len(nodes)
        t = time.time()
        self.log.info("Stoping nodes...")
        while nodes and (time.time() - t) <= timeout :
            print(".", end="")
            sys.stdout.flush()
            node = nodes.pop(0)
            tries = total_tries.pop(0)
            tries += 1
            status = node.setToIdle()
            if status.complete(1000): # and node.quickPing():
                print("")
                self.log.info("Node %d stopped at %d tries" % (node.nodeAddress(), tries))
            else:
                total_tries.append(tries)
                nodes.append(node)`

In this case also it stays stuck in the attempt and doesn't respond to the stop command

rwslord, 2018-11-28:
Ok, a few more questions:

What method/instructions did you follow to build mscl for python on Linux? 

What version of Linux are you using?

Does any node communication work? Can you try the node ping example code? Or maybe try printing out the serial number to see if you can get that far?

mt2h, 2018-11-28:
I'm using Ubuntu 16.04 Server LTS and the installation method was with installer for linux https://github.com/LORD-MicroStrain/MSCL/releases/download/v44.0.2/python2-mscl_44.0.2_amd64_ubuntu16.04.deb with sudo apt install -f. This works very well for the old nodes and old bases for example SG-Link OEM with basestation wsda LXRS 104, but with WSDA-200-USB I'm having the problem where I'm can't stop and ping all nodes, as I can mention I can see the information of a node and find it using the cycle power function also ping but only basestation more specifically  t stays stuck in the attempt and doesn't respond to the stop command with node.setToIdle() and  if status.complete(300).

closed 2024-10-24 by github-actions[bot]